zephyrproject-rtos/zephyr #30117: Cannot compile Zephyr project with standard macros INT8_C, UINT8_C, UINT16_C

rohand87, 2020-11-18:
Hello,

I am trying to build [this library](https://github.com/BoschSensortec/BMP3-Sensor-API) in Zephyr. I am getting compiler errors for `implicit declaration of function 'INT8_C', UINT8_C, UINT16_C`. These are standard macros from `stdint.h`. I can build the same code in other platforms but not in Zephyr. I am not sure whether it is a issue with my setup. Please help.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the attached project
2. Create an alias in device tree called i2c0. I am using a custom board
3. west build -b <board>
4. Get the below warnings which ultimately lead to errors

**Impact**
showstopper

**Logs and console output**
```
In file included from ../src/bmp3.h:47,
                 from ../src/bmp3.c:41:
../src/bmp3.c: In function 'bmp3_init':
../src/bmp3_defs.h:280:49: warning: implicit declaration of function 'INT8_C'; did you mean 'INT8_MAX'? [-Wimplicit-function-declaration]
  280 | #define BMP3_OK                                 INT8_C(0)
      |                                                 ^~~~~~
../src/bmp3.c:708:17: note: in expansion of macro 'BMP3_OK'
  708 |     if (rslt == BMP3_OK)
      |                 ^~~~~~~
../src/bmp3_defs.h:167:49: warning: implicit declaration of function 'UINT8_C'; did you mean 'UINT8_MAX'? [-Wimplicit-function-declaration]
  167 | #define BMP3_REG_CHIP_ID                        UINT8_C(0x00)
      |                                                 ^~~~~~~
```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.11.4
 - Version: 2.4.99

**Attachment**
[bmp3_zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/5561257/bmp3_zephyr.zip)

Thanks

galak, 2020-11-18:
I think this is just a missing feature of `lib/libc/newlib/include/stdint.h`

rohand87, 2020-11-19:
Thanks for your reply @galak. What should be a good workaround to resolved this issue?

galak, 2020-11-19:
> Thanks for your reply @galak. What should be a good workaround to resolved this issue?

you can submit a PR to add the macro's to `lib/libc/newlib/include/stdint.h`

PR #30498 (2021-01-21): minimal libc: close gap in functionality for integer types
Add the macros `INTn_C()` which expand to constant integer expressions of the appropriate type.  E.g. `INT64_C(1)` should become `1`, `1L`, or `1LL` depending on host capabilities.

This will currently only work if a toolchain that supports the GNUC intrinsic `__INTn_C()` functions is used.

Also fix a problem with the c_lib test, and correct the values of the `PRI?64/MAX` macros to use length modifiers that are more correct.

Fixes #30117
closed 2021-01-21 by nashif