zephyrproject-rtos/zephyr #30121: Make log subsystem power aware

ceolin, 2020-11-18:
**Is your enhancement proposal related to a problem? Please describe.**
Currently the log subsystem is not power aware, so its thread can change the behavior / timing expected by an application.

The idle thread can take longer time to execute and start a power management state transition than expected.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/issues/29330

nashif, 2020-11-18:
I do not think we should be doing much here and this is not limited to logging, there are other service threads that can interfere. k_sleep() is right now per thread, is we are looking at putting the system to sleep we need some other way for forcing a transition by not scheduling other lower priority threads (service).

ceolin, 2020-11-19:
I put here to lose this information, may I should re-phrase and say something like "system services".  I don't particular think that the k_sleep is a problem, any application should be relying on it for power management IMHO, but if one manually call something like "sys_suspend()" then there should be a way to notify all system threads about it.

closed 2021-03-11 by nashif