zephyrproject-rtos/zephyr #30126: xtensa-asm2-util.s hard coding

hongshui3000, 2020-11-19:
**Describe the bug**
```
/*
 * _restore_context
 *
 * Arrive here via a jump.  Enters into the restored context and does
 * not return.  A1 should have a context pointer in it as received
 * from switch or an interrupt exit.  Interrupts must be disabled,
 * and register windows should have been spilled.
 *
 * Note that exit from the restore is done with the RFI instruction,
 * using the EPCn/EPSn registers.  Those will have been saved already
 * by any interrupt entry so they are save to use.  Note that EPC1 and
 * RFE are NOT usable (they can't preserve PS).  Per the ISA spec, all
 * RFI levels do the same thing and differ only in the special
 * registers used to hold PC/PS, but Qemu has been observed to behave
 * strangely when RFI doesn't "return" to a INTLEVEL strictly lower
 * than it started from.  So pick level 6 (the highest that works on
 * Qemu, hardware doesn't care so it doesn't matter).  In theory we
 * should test to be able to support hardware with less than 6 levels,
 * though...
 */
.global _restore_context
_restore_context:
	call0 xtensa_restore_high_regs

	l32i a0, a1, BSA_PC_OFF
	wsr.EPC6 a0 ---------------------------------------------->hard coding
	l32i a0, a1, BSA_PS_OFF
	wsr.EPS6 a0   ---------------------------------------------->hard coding

	l32i a0, a1, BSA_SAR_OFF
	wsr.SAR a0
#if XCHAL_HAVE_LOOPS
	l32i a0, a1, BSA_LBEG_OFF
	wsr.LBEG a0
	l32i a0, a1, BSA_LEND_OFF
	wsr.LEND a0
	l32i a0, a1, BSA_LCOUNT_OFF
	wsr.LCOUNT a0
#endif
#if XCHAL_HAVE_S32C1I
	l32i a0, a1, BSA_SCOMPARE1_OFF
	wsr.SCOMPARE1 a0
#endif
#if XCHAL_HAVE_THREADPTR && defined(CONFIG_THREAD_LOCAL_STORAGE)
	l32i a0, a1, BSA_THREADPTR_OFF
	wur.THREADPTR a0
#endif
	rsync

	l32i a0, a1, BSA_A0_OFF
	l32i a2, a1, BSA_A2_OFF
	l32i a3, a1, BSA_A3_OFF
	addi a1, a1, BASE_SAVE_AREA_SIZE

	rfi 6                                   ---------------------------------->hard coding
```

If the core architecture of the chip does not support EPC6，EPS6 , the compilation will fail。There is no external option to change this result。

dcpleung, 2020-12-11:
Note that `DEF_INT_C_HANDLER()` does not refer to EPS/EPC registers so no issue there.

hongshui3000, 2020-12-14:
> Note that `DEF_INT_C_HANDLER()` does not refer to EPS/EPC registers so no issue there.
yes,But there will be other warnings.When there is no interruption of leve6 and leve7, the interrupt function _xtensa_handle_one_int6 and _xtensa_handle_one_int7 will not exist
msg：
```
E:\work\rtos_work\zephyrproject\zephyr>west build -b my_demo samples/hello_world
[1/7] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/xtensa-asm2.c.obj
E:/work/rtos_work/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c: In function 'xtensa_int6_c':
E:/work/rtos_work/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:158: warning: implicit declaration of function '_xtensa_handle_one_int6'
E:/work/rtos_work/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c: In function 'xtensa_int7_c':
E:/work/rtos_work/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:159: warning: implicit declaration of function '_xtensa_handle_one_int7'
```

PR #30668 (2021-01-05): xtensa: use highest available EPC/EPS regs in restore context
There may be Xtensa SoCs which don't have high enough interrupt
levels for EPC6/EPS6 to exist in _restore_context. So changes
these to those which should be available according to the ISA
config file.

Fixes #30126
commit 0d7bdbc8 (2021-01-05): xtensa: use highest available EPC/EPS regs in restore context

There may be Xtensa SoCs which don't have high enough interrupt
levels for EPC6/EPS6 to exist in _restore_context. So changes
these to those which should be available according to the ISA
config file.

Fixes #30126
closed 2021-01-05 by andrewboie