zephyrproject-rtos/zephyr #30129: TCP2 send test

liang8057, 2020-11-19:
**Describe the bug**
I want to test send data via TCP socket, but the device stopped after a minute.

**To Reproduce**

	int sock;

	if((sock = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP)) < 0)
	{
	 return;
	}

	struct sockaddr_in addr;
	addr.sin_family = AF_INET;
	addr.sin_port = htons(950);
	inet_pton(AF_INET, "10.144.8.81", &addr.sin_addr);

	if(connect(sock, &addr, sizeof(addr)) < 0)
	{
	 return;
	}
	
	while(1)
	{
	 send(sock, "test", 4, 0);
	 k_sleep(K_SECONDS(1));
	}

**Logs and console output**
[image]

[image]

**Environment (please complete the following information):**
 - OS: Zephyr2.4
 - Toolchain:  zephyr-sdk-0.11.2
 - Device:  MIMXRT1060-EVK

liang8057, 2020-11-19:
configuration:
CONFIG_NET_TCP2=y
CONFIG_NET_TCP=n

jukkar, 2020-11-19:
> configuration:
> CONFIG_NET_TCP2=y
> CONFIG_NET_TCP=n

Note that the above will turn off TCP.
Anyway, I will investigate this.

jukkar, 2020-11-19:
You seem to be using 2.4, can you try with the latest master as we have fixed lot of TCP2 issues.

liang8057, 2020-11-19:
I tried to send a larger packet(536 bytes) and then the send function returned an error code of -1.
I tried with the latest master, but the result is still the same.

[image]

[image]

liang8057, 2020-11-19:
[image]

jukkar, 2020-11-19:
Is this socket connection happening when the device boots?
If yes, does it crash if you add `k_sleep(K_SECONDS(5));` before `socket()` call?

liang8057, 2020-11-20:
I have added k_sleep(K_SECONDS(5)) before socket().
The result is the same.

jukkar, 2020-11-20:
> The result is the same.

Ok, then it is not the same issue I saw. There is an issue if the network interface does not have IP address set when the socket is bound. So in my tests (using you test app), it helped to delay the socket creation. In various network sample applications in `samples/net/sockets/`, the app will "start" after the device is properly setup (has IP addresses etc).

jukkar, 2020-11-20:
I have not been able to reproduce this issue. I used mimxrt1050_evk, qemu_x86 and native_posix boards.
My config file looks like this:
```
CONFIG_NETWORKING=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SHELL=y

# Network application options and configuration
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
```
and the application looks like this:
```
#include <zephyr.h>
#include <errno.h>
#include <stdio.h>

#include <net/socket.h>

void main(void)
{
	struct sockaddr_in addr;
	int sock;

	// Let the system to have some time set the source address to the network interface
	k_sleep(K_SECONDS(1));

	if((sock = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP)) < 0) {
		return;
	}

	addr.sin_family = AF_INET;
	addr.sin_port = htons(9500);

	inet_pton(AF_INET, CONFIG_NET_CONFIG_PEER_IPV4_ADDR, &addr.sin_addr);

	printk("Connecting....\n");
	if(connect(sock, (const struct sockaddr *)&addr, sizeof(addr)) < 0) {
		return;
	}

	printk("Connected, sending data to sock %d\n", sock);
	while(1) {
		send(sock, "test", 4, 0);
		k_sleep(K_SECONDS(1));
	}
}
```
In this test, the system worked ok and did not crash. Before trying to create the socket, it is mandatory that the network interface has proper IP address in it. In this simple app, I used a small sleep for that so that the config library has time to set the address. In a proper application, we would wait until the system send an event that tells the application that the network connectivity is ok.

jukkar, 2021-02-09:
There has been important fixes in TCP2 recently and I have not seen this issue. Closing this atm, please reopen if needed.

closed 2021-02-09 by jukkar