zephyrproject-rtos/zephyr #30130: nrf_radio_power_set() should use bool

nbrunner, 2020-11-19:
in radio_reset() found in /subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio.c.
The function nrf_radio_power_set() need a bool for is second parameter.
So the call should look like that:
nrf_radio_power_set(NRF_RADIO, false);
nrf_radio_power_set(NRF_RADIO, true);

cvinayak, 2020-11-20:
as `nrf_radio_power_set` in nrf_radio.h does not use the bit position and bit mask like it is done for all other registers, BLE controller code will keep the existing code to pass the value correctly using the `RADIO_POWER_POWER_Disabled << RADIO_POWER_POWER_Pos) & RADIO_POWER_POWER_Msk` construct. (bool does evaluate to 0 or 1, also, existing code is not incorrect in passing 0 or 1 too.).  The BLE code can be changed when nrf_radio.h takes care of bit values and does not directly assign bool value.

refer to: https://github.com/zephyrproject-rtos/hal_nordic/blob/master/nrfx/hal/nrf_radio.h#L1896

nbrunner, 2020-11-20:
Yes you're right, i will ask Nordics to correct there bug.

carlescufi, 2020-11-26:
Closing this since this is really an nrfx issue.

closed 2020-11-26 by carlescufi