zephyrproject-rtos/zephyr #30133: sensor: driver: lis2dh interrupt definitions

WilliamGFish, 2020-11-19:
**Enhancement proposal description.**
With the lis2dh sensors would it be possible to have the interrupts triggers user assigned? Currently int1 is drdy and int2 motion, I believe that the either can be used for both, therefore could be assigned dependant on need.

Create ability for user to assign trigger type to sensor interrupt dependent on need.

avisconti, 2020-11-20:
Yes, I got this point. I'll address it as soon as possible.

WilliamGFish, 2020-11-20:
Thanks, will make the thingy even better as a prototyping tool....

Billy..

avisconti, 2020-11-26:
Hi @WilliamGFish 

I started today looking into that issue, but I think that unfortunately the lis2dh12 device can
only have DRDY on INT1 and AnyMotion (or other event combinations) on INT2.
So, we cannot do antything about it.

Just a question related to thingy52: isn't the DRDY fixed on INT1 in hardware?

WilliamGFish, 2020-11-30:
@avisconti 
If you look at the st spec it suggests this is a mater of cfg (https://www.st.com/resource/en/application_note/dm00365457-lis2dh12-mems-digital-output-motion-sensor-ultralowpower-highperformance-3axis-nano-accelerometer-stmicroelectronics.pdf)
> The LIS2DH12 interrupts signals can behave as free-fall, wake-up, 6D and 4D orientation
> detection, and click detection. These signals can be driven to the two interrupt pins (INT1
> and INT2).

I think the zephyr driver was done that way for convenience. I'm not that great with MEM drivers so I always look at the manufacture for guidance; here is their github c code driver: 
https://github.com/STMicroelectronics/STMems_Standard_C_drivers/tree/master/lis2dh12_STdC

Hope this helps.
Billy..

avisconti, 2020-11-30:
> @avisconti
> If you look at the st spec it suggests this is a mater of cfg [...]

I think that you can drive the events interrupts to INT1 or INT2, but you cannot drive DRDY on INT2.

> I think the zephyr driver was done that way for convenience. I'm not that great with MEM drivers so I always look at the manufacture for guidance; here is their github c code driver:
> https://github.com/STMicroelectronics/STMems_Standard_C_drivers/tree/master/lis2dh12_STdC

Yes, I sit beside the guy that writes the standard C drivers, and I as well wrote some of them in the past. I can ask to him as well as to designers to be 100% sure of what I told you.

Just to be sure of what you are asking with this request, are you just interested to define INT1/INT2 just for events interrupt (like Any Motion) or you want also to define it for DRDY? Because for DRDY I think you cannot.

WilliamGFish, 2020-11-30:
@avisconti 
I think it would be super useful if INT1 could be configured to be either DRDY or 'Motion' events. If INT2 can only be event driven then that is that.

This is in part because a number of board only connect INT1, Thingy:52 is an example of one that is on my desk.

Billy..

avisconti, 2020-11-30:
> @avisconti
> I think it would be super useful if INT1 could be configured to be either DRDY or 'Motion' events. If INT2 can only be event driven then that is that.
> 
> This is in part because a number of board only connect INT1, Thingy:52 is an example of one that is on my desk.

@WilliamGFish 
Ah ok, so you would like the opposite of what I thought. You would like to have both DRDY and AnyMotion on INT1 (so avoid using INT2 at all). Let me think about that.

WilliamGFish, 2021-01-15:
@avisconti Any news? Can I help?

avisconti, 2021-01-20:
> @avisconti Any news? Can I help?

Sorry, let me try to take alook again. I fotgot it!

avisconti, 2021-01-20:
> Hi @WilliamGFish
> 
> I started today looking into that issue, but I think that unfortunately the lis2dh12 device can
> only have DRDY on INT1 and AnyMotion (or other event combinations) on INT2.
> So, we cannot do antything about it.
> 

I think that the answer is here. IT is not possible on this sensor to do what you requested.

avisconti, 2021-01-20:
I'm gonna close this request. If you feel that the answwer is not satisfactory you can re-open it.

closed 2021-01-20 by avisconti