zephyrproject-rtos/zephyr #30135: Firmware URI writing does not work anymore

JusbeR, 2020-11-19:
**Describe the bug**
Firmware URI(5/0/1) writing is broken after this addition: https://github.com/zephyrproject-rtos/zephyr/commit/69ca5892567655be1cb7b665b018506d9e2f6aeb#diff-09b6eaae5450ebeecff6e135e0a550242a3256198b6a6600b550b39848e08b2aR922

I am using LWM2M 5/0/1 to deliver a https url to device for FW downloading. After updating LWM2M lib to a version where that line is introduced, package_uri_write_cb in firmware object is not called anymore

I am using leshan in server side and I first read the whole object 5 to understand the current status of the upgrade progress in device. After that I write 5/0/1 to deliver an uri for the device, for getting the FW. If I print the raw data when client receives the URI packet it looks like this:
 ```
48 02 2d 95 d4 80 09 54  74 d8 2e de b1 35 01 30 |H.-....T t....5.0
12 2d 16 ff c8 01 50 68  74 74 70 73 3a 2f 2f 66 |.-....Ph ttps://f

48 COAP start
0100 1000
01-- ---- ver
--00 ---- type=non-conf
---- 1000 token len=8
02 Code
0000 0010
000- ---- Class 0 = request
---0 0010 Code=2=POST

2d 95 MSG ID
d4 80 09 54  74 d8 2e de Token

b1 OPTION
1011 0001
1011 ---- Option delta 11=uri-path
---- 0001 option len=1
35 Option value 35 ascii == 5 == firmware

01 30 reserved, len=1? wtf?

12 If match, len=2?
2d 16 Etag? Reference to some previous GET?

ff Payload marker
c8
1100 1000
11-- ---- Resource with value
--0- ---- Identifier is 8 bits long
---0 1--- Length is 8 bits
---- -000 Ignored
01 ID = 1
50 Len=80
68  74 74 70 73 3a 2f 2f 66 https://f...
```
Pay special attention in how the resource path is given. `5` comes in as COAP option and Resource ID `1` comes in as TLV.

But the offending code line about `5/0/0` is there before the stack actually parses the TLV value and thus msg->path.res_id is always 0.

I have no idea if it is sane/allowed to give these paths in little pieces like this. Maybe this issue should be made to leshan instead?

**To Reproduce**
Might be tricky as you need, Leshan, NRF9160 with SDK 1.4 and the FW object sequence described above. But with unit test it should be quite easy.

**Expected behavior**
Writing an uri will write uri to correct object

**Impact**
Showstopper?, have to fork and remove that line to proceed. Not a big deal for me really, but it simply does not work.

**Logs and console output**
N/A

**Environment (please complete the following information):**
NRF9160, leshan, custom HW

rlubos, 2020-11-19:
Thanks for reporting, I haven't tested writing into `5/0` that way before. If you write directly into `5/0/1` the issue obviously does not occur. I think the most reasonable fix here would be to add an additional check in there, to verify if block transfer is actually used, since the whole point of that code is to prevent TLV parsing issues when FOTA in PUSH mode is used. I'll push a PR rightaway.

rlubos, 2020-11-19:
@JusbeR I've opened https://github.com/zephyrproject-rtos/zephyr/pull/30138 that should address this issue. I've tested with Leshan and all possible combinations of Package/Package URI writes I could come up with and it worked fine, but please let me know if fixes the issue for you.

PR #30138 (2020-12-01): net: lwm2m: Verify if block transfer is used before skipping TLV parsing
Verify if block tranfer is used and not an initial block when skipping
directly to data processing during FW update in PUSH mode.

This fixes a bug, which caused TLV not to be processed when the FW
object was updated as a whole, and actual resource number was encoded in
a TLV (for instance when writing FW Update URI).

Fixes #30135
commit b27c14e3 (2020-12-01): net: lwm2m: Verify if block transfer is used before skipping TLV parsing

Verify if block tranfer is used and not an initial block when skipping
directly to data processing during FW update in PUSH mode.

This fixes a bug, which caused TLV not to be processed when the FW
object was updated as a whole, and actual resource number was encoded in
a TLV (for instance when writing FW Update URI).

Fixes #30135
closed 2020-12-01 by nashif