zephyrproject-rtos/zephyr #30141: qemu_x86 unexpected thread behavior

pabigot, 2020-11-19:
qemu_x86_64 appears to handle semaphores differently with respect to counting and impact on ready but not active threads: it allows cooperative threads to be pre-empted without yielding, and may mishandle semaphore counts.

Using the modified hello-world at https://github.com/pabigot/zephyr/commits/nordic/20201119a  with  qemu_x86:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1701-g414ba6b0741e  ***
Main thread prio: -1
main giving 1
main back 1
main giving 2
main back 2
main giving 3
main back 3
coop_main 0x105000 at prio -3  <- coop_main doesn't start until main yields
coop_main blocking
coop_main took: 0
coop_main blocking
```
With qemu_x86_64:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1701-g414ba6b0741e  ***
Main thread prio: -1        <- coop_main gets to run without main yield
coop_main 0x105000 at prio -3
main giving 1
coop_main blocking
main back 1
main giving 2
main back 2
main giving 3
coop_main took: 0   <- coop_main gets to run without main yield
main back 3
coop_main blocking
coop_main took: 0   <- coop_main gets to run again?
coop_main blocking
```

This causes test failures at https://buildkite.com/zephyr/zephyr/builds/13950#ee0cf79b-78b7-4a52-b4f6-2bd8707822f3 in code that relies on consistent behavior of thread preemption across platforms.

andyross, 2020-11-21:
Are you sure you haven't simply discovered the second CPU?  qemu_x86_64 is an SMP platform, coop_main absolutely will start before main yields if you made it runnable.

pabigot, 2020-11-21:
> Are you sure you haven't simply discovered the second CPU? 

No, that explains it.  I just hadn't encountered that behavior before.  It's definitely non-deterministic with a second processor available, so I'll make sure the tests I'm adding exclude that option, or make use of it judiciously.

closed 2020-11-21 by pabigot