zephyrproject-rtos/zephyr #30153: BSD recv() can not received huge packge(may be 100kB) sustain .

GasperChen, 2020-11-20:
**Describe the bug**
zephyr 2.2.99, mimxrt1064

I want to receive a huge tcp package(up 100KB), no response. then the BSD socket recv() api cound not sustain package by package, It may be have interval between 1 to 5 seconds.
code following:
sock = socket(AF_INET, SOCK_STREAM,IPPROTO_TCP);
bind(sock, bind_addr, bind_addrlen);
listen(sock, 10);
client = accept(sock, (struct sockaddr *)&client_addr,&client_addr_len);
while(1)
{
    iRecvLen = recv(client , pucRecvBuf, 4096, 0);
    printf("TCP Received %d bytes", iRecvLen);
    k_sleep(10);
}

**Expected behavior**
I send a huge package(up 100kB) from client, but recv have a big interval like the following:
[00:00:50.667,000] <dbg> net_echo_server_sample.handle_data: TCP Received 1072 bytes
[00:00:55.668,000] <dbg> net_echo_server_sample.handle_data: TCP Received 208 bytes
[00:00:55.668,000] <dbg> net_echo_server_sample.handle_data: TCP Received 1072 bytes
[00:01:00.670,000] <dbg> net_echo_server_sample.handle_data: TCP Received 208 bytes
[00:01:00.671,000] <dbg> net_echo_server_sample.handle_data: TCP Received 1072 bytes
[00:01:05.671,000] <dbg> net_echo_server_sample.handle_data: TCP Received 208 bytes
[00:01:05.672,000] <dbg> net_echo_server_sample.handle_data: TCP Received 1072 bytes
[00:01:10.673,000] <dbg> net_echo_server_sample.handle_data: TCP Received 208 bytes
[00:01:10.674,000] <dbg> net_echo_server_sample.handle_data: TCP Received 1072 bytes
[00:01:15.675,000] <dbg> net_echo_server_sample.handle_data: TCP Received 208 bytes
[00:01:15.675,000] <dbg> net_echo_server_sample.handle_data: TCP Received 1072 bytes

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain ( Zephyr SDK, ...)

**************
If receviced the package then send to the source socket immediately, the exception will not appear.
*************

jukkar, 2020-11-20:
How many network buffers you have allocated in your app? Can you try to increase the number of RX (and TX) buffers and packets to some higher value.
You are using quite old zephyr version with TCP1 as a default. Can you try with latest master if it helps in your scenario? In master, the TCP2 is the new TCP implementation and it has fixed lot of issues that were present in the legacy TCP implementation.

GasperChen, 2020-11-20:
> How many network buffers you have allocated in your app? Can you try to increase the number of RX (and TX) buffers and packets to some higher value.
> You are using quite old zephyr version with TCP1 [...]

HI, jukkar, Thankyou for you response, I try many config (include increase the number of TX\RX buffer and packets), but the exception is still appear.
I will try to update the version of zephyr to test the exception. then give the result for you.
Thank you!

jukkar, 2021-02-09:
There has been lot of changes to TCP2 which is the default TCP implementation. I have been running `samples/net/sockets/dumb_http_server_mt/` which can serve large file (100kb) to HTTP client. Currently this combo works ok so I will close this issue.
As the TCP1 implementation is deprecated, we do not expected any fixes to it any more and the plan is to remove it in 2.6 release.

closed 2021-02-09 by jukkar