zephyrproject-rtos/zephyr #30159: Clean code related to dts fixup files

erwango, 2020-11-20:
**Is your enhancement proposal related to a problem? Please describe.**
dts fixup files where used prior introduction of new devicetree API.
This is not in use anymore, but walking thought root CMakeLists.txt some references and related code could be found.

**Describe the solution you'd like**
Remove references to fixup files ("fixup_file", "FIXUP_FILE", "dts_fixup", ...) across the tree before we forget why it is there and that it could be safely removed.

**Describe alternatives you've considered**

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2020-11-20:
@galak

erwango, 2021-04-26:
@tejlmand you might be interested by this point

tejlmand, 2021-04-27:
@erwango thanks, I think this could deserve a cleanup. Thanks for pinging me, it's now on my radar.