zephyrproject-rtos/zephyr #30162: Build zephyr with Metaware toolchain for HSDK fails

jacoba-at, 2020-11-20:
**Describe the bug**
Hi,
I'm trying to build the hello_world example from Zephyr 2.4.0 release to the ARC HSDK board using my installed MetaWare toolchain, but the build fails.
I did so by setting the environment variable ZEPHYR_TOOLCHAIN_VARIANT to _arcmwdt_ and ARCMWDT_TOOLCHAIN_PATH to $METAWARE_ROOT/../  per the instructions in the Zephyr documentation.
It seems that the code is compiled for the ARC EM processor instead of the ARC HS, and at some point the compilation fails.

**Logs and console output**
Build log attached.
[build_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/5574407/build_log.txt)

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Metaware P_2019.09

nashif, 2020-11-20:
@abrodkin  @ruuddw FYI

ruuddw, 2020-11-20:
Acknowledged, in 2.4 the MetaWare compiler options have been added for nSIM targets, not yet for other ARC boards.
Eugeniy is addressing this.

evgeniy-paltsev, 2021-01-25:
There are several issues with building with MWDT toolchain for HSDK.

some of them have already been fixed:
- https://github.com/zephyrproject-rtos/zephyr/pull/30168
- https://github.com/zephyrproject-rtos/zephyr/pull/30402

however, we are still missing:
 - https://github.com/zephyrproject-rtos/zephyr/pull/31602
 - atomics fixed for MWDT toolchain (SNPS Jira P10019563-43273)

evgeniy-paltsev, 2021-01-26:
Ok, now we have #31602 merged as well.

carlescufi, 2021-02-09:
> Ok, now we have #31602 merged as well.

Does this mean that the last item missing is:
> atomics to be fixed for MWDT toolchain (SNPS Jira P10019563-43273)

abrodkin, 2021-02-09:
Indeed, atomics are not yet fixed in MetaWare toolchain.
That doesn't interfere with building, but does affect execution on SMP system.
Still it's only one particular atomic operation which is implemented incorrectly (`__atomic_fetch_nand()`), so if you're lucky enough it might not affect your particular application... yet it doesn't mean we don't need to fix it, which we'll obviously do soonish (though I don't have any ETAs yet for this, sorry).

jacoba-at, 2021-04-11:
I tried re-building with MW toolchain on a zephyr 2.5 tree, this time no errors, but still got some warnings which indicate not clean build:

..........
-- The C compiler identification is Clang 9.0.1
-- The CXX compiler identification is Clang 9.0.1
-- The ASM compiler identification is **unknown**
-- Found assembler: /tools/synopsys/metaware/safety_arc/Q-2020.06/MetaWare/arc/bin/ccac
-- **Warning: Did not find file Compiler/-ASM**
CMake Warning at ../../cmake/linker/arcmwdt/target.cmake:137 (message):
  MWDT toolchain does not support

             CONFIG_LINKER_ORPHAN_SECTION_WARN
Call Stack (most recent call first):
  ../../CMakeLists.txt:242 (toolchain_ld_baremetal)

-- Configuring done
-- Generating done
-- Build files have been written to: /space/users/jacoba/zephyr-ws/zephyr/build
-- west build: building application
[1/121] Preparing syscall dependency handling

[118/121] Generating isr_tables.c
MetaWare Hex File Generator Q-2020.06 (build 005)
(c) Copyright 1995-2020, Synopsys, Inc.
[elf2hex] (Warning) Emitting padding from 0x90004742 to 0x9000730c
[elf2hex] (Warning) Emitting padding from 0x900074b0 to 0xfffff800
[elf2hex] FILL LIMIT EXCEEDED BETWEEN ADDRESS 0x900074b0 and 0xfffff800

[121/121] Linking C executable zephyr/zephyr.elf
text	data	bss	dec	hex	filename
546	18116	0	18662	48e6	zephyr.elf
MetaWare Hex File Generator Q-2020.06 (build 005)
(c) Copyright 1995-2020, Synopsys, Inc.

IRISZZW, 2021-04-20:
this warnings is expected:

https://github.com/zephyrproject-rtos/zephyr/blob/facd0f57db85163432bdca7767a05e3f1d742b9f/cmake/linker/arcmwdt/target.cmake#L133-L142

ruuddw, 2021-05-18:
Build issues have been resolved, so closing. Feel free to reopen if there are new issues.

closed 2021-05-18 by ruuddw