zephyrproject-rtos/zephyr #30173: OpenThread commissioning of MTD/SED is broken after "Update nRF5 ieee802154 driver to v1.9"

retfie, 2020-11-21:
**Describe the bug**
After update to latest ToT (03df709a46), commissioning of MTD/SED devices is partial and devices can't join OT network. On FTD devices it's OK.

**To Reproduce**
Steps to reproduce the behavior:

After `git bisect`, last good commit is:
commit: b6b613d4db modules: nordic: Introduce nRF IEEE 802.15.4 Service Layer selection

```
git checkout b6b613d4db
west update -f always
west build -s samples/net/lwm2m_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf overlay-dtls.conf" -DCONFIG_OPENTHREAD_JOINER=y -DCONFIG_OPENTHREAD_JOINER_AUTOSTART=y -DCONFIG_OPENTHREAD_MTD=y -DCONFIG_OPENTHREAD_MTD_SED=y -DCONFIG_OPENTHREAD_POLL_PERIOD=1000 -DCONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768 -DCONFIG_MBEDTLS_HEAP_SIZE=16384
west flash
```

*on border router*
```
sudo wpanctl commissioner joiner-add <eui64> 60 <passphrase>
```

*on node*
```
[00:00:13.045,013] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: 0
[00:00:13.101,135] <inf> net_l2_openthread: State changed! Flags: 0x08040100 Current role: 0
[00:00:13.201,995] <inf> net_l2_openthread: Join success
[00:00:13.202,301] <inf> net_l2_openthread: State changed! Flags: 0x0800101f Current role: 1
[00:00:13.773,071] <inf> net_l2_openthread: State changed! Flags: 0x00000100 Current role: 1
[00:00:14.611,755] <inf> net_l2_openthread: State changed! Flags: 0x001332b7 Current role: 2
uart:~$

uart:~$ ot state
child

uart:~$ ot ipaddr
fd01:2345:6789:0:0:ff:fe00:d42b
fd01:2345:6789:0:e7fe:c39b:2779:4f16
fe80:0:0:0:2c7f:e2b2:8eef:8e7e
fdde:ad00:beef:0:0:0:0:1
```

*from border router*
```
pi@otbr:~ $ ping6 fd01:2345:6789:0:0:ff:fe00:d42b
PING fd01:2345:6789:0:0:ff:fe00:d42b(fd01:2345:6789::ff:fe00:d42b) 56 data bytes
64 bytes from fd01:2345:6789::ff:fe00:d42b: icmp_seq=1 ttl=64 time=771 ms
64 bytes from fd01:2345:6789::ff:fe00:d42b: icmp_seq=2 ttl=64 time=772 ms
```

*First bad commit:*
commit: 35ec164e36 west.yml: Update nRF5 ieee802154 driver to v1.9

```
git checkout 35ec164e36
west update
rm -rf build
west build -s samples/net/lwm2m_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf overlay-dtls.conf" -DCONFIG_OPENTHREAD_JOINER=y -DCONFIG_OPENTHREAD_JOINER_AUTOSTART=y  -DCONFIG_OPENTHREAD_MTD=y -DCONFIG_OPENTHREAD_MTD_SED=y  -DCONFIG_OPENTHREAD_POLL_PERIOD=1000 -DCONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768 -DCONFIG_MBEDTLS_HEAP_SIZE=16384
west flash
```

*during build a warning pop up*
```
zephyrproject/modules/hal/nordic/drivers/nrf_radio_802154/src/nrf_802154_trx_ppi.c:225:19: warning: implicit declaration of function 'nrf_fem_prepare_powerdown' [-Wimplicit-function-decl
aration]
     bool result = nrf_fem_prepare_powerdown(p_instance, compare_channel, PPI_EGU_TIMER_START);
```

*on non working node*
```
[00:00:06.559,478] <inf> net_l2_openthread: State changed! Flags: 0x0800c00b Current role: 0
[00:00:13.672,363] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: 0
[00:00:13.742,767] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: 0
[00:00:13.806,579] <inf> net_l2_openthread: State changed! Flags: 0x08040100 Current role: 0
[00:00:13.906,494] <inf> net_l2_openthread: Join success
[00:00:13.906,860] <inf> net_l2_openthread: State changed! Flags: 0x0800101f Current role: 1
[00:00:14.669,464] <inf> net_l2_openthread: State changed! Flags: 0x00000100 Current role: 1
uart:~$

uart:~$ ot state
detached
Done

uart:~$ ot ipaddr
fdde:ad00:beef:0:4cdc:9890:2dea:36ef
fe80:0:0:0:18a9:79c7:a2d4:27fd
fdde:ad00:beef:0:0:0:0:1
Done

uart:~$ ot dataset active
Channel: <channel>
Master Key: <masterkey>
PAN ID: <panid>
Done
```

Even it states that Join is successful it never reach state greather than 1, and can't get IP address from OTBR. Active dataset is partially populated.

**Expected behavior**
MTD/SED able to join OT successfully and communicate with other nodes into network.

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain gnuarmemb
 - Zephyr version: 2.4.99
 - Commit 35ec164e36 (non working)
 - Commit b6b613d4db (working)

rlubos, 2020-11-24:
I did some testing but could only reproduce the issue with SED (MED worked just fine). And it isn't related to the commissioning process (which completes correctly) but to the attachment which happens afterwards. There is some issue with CCA handling in the radio driver in SED mode, which can get stuck, I've reported the issue internally for further investigation.

retfie, 2020-11-24:
@rlubos, You're right, after upgrading to MED role, node is able to join and attach to network successful. We could update the issue heading with more appropriate description now or once the root cause is found.

rlubos, 2020-12-02:
@retfie The fix is on its way (https://github.com/zephyrproject-rtos/zephyr/pull/30387) thanks to @czeslawmakarski.

PR #30387 (2020-12-07): west.yml: Update hal_nordic to the latest version
This PR updates the hal_nordic repository to the latest
version.

Fixes #30173.
commit e8f2c341 (2020-12-07): west.yml: Update hal_nordic to the latest version

This PR updates the hal_nordic repository to the latest
version.

Fixes #30173.
closed 2020-12-07 by carlescufi