zephyrproject-rtos/zephyr #30181: OpenPOWERLINK V2 Intergation

Nukersson, 2020-11-22:
**Is your feature request related to a problem? Please describe.**
Currently Zephyr does not have any real time ethernet support.

**Describe the solution you'd like**
Integrate [openPOWERLINK V2](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2), which is under BSD and GPLv2 licenses.

```
Ethernet Powerlink expands Ethernet with a mixed polling and timeslicing mechanism. This provides:

    Guaranteed transfer of time-critical data in very short isochronic cycles with configurable response time
    Time-synchronisation of all nodes in the network with very high precision of sub-microseconds
    Transmission of less timecritical data in a reserved asynchronous channel

Modern implementations reach cycle-times of under 200 µs and a time-precision (jitter) of less than 1 µs. 
```
[***source**](https://en.wikipedia.org/wiki/Ethernet_Powerlink)

**Describe alternatives you've considered**
No other alternatives in this field as free solution.

Nukersson, 2020-11-22:
@jukkar what do you think?

jukkar, 2020-11-23:
I am ok with this. You need to port the openpowerlink code as a module.

Cherish-Gww, 2023-05-25:
@KozhinovAlexander 
so cool !
How is the progress now?

closed 2024-01-21 by KozhinovAlexander