zephyrproject-rtos/zephyr #30184: RFC: Extend Secure Sockets with engine support

drandreas, 2020-11-22:
## Introduction

Zephyr implements the concept of Secure Sockets. Those ease the integration of TLS and DTLS into arbitrary applications. However, the secure sockets hide the underlaying crypto library. Moreover, the current code lacks support for opaque private keys and engine callbacks.

The secure socket API should be extended with an API that allows replacing any of the asymmetric functions on a per context basis.

### Problem description

mbedTLS has support for alternative crypto implementations and limited (RSA: yes, ECP: no) support of opaque keys. However, Zephyr’s code](subsys/net/lib/sockets/sockets_tls.c) that setups the pk context is very rigid and cannot cope with situations where `TLS_CREDENTIAL_PRIVATE_KEY` is not set.

### Proposed change

Add support for adding opaque private keys with `TLS_CREDENTIAL_PRIVATE_KEY_BLOB` in addition to `TLS_CREDENTIAL_PRIVATE_KEY` and allow for definition of alternative cryptographic functions.

## Detailed RFC

### Proposed change (Detailed)

The Secure Socket subsystem runs to the best of my knowledge in kernel space. The proposed change therefore needs to take privilege separation into account. User space should not be able to set functions that are executed in kernel mode.

mbedTLS features [mbedtls_pk_setup_rsa_alt](https://tls.mbed.org/api/pk_8h.html#ab719d0b89ef0a49fef2bc8ba20e4726b) which allows the definition of `mbedtls_pk_rsa_alt_decrypt_func`, `mbedtls_pk_rsa_alt_sign_func` and `mbedtls_pk_rsa_alt_key_len_func`. Those could be exposed to the developer to be defines at compile time. Additionally, `tls_set_own_cert` could be extended to detect opaque keys and call `mbedtls_pk_setup_rsa_alt` instead of `mbedtls_pk_setup` with the above alt function pointers. Finaly Zephyr should provide a macro/function that allows for a type safe recovery of the `KEY_BLOB` within the alt-functions from `void *ctx`.

### Dependencies

The purposes change addresses RSA only. The three alt functions described above are not yet available for ecp. We would need to create a PR in the mbedTLS project to upstream `mbedtls_pk_setup_ecp_alt`.

### Concerns and Unresolved Questions

Depending on the application an engine might need to communicate with a UI to request a password. The current RFC only addresses one way data transfers.

## Alternatives

I've created a [Proof of Concept](https://github.com/drandreas/tpm2-tss-zephyr) for using a Trusted Platform Modules (TPM2) as secure storage of private keys. The PoC defines `MBEDTLS_ECDSA_SIGN_ALT` to catch all sign-requests and forward them to the TPM2. However this approach has no context. Therefore the `KEY_BLOB` needs to be passes on a side channel (global variable). Moreover the function lacks context. Therefore this approach only works for a single key per algorithm (1x RSA, 1xECP).

jukkar, 2020-11-23:
This looks more like a security enhancement than network one to me. cc: @d3zd3z @ceolin

d3zd3z, 2020-12-04:
I think this is a good start. Opaque keys will be important when wanting to store keys in a secure environment such as TF-M/PSA.

tpennors, 2022-10-20:
Hello,
Any news on this feature which could really enhance security ? Would be interested!

d3zd3z, 2023-07-31:
At this point, we are waiting for some more progress in how this is done within TF-M. It works when chosen at build time, and there was additional information added to the keys, but it isn't yet possible to have keys that live in different places from how the system was configured.