zephyrproject-rtos/zephyr #30187: usb: stm32: MCU fall in deadlock when calling sleep API during USB transfer

easternblack, 2020-11-23:
**Describe the bug**
 during USB transfer, If main or thread calls sleep API(ex. k_sleep, k_msleep, k_usleep), MCU fall in deadlock.

**To Reproduce**
To simplify the situation, I made simple usb stack and sample program for test.
https://github.com/easternblack/zephyr/commit/d1cf0a6792101b397e91a25d22208244adae8ef3

Before make, you should modify a little in main.c
There's 2 ways to reproduce the matter.
1. Use k_sleep() or k_msleep(), or k_usleep() instead of k_yield() in the main while().
2. Uncomment thread_create() in the main().

then, make and flash.
1. mkdir build; cd build
2. cmake -DBOARD=stm32f746g_disco ..
3. make

**Expected behavior**
If host send BULK_OUT data start with 'a' 'b' 'c', then device response 100 bytes length of  BULK_IN  data.

**Impact**
If there's sleep API in main() or other thread, MCU falls in deadlock.

**Logs and console output**
BULK_OUT data sended by host start with 'a', 'b', 'c'
[image]

BULK_IN data from device after host send correct data.
[image]

With repeating this transfer, and main() or other threads calls sleep API periodically, MCU fall in deadlock and never wake up until hardware reset.

If there's more often USB transfer, the situation also occurs more often too. 
Shorter period of sleep API makes the problem more often.(k_usleep(1) > k_msleep(1)).

**Environment (please complete the following information):**
TEST environment:
 - HOST: Window 7 USB Host Program.
 - Test Board: stm32f746i_disco.
 - based on bc6d16e422be4e317c4f005ebf1a1af72bb5cf42

**Additional context**
My purpose was make mass transfer via USB.    
at first, I thought other functions and works are cause of the problem.  so, I cut off other functions one by one and finally found k_sleep-things makes this situation.
 I made USB stack by reference USB loopback in zephyr origin.

 I also read https://github.com/zephyrproject-rtos/zephyr/issues/28333 and https://github.com/zephyrproject-rtos/zephyr/pull/28408 but did not work on this issue.

jfischer-no, 2020-11-23:
@easternblack Could you please provide an more simple example, for example from the Zephyr tree with minimal changes, where the problem occurs? Sorry, but your example from [above](https://github.com/easternblack/zephyr/commit/d1cf0a6792101b397e91a25d22208244adae8ef3) does not look right. I have marked it as a question until a more clear description follows.

easternblack, 2020-11-24:
@jfischer-no thx for reply. Do you mean you want tweaked sample apps of zephyr main tree?

closed 2020-12-04 by easternblack