zephyrproject-rtos/zephyr #30192: mec15xxevb_assy6853: running tests/subsys/power/power_mgmt_soc failed

maksimmasalski, 2020-11-23:
**Describe the bug**
1. Test `test_pm_multithread` and `test_pm_singlethread` fail, Assertion failed in both cases. 
2. Also at the end of test happens kernel error 
```
E: ***** USAGE FAULT ***** 
E:   Illegal use of the EPSR
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b mec15xxevb_assy6853 tests/subsys/power/power_mgmt_soc
2. west flash
3. Open terminal and See error

**Expected behavior**
Test must pass.

**Impact**
Power Management issue.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.4.0-1755-g0a0404df29ed  ***
Running test suite test_power_mgmt
===================================================================
START - test_pm_dummyinit
I: PM dummy single-thread test started for one cycle
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.502 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.102 seconds
I: PM dummy single-thread completed
 PASS - test_pm_dummyinit
===================================================================
START - test_pm_multithread
I: Thread task A init
I: Thread task B init
I: PM multi-thread test started for cycles: 2
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: Resume
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.502 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.102 seconds
I: Resume
I: PM multi-thread completed
I: PM state[0] entry counter 2

I: PM state[0] exit counter 2

I: PM state[1] entry counter 2

I: PM state[1] exit counter 1

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt_soc/sr)
PM counters entry/exit mismatch
 FAIL - test_pm_multithread
===================================================================
START - test_pm_singlethread
I: PM single-thread test started for cycles: 2
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: PM single-thread completed
I: PM state[0] entry counter 2

I: PM state[0] exit counter 2

I: PM state[1] entry counter 4

I: PM state[1] exit counter 3

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt_soc/sr)
PM counters entry/exit mismatch
 FAIL - test_pm_singlethread
===================================================================
Test suite test_power_mgmt failed.
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000023  r1/a2:  0x001185a8  r2/a3:  0x001185a8
E: r3/a4:  0x00000001 r12/ip:  0x00000000 r14/lr:  0x000e2703
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x0000000a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x1185a8 (main)
E: Halting system

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.4 
 - Commit SHA 0a0404df29ed890cacadc94ecbd6e1e7f79c3d75

s-kelvin, 2020-12-02:
The testcase worked well on zephyr version v2.3.0. The failure might be because of the code change in versions. 
`
*** Booting Zephyr OS build zephyr-v2.3.0-1348-g32d6aa724f5c  ***
Running test suite test_power_mgmt
===================================================================
START - test_pm_dummyinit
I: PM dummy single-thread test started for one cycle
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.502 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.102 seconds
I: PM dummy single-thread completed
PASS - test_pm_dummyinit
===================================================================
START - test_pm_multithread
I: Thread task A init
I: Thread task B init
I: PM multi-thread test started for cycles: 2
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.102 seconds
I: Resume
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: Resume
I: PM multi-thread completed
I: PM state[0] entry counter 2
I: PM state[0] exit counter 2
I: PM state[1] entry counter 2
I: PM state[1] exit counter 2
PASS - test_pm_multithread
===================================================================
START - test_pm_singlethread
I: PM single-thread test started for cycles: 2
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: PM single-thread completed
I: PM state[0] entry counter 2
I: PM state[0] exit counter 2
I: PM state[1] entry counter 2
I: PM state[1] exit counter 2
PASS - test_pm_singlethread
===================================================================
Test suite test_power_mgmt succeeded
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000000  r1/a2:  0x00118598  r2/a3:  0x00118598
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000e246f
E:  xpsr:  0x60000000
E: Faulting instruction address (r15/pc): 0x00000014
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x00118598 (main)
E: Halting system
`

maksimmasalski, 2020-12-02:
> The testcase worked well on zephyr version v2.3.0. The failure might be because of the code change in versions.
>
> ```
> *** Booting Zephyr OS build zephyr-v2.3.0-1348-g32d6aa724f5c  ***
> Running test [...]

Yes, it might be a code change. You have to find the latest stable zephyr commit which worked well, and then investigate what code changes where did right after that causing an error.

s-kelvin, 2020-12-03:
The testcase passed without failures by cherry-pick the testcase on top of the commit of the release of v2.4.0 on Sep.26. Then I used bisect between the cherry-pick commit and my merged commit [#27733](https://github.com/zephyrproject-rtos/zephyr/pull/27733), which gave me this [commit](https://github.com/zephyrproject-rtos/zephyr/commit/f5a7e1a108296d00a09d5a6913f31d7c2dc96a00) as the exact one that introduced the change causing the failure of this testcase.

s-kelvin, 2021-02-01:
This has not been fixed. See the latest log on micro chip below:
```
*** Booting Zephyr OS build v2.5.0-rc1-133-gdfd8e7522d11  ***
Running test suite test_power_mgmt
===================================================================
START - test_pm_dummyinit
I: PM dummy single-thread test started for one cycle
I: About to enter light sleep
PM >
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: PM dummy single-thread completed
PASS - test_pm_dummyinit
===================================================================
START - test_pm_multithread
I: Thread task A init
I: Thread task B init
I: PM multi-thread test started for cycles: 2
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.502 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: Resume
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.102 seconds
I: Resume
I: PM multi-thread completed
I: PM state[0] entry counter 2
I: PM state[0] exit counter 2
I: PM state[1] entry counter 2
I: PM state[1] exit counter 1
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt_soc/src/power_mgmt.c:117: pm_check_counters: (pm_counters[i].entry_cnt not equal to pm_counters[i].exit_cnt)
PM counters entry/exit mismatch
E: ***** USAGE FAULT *****
E:   Illegal load of EXC_RETURN into PC
E: r0/a1:  0x00000000  r1/a2:  0xe000ed00  r2/a3:  0x00118dd0
E: r3/a4:  0x00000000 r12/ip:  0x0000000a r14/lr:  0x000e2abd
E:  xpsr:  0x00119c00
E: Faulting instruction address (r15/pc): 0x00119cb8
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x1185b8 (main)
E: Halting system
```

ceolin, 2021-02-02:
That is not the same error message, I'd used `cc1352r1_launchxl` to fix the first report that was consistently happening there as well. But this one I'm not being able to reproduce. It seems though that the stack is being corrupted, would mind to test in this board with the following option:
`CONFIG_IDLE_STACK_SIZE=2048`

carlescufi, 2021-02-09:
> would mind to test in this board with the following option:
> `CONFIG_IDLE_STACK_SIZE=2048`

@maksimmasalski could you re-run the test with that option to see if you get rid of the fault?

s-kelvin, 2021-02-18:
```
*** Booting Zephyr OS build zephyr-v2.5.0-197-gca5272f239cd  ***
Running test suite test_power_mgmt
===================================================================
START - test_pm_dummyinit
I: PM dummy single-thread test started for one cycle
I: About to enter light sleep
PM >
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: PM dummy single-thread completed
PASS - test_pm_dummyinit
===================================================================
START - test_pm_multithread
I: Thread task A init
I: Thread task B init
I: PM multi-thread test started for cycles: 2
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.000 seconds
*** Booting Zephyr OS build zephyr-v2.5.0-197-gca5272f239cd  ***
Running test suite test_power_mgmt
===================================================================
START - test_pm_dummyinit
I: PM dummy single-thread test started for one cycle
I: About to enter light sleep
PM >
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: PM dummy single-thread completed
PASS - test_pm_dummyinit
===================================================================
START - test_pm_multithread
I: Thread task A init
I: Thread task B init
I: PM multi-thread test started for cycles: 2
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep

...skipping 1 line
I: PM sleep residency 1.503 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: Resume
I: Suspend...
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.502 seconds
I: Resume
I: Suspend...
I: About to enter deep sleep
PM >
I: PM sleep entry latency 0.001 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.102 seconds
I: Resume
I: PM multi-thread completed
I: PM state[0] entry counter 2
I: PM state[0] exit counter 2
I: PM state[1] entry counter 2
I: PM state[1] exit counter 1
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt_soc/src/pow
er_mgmt.c:117: pm_check_counters: (pm_counters[i].entry_cnt not equal to pm_coun
ters[i].exit_cnt)
PM counters entry/exit mismatch
FAIL - test_pm_multithread
===================================================================
START - test_pm_singlethread
I: PM single-thread test started for cycles: 2
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep
PM <
I: PM sleep residency 3.103 seconds
I: About to enter light sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Light Sleep
PM <
I: PM sleep residency 1.503 seconds
I: About to enter deep Sleep
PM >
I: PM sleep entry latency 0.000 seconds
I: Wake from Deep Sleep

...skipping 1 line
I: PM sleep residency 3.103 seconds
I: PM single-thread completed
I: PM state[0] entry counter 2
I: PM state[0] exit counter 2
I: PM state[1] entry counter 4
I: PM state[1] exit counter 2
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt_soc/src/pow
er_mgmt.c:117: pm_check_counters: (pm_counters[i].entry_cnt not equal to pm_coun
ters[i].exit_cnt)
PM counters entry/exit mismatch
FAIL - test_pm_singlethread
===================================================================
Test suite test_power_mgmt failed.
===================================================================
PROJECT EXECUTION FAILED
```

ceolin, 2021-02-22:
that is a different problem, let me look it. This one seems to be a real issue and not only a stack overflow

s-kelvin, 2021-03-01:
This is no longer a failure since 0ba94379381f.

PR #31765 (2021-01-29): tests: power_mgmt_soc: Increase idle stack size
Test suite test_power_mgmt succeeded
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000000  r1/a2:  0x00118598  r2/a3:  0x00118598
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000e246f
E:  xpsr:  0x60000000
E: Faulting instruction address (r15/pc): 0x00000014
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x00118598 (main)
E: Halting system

Fixes: #30192
commit 7b47c18d (2021-01-29): tests: power_mgmt_soc: Increase idle stack size

Test suite test_power_mgmt succeeded
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000000  r1/a2:  0x00118598  r2/a3:  0x00118598
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000e246f
E:  xpsr:  0x60000000
E: Faulting instruction address (r15/pc): 0x00000014
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x00118598 (main)
E: Halting system

Fixes: #30192
closed 2021-03-01 by s-kelvin