zephyrproject-rtos/zephyr #30193: reel_board: running tests/subsys/power/power_mgmt_soc failed

maksimmasalski, 2020-11-23:
**Describe the bug**
1. Test `test_pm_dummyinit` fail, happens `***** MPU FAULT *****`
That can be related to #30192 , but reel_board board and the Microchip board from #30192 issue have different error types and failed tests.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b reel_board tests/subsys/power/power_mgmt_soc
2. west flash
3. Open terminal and See error

**Expected behavior**
Test must pass.

**Impact**
Low pass rate.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.4.0-1755-g0a0404df29ed  ***
Running test suite test_power_mgmt
===================================================================
START - test_pm_dummyinit
I: PM dummy single-thread test started for one cycle
I: About to enter light sleep
PM >
I: PM sleep entry latency E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x200015c8
E: r0/a1:  0x20000ca0  r1/a2:  0x00000000  r2/a3:  0x04010102
E: r3/a4:  0x00002800 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x000001d0
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x20000400 (idle 00)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux Fedora 28
 - Toolchain Zephyr SDK 0.11.4
 - Commit SHA 0a0404d

s-kelvin, 2020-11-25:
According to the platform exclusion in the yaml file, reel board should be skipped for this case. In the latest mainline daily test report, I didn't see this failing on reel board.

nashif, 2020-12-01:
based on the comment above, this should not be run on reel_board. Also not sure how this is related to memory protection. It is just a stack overflow.

closed 2020-12-01 by nashif