zephyrproject-rtos/zephyr #30194: qemu_x86 crashes when printing floating point.

nordic-krch, 2020-11-23:
**Describe the bug**
Printing floating point (`%f`) crashed on qemu_x86. Same sample runs correctly on `qemu_cortex_m3` and `native_posix`.

**To Reproduce**
Modify `hello_world` sample in the following way:
- add `printk("%f", 1.11);` to main.c 
- add `CONFIG_CBPRINTF_FP_SUPPORT=y` to `prj.conf`
run it on `qemu_x86` and see that it fails:
```
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200516_152813-e7ee184eb100-zephyr
)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.4.0-1755-g0a0404df29ed  ***
FAILED: zephyr/CMakeFiles/run 
```

**Expected behavior**
Standard hello world output extended with floating point number should be seen on the console.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - Commit 0a0404df29

nordic-krch, 2020-11-23:
@pabigot can you take a look?

nordic-krch, 2020-11-23:
Ok, i got it. Apparently, you need to enable FPU support (`CONFIG_FPU=y`) to print floats on qemu_x86.

closed 2020-11-23 by nordic-krch

pabigot, 2020-11-23:
> Ok, i got it. Apparently, you need to enable FPU support (`CONFIG_FPU=y`) to print floats on qemu_x86.

You can just enable `CONFIG_CBPRINTF_FP_SUPPORT=y`, which is the default when `CONFIG_FPU=y`.  This is what most samples with float output use.