zephyrproject-rtos/zephyr #30209: TCP2 : How to add MSS option on sending [SYN, ACK] to client?

liang8057, 2020-11-24:
I want to add TPC MSS

jukkar, 2020-11-24:
Currently the TCP stack does not add any tcp options when sending data. We could support `TCP_MAXSEG` socket option that the application could set. Patches are welcome if you want to implement it.

liang8057, 2020-11-25:
Can TCP1 add mss tcp option?

liang8057, 2020-11-25:
I did not find the TCP_MAXSEG socket option, is there an example for reference?

jukkar, 2020-11-26:
> Can TCP1 add mss tcp option?

Yes, there is some option setting in TCP1. Seach `net_tcp_set_syn_opt()` function.

jukkar, 2020-11-26:
> I did not find the TCP_MAXSEG socket option, is there an example for reference?

Yes, see for example `subsys/net/lib/sockets/sockets.c` and ` zsock_setsockopt_ctx()`

carlescufi, 2021-10-06:
Closing since it's out of date (TCP1 is no longer in the tree).

closed 2021-10-06 by carlescufi