zephyrproject-rtos/zephyr #30212: Disk rewrites same flash page multiple times.

te-johan, 2020-11-24:
While testing USB mass storage on nrf52 I noticed that the write performance was very bad. The reason for this is because disk flash uses a sector size of 512 byte but the erase size of flash is 4096 bytes. For each written sector the disk_access_flash driver reads 4096 bytes, updates 512 bytes, erases the flash pages and writes it. This is highly inefficient since writing multiple contiguous sectors is a common operation.

Changing sector size in disk to match the erase flash size is not an option since 512 byte it often needed for compatibility.

I propose to add a cache in disk_access_flash that stores last writes and flushes them when writes to another flash page is issued. USB mass storage class should use the DISK_IOCTL_CTRL_SYNC ioctl call to flush the cache.
I have retrofitted this as a PoC and it seems to work fine, owever it would be good if the whole driver was written with this intention.

I would also recommend to rewrite disk_access_flash so that it instead uses flash_map.h so that the disks can be described in devicetree instead of hardcoded configuration parameters. Doing that would take one step towards unifying the file system backend APIs.

de-nordic, 2021-01-18:
> 
> 
> This is highly inefficient since writing multiple contiguous sectors is a common operation.
> 
 
... and is probably killing the flash 4 times faster.

desowin, 2022-10-21:
> I would also recommend to rewrite disk_access_flash so that it instead uses flash_map.h so that the disks can be described in devicetree instead of hardcoded configuration parameters. Doing that would take one step towards unifying the file system backend APIs.

This is done in #50856

I have also observed the erase inefficiency and I agree that the caching in flashdisk is the way to go. My idea is that there should be a simple timeout (e.g.1 second) after which the page is unconditionally flushed if no writes to other pages are done.

desowin, 2022-10-21:
> I have also observed the erase inefficiency and I agree that the caching in flashdisk is the way to go. My idea is that there should be a simple timeout (e.g.1 second) after which the page is unconditionally flushed if no writes to other pages are done.

Apparently the `DISK_IOCTL_CTRL_SYNC` would be the way to go. However, it would require the Mass Storage class to report the caching information to host so host can issue synchronize cache commands.

desowin, 2022-10-25:
> Apparently the `DISK_IOCTL_CTRL_SYNC` would be the way to go. However, it would require the Mass Storage class to report the caching information to host so host can issue synchronize cache commands.

Not necessarily. The cache, at least initially would be there to avoid the multiple flash page rewrites and therefore the `DISK_IOCTL_CTRL_SYNC` can be unconditionally issued on write transfer end. That way, the host would not need to be aware of the caching and would not need to issue synchronize cache commands.

PR #51748 (2022-11-02): drivers: flashdisk: implement basic write caching
Cache written data to avoid rewriting same flash page multiple times when writing subsequent flash pages. The cache is used for reads to account for reading not yet committed (i.e. dirty) page data. Speeding up reads is not intention of this patch and therefore the read path does not modify cache state.

Fixes: #30212

This simple caching quite significantly improves write performance. On nRF52840 DK the multiblock writes (SCSI Write(10) Request) are around 9.5 times faster. The speedup is more than would be expected due to only performing 1/8  [... cut]
commit a4c59a33 (2022-11-02): drivers: flashdisk: implement basic write caching

Cache written data to avoid rewriting same flash page multiple times
when writing subsequent flash pages. The cache is used for reads to
account for reading not yet committed (i.e. dirty) page data. Speeding
up reads is not intention of this patch and therefore the read path
does not modify cache state.

Fixes: #30212
closed 2022-11-02 by carlescufi