zephyrproject-rtos/zephyr #30227: Improve toolchain capabilities caching / broken toolchain detection

tejlmand, 2020-11-25:
**Is your enhancement proposal related to a problem? Please describe.**
The current testing toolchain of the toolchain and its capabilities offers a significant performance improvement, as it remove the need of testing the toolchain when running CMake in a new build folder.

Unfortunately, this means that if the installation was broken when testing compiler flags, then the cached capabilities will be marked as not supported.

Re-installing the toolchain will not clear the toolchain capabilities database, and user must do this manually, as is discussed in: https://github.com/zephyrproject-rtos/zephyr/issues/22060.

**Describe the solution you'd like**
There should be a way to force a re-run of the toolchain testing, and if possible, detect if the toolchain installation itself might be broken, and then inform the user.

This should be done in a way that does not require the testing of the toolchain everytime CMake is run in a new build folder.
Testing of toolchain capabilities may take up to 40 seconds.

baw-serafin, 2020-11-25:
What is the usecase for new build folder? There is `--prisitne` for clean builds if that is not enough and I have to delete the build folder I would want to have a really clean build not caches somewhere else.

tejlmand, 2020-11-25:
> What is the use case for new build folder? There is `--pristine` for clean builds if that is not enough and I have to delete the build folder I would want to have a really clean build not caches somewhere else.

A `--pristine` build is identical to a new build folder, because the CMakeCache.txt file is removed during pristine cleaning.
CMake has no concept of pristine, only whether it is:
- Existing build folder (contains a CMakeCache.txt)
- New build folder (doesn't contain a CMakeCache.txt)