zephyrproject-rtos/zephyr #30232: CMake 3.19 doesn't work with Zephyr

galak, 2020-11-25:
In cmake 3.19 `cmake_check_compiler_flag` behaves differently than previous versions and breaks Zephyr.

On ARM you might see something like:

```
CMake Error at ../../cmake/extensions.cmake:1561 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  ../../CMakeLists.txt:42 (assert)
```

and

```
Determining if the  option is supported for C language failed with the following output:
/opt/zephyr-sdk-0.12.0-beta-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: /opt/zephyr-sdk-0.12.0-beta-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x60): undefined reference to `_exit'
collect2: error: ld returned 1 exit status
```

galak, 2020-11-25:
This issue is really to track the issue w/upstream CMake and convey status of what's going on.

Here's the CMake issue:

https://gitlab.kitware.com/cmake/cmake/-/issues/21497

galak, 2020-11-25:
Note, its possible you get into a weird state with the Zephyr cmake cache and best to possibly remove it:

`rm -fr ~/.cache/zephyr`

microbuilder, 2020-11-25:
setup workaround for Ubuntu, just as an example:

```bash
$ wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | sudo apt-key add -
$ sudo apt-add-repository 'deb https://apt.kitware.com/ubuntu/ bionic main'
$ sudo apt update
$ sudo apt install cmake-data=3.18.4-0kitware1 cmake=3.18.4-0kitware1
```

manchoz, 2020-11-25:
Quick-and-dirty workaround for MacOS + `brew`:

Ignore all the `brew` complaints. YMMV.

```sh
$ export FIX_CMAKE_VERSION=3.19.0
$ brew tap-new $USER/homebrew-repo
$ brew extract --version $FIX_CMAKE_VERSION cmake $USER/homebrew-repo
$ brew unlink cmake
$ brew install cmake@$FIX_CMAKE_VERSION
```

aunsbjerg, 2020-11-26:
Cmake is available as a pip package, which is already mentioned in the getting started docs. Would it make sense to add cmake to the python requirements instead of having to manually install it? This would also make it easier to pin a specific cmake version as the officially supported version.

carlescufi, 2020-11-26:
> Cmake is available as a pip package, which is already mentioned in the getting started docs. Would it make sense to add cmake to the python requirements instead of having to manually install it? This would also make it easier to pin a specific cmake version as the officially supported version.

We did consider that, but it was ultimately decided that this was not a good idea because many people choose to use the CMake that comes with their distribution instead, and by putting it in the requirements we'd effectively be asking users to have 2 copies of CMake.

tejlmand, 2020-11-26:
Workaround posted: https://github.com/zephyrproject-rtos/zephyr/pull/30272

PR #30272 (2020-12-01): cmake: manual handling of CMakeCheckCompilerFlag.cmake inclusion
Fixes: #30232

This is a workaround for #30232.

During Zephyr CMake invocation a plain C compiler is used for DTS.
This results in the internal `CheckCompilerFlag.cmake` being included
by CMake

Later, when the full toolchain is configured, then
`CMakeCheckCompilerFlag.cmake` is included.
This overloads the `cmake_check_compiler_flag()` function, thus
causing #30232.

By manualy loading `CMakeCheckCompilerFlag.cmake` then
`CheckCompilerFlag.cmake` will overload the functions (and thus win the
battle), and because `include_guard( [... cut]
commit 6ca2bf25 (2020-12-01): cmake: manual handling of CMakeCheckCompilerFlag.cmake inclusion

Fixes: #30232

This is a workaround for #30232.

During Zephyr CMake invocation a plain C compiler is used for DTS.
This results in the internal `CheckCompilerFlag.cmake` being included
by CMake

Later, when the full toolchain is configured, then
`CMakeCheckCompilerFlag.cmake` is included.
This overloads the `cmake_check_compiler_flag()` function, thus
causing #30232.

By manualy loading `CMakeCheckCompilerFlag.cmake` then
`CheckCompilerFlag.cmake` will overload the functions (and thus win the
battle), and because `include_guard [... cut]
closed 2020-12-01 by galak