zephyrproject-rtos/zephyr #30236: Main thread sometimes looping forever before user application is reached when using UDP and IPv6 on Nucleo F767ZI

rtzu6, 2020-11-25:
**Describe the bug**
When booting a Nucleo f767zi with the network stack enabled (UDP and IPv6 only), sometimes the main thread blocks. In most cases, the example works with the TCPtest.py offered by net-tools (modified the socket to work with UDP, otherwise untouched).
In rare cases, it does not reach the application but is stuck in an endless loop in z_clock_announce (timeout.c). 
Attaching gdb and catching the rare event shows the timeout struct returned by `first()` has dticks set to 0 *before* it is zeroed later. A struct with dticks of 0 seems to be returned every time first is called in this situation. Consequently, `announce_remaining` always stays the same and the loop never ends. The user application is never reached.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Linux. I was unable to reproduce this problem on a Mac.
2. build the echo_server example (`west build -p auto -b nucleo_f767zi zephyr/samples/net/sockets/echo_server`) for IPv6 and UDP (see KConfig below)
3. flash it to the board (`west flash`)
4. reset until the behaviour is exposed

**Expected behavior**
The setup should always complete normally and pass control to user code.

**Logs and console output**

timeout struct t in z_clock_announce in error case, breakpoint in line `curr_tick += dt;` Note the dticks of 0. This is before it is zeroed later.

` {node = {{head = 0x0, next = 0x0}, {tail = 0x0, prev = 0x0}}, fn = 0x800eb2b <work_timeout>, dticks = 0} `

KConfig:
```
# Generic networking options
CONFIG_NETWORKING=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=n
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=n
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_POSIX_MAX_FDS=6
CONFIG_NET_CONNECTION_MANAGER=y

# Kernel options
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y

# Logging
CONFIG_NET_LOG=y
CONFIG_LOG=y
CONFIG_NET_STATISTICS=y
CONFIG_PRINTK=y

# Network buffers
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=64
CONFIG_NET_BUF_TX_COUNT=64
CONFIG_NET_CONTEXT_NET_PKT_POOL=y

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=10

# Network shell
#CONFIG_NET_SHELL=y
#CONFIG_SHELL=y

# Network application options and configuration
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_NEED_IPV4=n
#CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
#CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"

# Number of socket descriptors might need adjusting
# if there are more than 1 handlers defined.
CONFIG_POSIX_MAX_FDS=12

# How many client can connect to echo-server simultaneously
CONFIG_NET_SAMPLE_NUM_HANDLERS=1
```

**Environment:**
 - OS: Linux
 - Toolchain: (arm-none-eabi-gcc 10.2.0)
 - Zephyr version: 2.4.99 (latest master)

**Additional context**
I was unable to reproduce this problem on a Mac.
I tried two different boards, with the same results. Both boards work with a Mac.

erwango, 2020-11-30:
@rtzu6 Thanks for reporting this issue

> When booting a Nucleo f767zi with the network stack enabled (UDP and IPv6 only), sometimes the main thread blocks [...]. Attaching gdb and catching the rare event shows the timeout struct returned by first() has dticks set to 0 before it is zeroed later.

Did you face the same issue w/o the network stack enabled ?

rtzu6, 2020-11-30:
No, I cannot reproduce this issue when changing `CONFIG_NETWORKING=n` (I assume this is what you mean).

avigreen1978, 2021-01-04:
I experienced a similar issue. In my case, when I I turned off NET_CONFIG options and just set my ip manually, it stopped happening.

rtzu6, 2021-01-05:
Interesting. Do you have a small example that I can test/steps to produce one? I can get rid of most of these, but for example removing CONFIG_NET_CONFIG_NEED_IPV6 is not possible.

erwango, 2021-01-07:
@avigreen1978 are using the same board ?

erwango, 2021-01-12:
> I experienced a similar issue. In my case, when I I turned off NET_CONFIG options and just set my ip manually, it stopped happening.

@jukkar does this ring any bell?

jukkar, 2021-01-12:
> does this ring any bell?

I have not seen anything like this. I have this board so I just tried it with the prj.conf file that is found in the description. I see no issues, no hanging etc. I can use `echo-client` from `net-tools` project and the IPv6 connectivity works just fine.

rtzu6, 2021-01-12:
Out of curiosity: what operating system are you on? I was unable to reproduce this issue on an apple system (I tried two different iMacs).

jukkar, 2021-01-12:
I am using Fedora 32

jukkar, 2021-01-12:
@rtzu6 was the issue happening always or just sometimes? I tried to replicate this only couple of times.
I am using zephyr sdk 0.11.3

rtzu6, 2021-01-12:
The issue does not happen very often, but also not consistently after a certain amount. Sometimes it happens after a few reboots, sometimes it takes many.

carlescufi, 2021-01-26:
> timeout struct t in z_clock_announce in error case, breakpoint in line `curr_tick += dt;` Note the dticks of 0. This is before it is zeroed later.
> 
> `{node = {{head = 0x0, next = 0x0}, {tail = 0x0, prev = 0x0}}, fn = 0x800eb2b <work_timeout>, dticks = 0}`

@andyross does this ring a bell?

andyross, 2021-01-27:
Something is definitely weird.  The entry in question (if I'm understanding the description) is the result of `sys_dlist_peek_head()` on the timeout list.  But the null list pointers say that the node is **not** part of a list and has already been removed.  The dlist API shouldn't ever produce that situation.  My guess would be that something just came by and zeroed the timeout struct, but the fact that the callback pointer is still present argues against that.  It's a work queue timeout, we know that much at least.

As a completely wild guess: we have a race somewhere in the work queue code that allows a timer interrupt to happen while it's mucking with the work item struct and its embedded timeout?

erwango, 2021-02-05:
@andyross Would you have any way to instrument the code to track a potential race ?

erwango, 2021-05-11:
@rtzu6 Thanks for feedback. I'm setting it as "Waiting for reponse".

closed 2021-08-02 by github-actions[bot]