zephyrproject-rtos/zephyr #30240: Bluetooth: Mesh: PTS Test failed in friend node

LingaoM, 2020-11-26:
**There are two issues related to friend.**

- 1: MESH/NODE/FRND/FN/BV-14-C MESH/NODE/FRND/FN/BV-15-C

Both the above two tests require IUT to cache the data(netkey_update) sent by LT2 to LPN as a friend node. However, according to the log, LPN does not receive valid data, resulting in LT2(waiting for netkey_status) timeout exit.

- 2: Latest PTS 7.6.2 Add new testcase MESH/NODE/FRND/FN/BV-13-C

From the simple description of PTS, it can be inferred that the test friend relay the LPN message. However, this item also failed. The reason for the failure was that Low Tester did not receive the specified data. I read the log from the IUT and IUT has already relayed.

LingaoM, 2020-11-26:
https://github.com/zephyrproject-rtos/zephyr/blob/41b7e5da7b453ce6ec648e29825f3e71048b498a/subsys/bluetooth/mesh/subnet.c#L616-L620

@trond-snekvik 
Here is a bug that I have found. I plan to submit PR together to solve these problems, but it is necessary to tell you. It will lead to the failure to establish a friendship relationship at all. The reason is that the friend node will set established to 1 only after receiving the poll from LPN, but the message has passed through friend_cred encryption.

LingaoM, 2020-11-26:
https://github.com/zephyrproject-rtos/zephyr/blob/41b7e5da7b453ce6ec648e29825f3e71048b498a/subsys/bluetooth/mesh/transport.c#L1446-L1447

In this way, the sequence number of each segment is changed, and the segment packets of the same sequence number are saved to the friend cache, resulting in LPN **replay attack**.

@trond-snekvik first issue has been solved, i'll submit pr for this. But until now, I haven't solved the **second** problem.

jhedberg, 2020-11-30:
Did #30260 fix this?

LingaoM, 2020-11-30:
> Did #30260 fix this?

Yes, But it is only a part, so it cannot be closed.

carlescufi, 2020-12-03:
> > Did #30260 fix this?
> 
> Yes, But it is only a part, so it cannot be closed.

Will you post a PR to fix the second part?

LingaoM, 2020-12-07:
> > > Did #30260 fix this?
> > 
> > 
> > Yes, But it is only a part, so it cannot be closed.
> 
> Will you post a PR to fix the second part?

Of course, I'm giving feedback to the Bluetooth support team and I'm waiting for an answer.

LingaoM, 2020-12-09:
> * 2: Latest PTS 7.6.2 Add new testcase MESH/NODE/FRND/FN/BV-23-C
> 
> From the simple description of PTS, it can be inferred that the test friend relay the LPN message(refer #29580). However, this item also failed. The reason for the failure was that Low Tester did not receive the specified data. I read the log from the IUT and IUT has already relayed(?_?).

DW
Dexter Wong
Additional comments•2020-12-01 09:51:12
https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=481160

https://www.bluetooth.org/tse/errata_view.cfm?errata_id=12773

@michal-narajowski  Do you have time to help solve this problem? Above is the detailed documentation I provided about this test item.

LingaoM, 2020-12-31:
https://bluetooth.service-now.com/ess/x_bsig_case_manage_case.do?sysparm_query=number=CASE0069829 PTS8.0.0 has been solved this problem.

closed 2020-12-31 by LingaoM