zephyrproject-rtos/zephyr #30243: Bluetooth: controller: IRK resolution in extended scanning breaks auxiliary PDU reception

cvinayak, 2020-11-26:
**Describe the bug**
`radio_irk_resolve` function takes ~300 us which prevents reception of auxiliary PDUs that are as close as 300us of T_MAFS from next auxiliary PDU.

Details be filled as we go about fixing.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

cvinayak, 2021-01-06:
@nashif This issue is against experimental and work in progress support for Advertising Extensions feature in Bluetooth Controller. We should move this out of v2.5 release bug list.

carlescufi, 2021-01-14:
Downgrading to low since this feature is still WIP in the controller.

carlescufi, 2021-06-17:
Since complete Advertising Extensions support is a main focus for 2.7 LTS, and this is a blocker for it, raising the priority to medium

commit 4661ae3c (2021-08-04): Bluetooth: Controller: Fix Extended Scanning Address Resolution

Fix the received Extended Advertising PDU's address
resolution latency due to missing reset of pending IRQ in
the NVIC which caused the wait loop from not exit correctly.

Fixes #30243.
closed 2021-08-04 by carlescufi