zephyrproject-rtos/zephyr #30244: Bluetooth: controller: Extended scan window time reservation prevents auxiliary channel reception

cvinayak, 2020-11-26:
**Describe the bug**
Time space is reserved for scan window, this will cancel single shot scheduling being used for auxiliary PDU.

- Add implementation for abort scan window
- Add new ticker interface to yield currently occupied timespace

More info be filled as we work on this issue...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

cvinayak, 2020-11-27:
@asmk-ot @mtpr-ot @wopu-ot @kruithofa 

Relates to the below design requirement (serial no. 2):
[image]

Discussion welcome...

cvinayak, 2021-01-06:
@nashif This issue is against experimental and work in progress support for Advertising Extensions feature in Bluetooth Controller. We should move this out of v2.5 release bug list.

carlescufi, 2021-01-14:
Downgrading to low since this feature is still WIP in the controller.

PR #35389 (2021-06-14): Bluetooth: controller: Add ticker_yield_abs interface
Add ticker_yield_abs interface to reduce ticks_slot_previous
value when radio events yield/stop earlier than their time
reservations.

Use ticker_yield_abs interface to make scan window on
primary channel to yield when trying to scan auxiliary PDUs.

Fixes #30244.
commit 97598c4d (2021-06-14): Bluetooth: controller: Yield scan window when scanning Aux PDUs

Use ticker_yield_abs interface to make scan window on
primary channel to yield when trying to scan auxiliary PDUs.

Fixes #30244.
closed 2021-06-14 by carlescufi