zephyrproject-rtos/zephyr #30245: Bluetooth: controller: event scheduling pipeling preemption by short schedule

cvinayak, 2020-11-26:
**Describe the bug**
A request to schedule an event shorter than preemption timeout could be delayed upto the preemption timeout making the event to be delayed beyond its hard deadline.

[image]

1 - Prepare callback of continuous scan window
2 - Preemption abort after 1.5 ms since 1
3 - New schedule, short offset, lower than 300 us, which is lesser than the 1.5 ms preemption timeout
4 - New scheduled event executed beyond the preemption timeout, instead of being scheduled at its hard deadline.

Further details to be filled as works starts towards fixing it.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

carlescufi, 2021-01-14:
Downgrading to low since this feature is still WIP in the controller.

cvinayak, 2021-06-29:
Increased to high priority as issue could be the cause of the #36381

PR #37440 (2021-08-12): Bluetooth: Controller: Fix to handle relative short preempt timeout
When a new radio event is scheduled with a relatively short
preempt timeout while there is already a started preempt
ticker, then stop the ticker, abort the previous event that
requested the preemption, and request to start ticker with
the new relatively short preempt timeout.

Fixes #30245.
commit 27b8beaa (2021-08-12): Bluetooth: Controller: Fix to handle relative short preempt timeout

When a new radio event is scheduled with a relatively short
preempt timeout while there is already a started preempt
ticker, then stop the ticker, abort the previous event that
requested the preemption, and request to start ticker with
the new relatively short preempt timeout.

Fixes #30245.
closed 2021-08-12 by carlescufi