zephyrproject-rtos/zephyr #30257: test: kernel: Test kernel.common.stack_protection_arm_fpu_sharing.fatal fails on nrf52 platforms

PerMac, 2020-11-26:
**Describe the bug**
The test kernel.common.kernel.common.stack_protection_arm_fpu_sharing.fatal fails on nrf52 platforms.

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr dir 
2. have nrf52 platform connected (e.g. nrf52dk_nrf52832)
3. run `./scripts/sanitycheck --device-testing -T tests/kernel/fatal/exception/ -p nrf52dk_nrf52832 --device-serial /dev/ttyACM0 -v -v`
4. See error

**Expected behavior**
test passes

**Impact**
Not clear

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.4.0-1784-g2ce570b03f97  ***
DEBUG   - DEVICE: Running test suite E: ***** MPU FAULT *****
DEBUG   - DEVICE: E:   Stacking error (context area might be not valid)
DEBUG   - DEVICE: E:   Data Access Violation
DEBUG   - DEVICE: E:   MMFAR Address: 0x20002c08
DEBUG   - DEVICE: E: r0/a1:  0x05fefef9  r1/a2:  0x020b8402  r2/a3:  0x3ae7e3a5
DEBUG   - DEVICE: E: r3/a4:  0x401c88f6 r12/ip:  0x3e3ef5eb r14/lr:  0x1e12a498
DEBUG   - DEVICE: E:  xpsr:  0x12423c00
DEBUG   - DEVICE: E: s[ 0]:  0x00000000  s[ 1]:  0xffffffff  s[ 2]:  0x00000000  s[ 3]:  0x00000000
DEBUG   - DEVICE: E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
DEBUG   - DEVICE: E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
DEBUG   - DEVICE: E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
DEBUG   - DEVICE: E: fpscr:  0x00000013
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0xec7237e9
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x20000358 (main)
DEBUG   - DEVICE: Caught system error -- reason 2
DEBUG   - DEVICE: Was not expecting a crash
```

**Environment (please complete the following information):**
 - OS:  Ubuntu 18.04
 - Toolchain Zephyr SDK
 - Commit SHA or Version used zephyr-v2.4.0-1784-g2ce570b03f97

s-kelvin, 2020-11-30:
This testcase fails on reel board as well.

ioannisg, 2020-12-07:
That's a stacking error, might be related to things that we fixed in other tests (@pabigot FYI). Otherwise, we can fix this by enlarging the stack size. Will take  a look.

pabigot, 2020-12-07:
When reproduced using `prj_arm_fpu_sharing.conf ` this correlates with the merge of #29876 so yes a stack size needs to be increased.  Prior to this printk formatting would have used less space as it wouldn't have supported float formatting, which requires a larger stack buffer.

If `CONFIG_CBPRINTF_NANO=y` is added to that config that particular test passes.  That's #30494.

    west build -b pca10056 -- -DOVERLAY_CONFIG=sentinel.conf

When reproduced using `sentinel.conf` this bisects to 79e63b421607ad078a82ab0b3483b9fcc5ac2575  @nordic-krch

ioannisg, 2021-01-25:
I think this has been already fixed, @PerMac , I cannot reproduce this on latest master. Can you also check?

PerMac, 2021-01-25:
I don't see these errors anymore. Closing the issue.

closed 2021-01-25 by PerMac