zephyrproject-rtos/zephyr #30275: up_squared:tests/kernel/common failed (tests freeze and then timeout)

maksimmasalski, 2020-11-27:
**Describe the bug**
When running test suite "kernel.common" on the up_squared board, tests stop running (freeze) and cause timeout error in sanitycheck.

1. All tests in the category `kernel.common.kernel.common.tls` freeze at the next point:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1822-gc08ef381c819  (delayed boot 500ms) ***
Running test suite common
===================================================================
START - test_bootdelay
```

2. All tests in the category `kernel.common.kernel.common.misra` freeze at the next point:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1822-gc08ef381c819  (delayed boot 500ms) ***
Running test suite common
===================================================================
START - test_bootdelay
PASS - test_bootdelay
===================================================================
START - test_irq_offload
PASS - test_irq_offload
===================================================================
START - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap
===================================================================
START - test_byteorder_mem_swap
```

3. All tests in the category `kernel.common.kernel.common.nano32` freeze at the next point:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1822-gc08ef381c819  (delayed boot 500ms) ***
Running test suite common
===================================================================
START - test_bootdelay
PASS - test_bootdelay
===================================================================
START - test_irq_offload
PASS - test_irq_offload
===================================================================
START - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap
===================================================================
START - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap
===================================================================
START - test_sys_get_be64
PASS - test_sys_get_be64
===================================================================
START - test_sys_put_be64
PASS - test_sys_put_be64
===================================================================
START - test_sys_get_be48
PASS - test_sys_get_be48
===================================================================
START - test_sys_put_be48
PASS - test_sys_put_be48
===================================================================
START - test_sys_get_be32
PASS - test_sys_get_be32
===================================================================
START - test_sys_put_be32
PASS - test_sys_put_be32
===================================================================
START - test_sys_get_be24
PASS - test_sys_get_be24
===================================================================
START - test_sys_put_be24
PASS - test_sys_put_be24
===================================================================
START - test_sys_get_be16
PASS - test_sys_get_be16
===================================================================
START - test_sys_put_be16
PASS - test_sys_put_be16
===================================================================
START - test_sys_get_le16
PASS - test_sys_get_le16
===================================================================
START - test_sys_put_le16
```

4. All tests in the category `kernel.common.kernel.common.nano64` freeze at the next point:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1822-gc08ef381c819  (delayed boot 500ms) ***
Running test suite common
===================================================================
START - test_bootdelay
PASS - test_bootdelay
===================================================================
START - test_irq_offload
PASS - test_irq_offload
===================================================================
START - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap
===================================================================
START - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap
===================================================================
START - test_sys_get_be64
PASS - test_sys_get_be64
===================================================================
START - test_sys_put_be64
PASS - test_sys_put_be64
===================================================================
START - test_sys_get_be48
PASS - test_sys_get_be48
===================================================================
START - test_sys_put_be48
PASS - test_sys_put_be48
===================================================================
START - test_sys_get_be32
PASS - test_sys_get_be32
===================================================================
START - test_sys_put_be32
PASS - test_sys_put_be32
===================================================================
START - test_sys_get_be24
PASS - test_sys_get_be24
===================================================================
START - test_sys_put_be24
PASS - test_sys_put_be24
===================================================================
START - test_sys_get_be16
PASS - test_sys_get_be16
===================================================================
START - test_sys_put_be16
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/common
2. Flash board
4. See the error

**Expected behavior**
Tests must pass without timeout errors or freezing

**Impact**
Low pass rate on the up_squared

**Environment (please complete the following information):**
 - OS: Fedora 31
 - Toolchain Zephyr SDK zephyr-sdk-0.11.4
 - Commit SHA c08ef381c819

**Additional context**
Probably the same behaviour can be found in that issue #29737

jenmwms, 2020-11-30:
Was this observed using sanitycheck only? I do not observe #1 `kernel.common.kernel.common.tls`, can try the others. The test is successful on Up2 using `west build -p -b up_squared zephyr/tests/kernel/common/ -DCONFIG_THREAD_LOCAL_STORAGE=y`.

maksimmasalski, 2020-12-01:
> Was this observed using sanitycheck only? I do not observe #1 `kernel.common.kernel.common.tls`, can try the others. The test is successful on Up2 using `west build -p -b up_squared zephyr/tests/kernel/common/ -DCONFIG_THREAD_LOCAL_STORAGE=y`.

Was observed using sanitycheck only.
I can prove that with the west it works correctly, probably that is a sanitycheck bug.

jenmwms, 2020-12-01:
@maksimmasalski  we need to do further root-cause analysis to resolve this issue. Although we can achieve deterministic, reproduceable, correctly passing results using west/flash and sanitycheck by adding workaround of a longer timeout in the tests - it is not advised. We need to find out why the delay is now needed instead of default. It is not a sanitycheck issue. Please see discussion in https://github.com/zephyrproject-rtos/zephyr/pull/30374#pullrequestreview-542380473.

maksimmasalski, 2020-12-02:
> @maksimmasalski we need to do further root-cause analysis to resolve this issue. Although we can achieve deterministic, reproduceable, correctly passing results using west/flash and sanitycheck by adding workaround of a longer timeout in the tests - it is not advised. We need to find out why the delay is now needed instead of default. It is not a sanitycheck issue. Please see discussion in [#30374 (review)](https://github.com/zephyrproject-rtos/zephyr/pull/30374#pullrequestreview-542380473).

Hello, today I observed freeze during 3min35sec in 
```
===================================================================
START - test_sys_put_le48
```

Your patch won't work for that case.

nashif, 2020-12-02:
@dcpleung  FYI

jenmwms, 2020-12-02:
Still looking at this, but found that the test suite starts experiencing delays at this commit. I used git bisect and up2 to narrow it down:
```
93cccf27b3fd43573a409603f923a26f17d9f1ad is the first bad commit
commit 93cccf27b3fd43573a409603f923a26f17d9f1ad
Author: Dominik Ermel <dominik.ermel@nordicsemi.no>
Date:   Wed Oct 21 07:20:08 2020 +0000

    west.yml: Snapshot update of mcumgr from upstream
    
    This commit updates mcumgr with latest snapshot from the upstream.
    
    Signed-off-by: Dominik Ermel <dominik.ermel@nordicsemi.no>

 west.yml | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
```

nashif, 2020-12-02:
>  I used git bisect and up2 to narrow it down:

this cant be it, complete unrelated.

jenmwms, 2020-12-02:
> this cant be it, complete unrelated.

Thank you, Anas. I'm revisiting this. It seems inconsistent so it's harder to chase down.

dcpleung, 2020-12-03:
This seems to only happen when booting through EFI. Booting through GRUB works fine.

jenmwms, 2020-12-10:
Closing this, it is now part of cluster in https://github.com/zephyrproject-rtos/zephyr/issues/30573

closed 2020-12-10 by jenmwms