zephyrproject-rtos/zephyr #30298: regression/change in master: formatting floats and doubles

coderkalyan, 2020-11-27:
After updating to latest zephyr master, I noticed that I can't format float values with libc printf() or the logging API, while I used to be able to. I took a quick in the commit history, and there are some commits recently (in the last 2-3 weeks) regarding formatting in the libc implementation, but I could figure out what was the culprit.

Is there something I should be changing at build time, or a replacement format function, to achieve float formatting?

pabigot, 2020-11-30:
Add `CONFIG_CBPRINTF_FP_SUPPORT=y` to your `prj.conf`.  This should be the default if you have `CONFIG_FPU=y`, but if you're doing software floating point it wouldn't be.

pabigot, 2020-11-30:
For more details on the new formatter: https://docs.zephyrproject.org/latest/reference/misc/formatted_output.html

coderkalyan, 2020-12-01:
This worked for me, thank you very much @pabigot

closed 2020-12-01 by coderkalyan