zephyrproject-rtos/zephyr #30304: NRF52832 consumption too high 220uA

tcpipchip, 2020-11-30:
Dear

I am using the NRF52832, using the following prj

##########
# for GPIO
##########
CONFIG_GPIO=y

##########
# for BLE
##########
CONFIG_BT=y

CONFIG_SENSOR=y

##########
# for BME680
##########
CONFIG_BME680=y

##########
# for ccs811
##########
CONFIG_CCS811=y

##########
# for bme280
##########
CONFIG_BME280=y

##########
# for i2c
##########
CONFIG_I2C=y
CONFIG_I2C_0=y

##########
# for console
##########
CONFIG_STDOUT_CONSOLE=n
CONFIG_PRINTK=n
CONFIG_SERIAL=n

##########
# Battery
##########
CONFIG_ADC=y
CONFIG_ADC_ASYNC=y
CONFIG_ADC_NRFX_SAADC=y
#CONFIG_ADC_0=y

##########
#LOW POWER
##########
CONFIG_OPENOCD_SUPPORT=n
CONFIG_ASSERT=y
CONFIG_SYS_POWER_MANAGEMENT=y
CONFIG_SYS_PM_STATE_LOCK=y

CONFIG_LOG=n

I would like to know how to disable the UART.

Maybe UART is my CONSUMPTION problem!

Thanks!

pabigot, 2020-11-30:
What board are you using?  How are you measuring the power?

It could be UART.  Look for `CONFIG_UART_` in `build/zephyr/.config` and make sure it's all off (what you specify in `prj.conf` may not be what you're getting).

`samples/boards/nrf/system_off` shows how to get the UART device for the console and set it into a low-power state, but that assumes that the UART is available.  It sounds like you're trying to exclude it.  That *should* result in the lowest possible power state for on-board peripherals, but it wouldn't for other devices like I2C sensors which would have to be put into a low-power state explicitly even if they're not being used.

The CCS811 is a pretty high-power device, so if it's powered up that could be a factor.

tcpipchip, 2020-11-30:
Good morning @pabigot (Peter)

I am using the EBYTE E73. Using Amperimeter and put it the cable  in serie with E73 power (VCC), mayBE is this my fault)

This is my .config

#
# Modules
#

#
# Optional modules. Make sure they're installed, via the project manifest.
#
# CONFIG_CANOPENNODE is not set
# CONFIG_CIVETWEB is not set
CONFIG_HAS_CMSIS_CORE=y
CONFIG_HAS_CMSIS_CORE_M=y
# CONFIG_LIBMETAL is not set
# CONFIG_HAS_SEMTECH_LORAMAC is not set
# CONFIG_HAS_SEMTECH_RADIO_DRIVERS is not set
# CONFIG_MBEDTLS is not set
# CONFIG_HAS_MEC_HAL is not set
CONFIG_HAS_NRFX=y

#
# nrfx drivers
#
CONFIG_NRFX_CLOCK=y
CONFIG_NRFX_CLOCK_LFXO_TWO_STAGE_ENABLED=y
# CONFIG_NRFX_COMP is not set
# CONFIG_NRFX_EGU is not set
# CONFIG_NRFX_EGU0 is not set
# CONFIG_NRFX_EGU1 is not set
# CONFIG_NRFX_EGU2 is not set
# CONFIG_NRFX_EGU3 is not set
# CONFIG_NRFX_EGU4 is not set
# CONFIG_NRFX_EGU5 is not set
# CONFIG_NRFX_GPIOTE is not set
# CONFIG_NRFX_I2S is not set
# CONFIG_NRFX_LPCOMP is not set
# CONFIG_NRFX_NFCT is not set
# CONFIG_NRFX_NVMC is not set
# CONFIG_NRFX_PDM is not set
# CONFIG_NRFX_POWER is not set
# CONFIG_NRFX_PPI is not set
# CONFIG_NRFX_PWM is not set
# CONFIG_NRFX_PWM0 is not set
# CONFIG_NRFX_PWM1 is not set
# CONFIG_NRFX_PWM2 is not set
# CONFIG_NRFX_QDEC is not set
# CONFIG_NRFX_RNG is not set
# CONFIG_NRFX_RTC is not set
# CONFIG_NRFX_RTC0 is not set
# CONFIG_NRFX_RTC1 is not set
# CONFIG_NRFX_RTC2 is not set
# CONFIG_NRFX_SAADC is not set
# CONFIG_NRFX_SPI is not set
# CONFIG_NRFX_SPI0 is not set
# CONFIG_NRFX_SPI1 is not set
# CONFIG_NRFX_SPI2 is not set
# CONFIG_NRFX_SPIM is not set
# CONFIG_NRFX_SPIM0 is not set
# CONFIG_NRFX_SPIM1 is not set
# CONFIG_NRFX_SPIM2 is not set
# CONFIG_NRFX_SPIS is not set
# CONFIG_NRFX_SPIS0 is not set
# CONFIG_NRFX_SPIS1 is not set
# CONFIG_NRFX_SPIS2 is not set
# CONFIG_NRFX_SYSTICK is not set
# CONFIG_NRFX_TEMP is not set
[... 1042 lines cut ...]
# CONFIG_BUILD_OUTPUT_S19 is not set
# CONFIG_BUILD_OUTPUT_STRIPPED is not set
# CONFIG_APPLICATION_DEFINED_SYSCALL is not set
# CONFIG_MAKEFILE_EXPORTS is not set
# CONFIG_LEGACY_DEVICETREE_MACROS is not set
# CONFIG_DEPRECATED_ZEPHYR_INT_TYPES is not set
# end of Build Options
# end of Build and Link Features

#
# Boot Options
#
# CONFIG_IS_BOOTLOADER is not set
# CONFIG_BOOTLOADER_MCUBOOT is not set
# CONFIG_REBOOT is not set
# CONFIG_MISRA_SANE is not set
# end of Boot Options

#
# Compatibility
#
CONFIG_COMPAT_INCLUDES=y
# end of Compatibility

I am using BME680

pabigot, 2020-11-30:
Based on the config UART is disabled.  It sounds like you're measuring the power consumption of the entire board, which would include any circuitry on the E73 module plus whatever it's connected to (BME680, LEDs, any other current drains).  I don't have any other advice.

tcpipchip, 2020-11-30:
@pabigot 

The e73 is not connected to nothing now! And i am getting 100uA! Just running BLE Advertising...

The only very strange stuff was that i had to disable the DC-DC to make the e73 runs my app. Is that a problem ?

I am using a CR coin battery connected to e73 vcc

No problem to me send for now my project for you see inconsistences!

https://github.com/tcpipchip/bmezephyrforced

tcpipchip, 2020-12-01:
Please
Send me a low energy example that really compiles! Thank you!

nordic-krch, 2020-12-01:
I think that you need to disable uart in device tree, too. Add overlay like this:
```
&uart0 {
    status = "disabled"
};

pabigot, 2020-12-01:
I wouldn't expect devicetree to have any effect since the uploaded config shows all UART drivers are disabled.

@tcpipchip Please test energy use with the [Nordic system-off demo](https://docs.zephyrproject.org/latest/samples/boards/nrf/system_off/README.html).  You should be able to measure multiple power levels, including one where everything associated with the MCU is disabled except the ability to restart on a button press.  If in that mode you still see current draw over, say, 10 uA, then it's your board, or failure to configure certain pins to avoid current drain.

Please follow up with measurements of power consumption in each of the two active levels, the long sleep, and the deep sleep.

tcpipchip, 2020-12-01:
Hi team

Thanks for all support that i got here!

you are the best

Well, something is Wrong on E73 that i dont know!  DCC,DCC1, DC-DC

i aborted

I will use the SDK 16 (NORDIC) on a CHINESE beacon that i got!

Thanks again!

I lost 80 hrs without success!

closed 2020-12-02 by pabigot

tcpipchip, 2020-12-02:
I back to the ZEPHYR, now with U-BLOX NINA B112.

Now is working nice :)