zephyrproject-rtos/zephyr #30305: up_squared:  tests/kernel/mem_protect/mem_protect failed.

chen-png, 2020-11-30:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/mem_protect/mem_protect
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
Booting Zephyr OS build zephyr-v2.4.0-1845-g6aa16e01dd91
Running test suite memory_protection_test_suite

**START - test_permission_inheritance**
E: Page fault at address 0x12e990 (error code 0x4)
E: Linear address not present in page tables
E: Access violation: user thread not allowed to read
E: PTE: not present
E: RAX: 0x0000000000000000 RBX: 0x0000000000000000 RCX: 0x0000000000100456 RDX: 0x0000000000000000
E: RSI: 0x0000000000d60fb4 RDI: 0x0000000000d79380 RBP: 0x0000000000000000 RSP: 0x0000000000d60fc8
E:  R8: 0x000000000010048c  R9: 0x8000000000000006 R10: 0x0000000000000000 R11: 0x0000000000000202
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000d60fc8 RFLAGS: 0x0000000000010202 CS: 0x003b CR3: 0x000000000083a000
E: RIP: 0x0000000000100499
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
E: Current thread: 0x12a940 (unknown)
Caught system error -- reason 0 1
fatal error expected as part of test case

**this error is an expected error, but the test it will stop at there, and fail with timeout.**

**Environment (please complete the following information):**
 - OS: Fedora32
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: 6aa16e01dd91

andrewboie, 2020-12-01:
Is this a recent regression or has this been broken for some time? On the off chance you have had time to bisect is there an offending commit?

jenmwms, 2020-12-02:
This is similar symptom as https://github.com/zephyrproject-rtos/zephyr/issues/30275. The workaround is not scalable https://github.com/zephyrproject-rtos/zephyr/pull/30374. This will pass if sanitycheck is informed to wait 120 seconds for the test to complete (TBD if its related to setup/flash/boot time).

jenmwms, 2020-12-03:
Interesting this does look like another case of the test taking longer than the default timeout that sanitycheck is using. The test suite completes quickly back at https://github.com/zephyrproject-rtos/zephyr/commit/b8242bff6441cc9bb310f1a4cac196155b761655, then next commit we start seeing it take longer for `test_permission_inheritance` in https://github.com/zephyrproject-rtos/zephyr/commit/1367c4a4b7f1. (note this allows us to see pre-existing `fatal error was unexpected, aborting` in `test_mem_domain_setup`).

jenmwms, 2020-12-10:
Closing this, it is now part of cluster in https://github.com/zephyrproject-rtos/zephyr/issues/30573

closed 2020-12-10 by jenmwms