zephyrproject-rtos/zephyr #30306: up_squared: tests/kernel/mem_protect/userspace failed.

chen-png, 2020-11-30:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p tests/kernel/mem_protect/userspace
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
Booting Zephyr OS build zephyr-v2.4.0-1845-g6aa16e01dd91 
Running test suite userspace

START - test_is_usermode
PASS - test_is_usermode

START - test_write_control
E: General protection fault (code 0x0)
E: RAX: 0x00000000ffffffff RBX: 0x0000000000117028 RCX: 0x0000000000000000 RDX: 0x0000000000000000
E: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x0000000000000000 RSP: 0x0000000000d4afb8
E:  R8: 0x00000000001056d7  R9: 0x8000000000000006 R10: 0x0000000000000000 R11: 0x0000000000000246
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000d4afb8 RFLAGS: 0x0000000000010246 CS: 0x003b CR3: 0x0000000000cae000
E: RIP: 0x0000000000100578
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
E: Current thread: 0x11b800 (ztest_thread)
Caught system error -- reason 0
System error was expected
PASS - test_write_control

START - test_disable_mmu_mpu
E: General protection fault (code 0x0)
E: RAX: 0x0000000000cae000 RBX: 0x0000000000117050 RCX: 0x0000000000000000 RDX: 0x0000000000000000
E: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x0000000000000000 RSP: 0x0000000000d4afb8
E:  R8: 0x00000000001056d7  R9: 0x8000000000000006 R10: 0x0000000000000000 R11: 0x0000000000000246
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000d4afb8 RFLAGS: 0x0000000000010246 CS: 0x003b CR3: 0x0000000000cae000
E: RIP: 0x00000000001005a9
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x11b800 (ztest_thread)
Caught system error -- reason 0
System error was expected
PASS - test_disable_mmu_mpu
......
START - test_domain_add_thread_context_switch
PASS - test_domain_add_thread_context_switch

START - test_domain_add_part_context_switch
PASS - test_domain_add_part_context_switch

START - test_domain_remove_part_context_switch
E: Page fault at address 0x119000 (error code 0x6)
E: Linear address not present in page tables
E: Access violation: user thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000cae000 RBX: 0x0000000000000000 RCX: 0x0000000000000000 RDX: 0x0000000000000000
E: RSI: 0x0000000000000000 RDI: 0x0000000000119000 RBP: 0x0000000000000000 RSP: 0x0000000000d46fd0
E:  R8: 0x000000000010032b  R9: 0x8000000000000006 R10: 0x0000000000000000 R11: 0x0000000000000246
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000d46fd0 RFLAGS: 0x0000000000010246 CS: 0x003b CR3: 0x0000000000cae000
E: RIP: 0x000000000010032b
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x11a000 (unknown)
Caught system error -- reason 0
System error was expected

**this error is an expected error, but the test it will stop at there, and fail with timeout.**

**Environment (please complete the following information):**
 - OS: Fedora32
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: 6aa16e01dd91

jenmwms, 2020-12-01:
I cannot replicate this 'fail and timeout'. The test suite PASSes for me on up2 at that revision `6aa16e0`. (I do not have a flash script, so I used USB boot media method with grub and zephyr.strip.)
[issue_30306_up2_tests_kernel_mem_protect_userspace.txt](https://github.com/zephyrproject-rtos/zephyr/files/5619633/issue_30306_up2_tests_kernel_mem_protect_userspace.txt)

chen-png, 2020-12-07:
the same timeout issue with #30307

jenmwms, 2020-12-10:
Closing this, it is now part of cluster in https://github.com/zephyrproject-rtos/zephyr/issues/30573

closed 2020-12-10 by jenmwms