zephyrproject-rtos/zephyr #30307: up_squared:  tests/kernel/device/ failed.

chen-png, 2020-11-30:
**Describe**
when running tests/kernel/device/ on up_squared, the test it will stop at "test_bogus_dynamic_name" and fail with timeout.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/device/
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
Booting Zephyr OS build zephyr-v2.4.0-1845-g6aa16e01dd91
Running test suite device

START - test_dummy_device_pm
SKIP - test_dummy_device_pm

START - test_build_suspend_device_list
SKIP - test_build_suspend_device_list

START - test_dummy_device
PASS - test_dummy_device

START - test_enable_and_disable_automatic_idle_pm
SKIP - test_enable_and_disable_automatic_idle_pm

START - test_pre_kernel_detection
PASS - test_pre_kernel_detection

**START - test_bogus_dynamic_name**

**Environment (please complete the following information):**
 - OS: Fedora32
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: 6aa16e01dd91

jenmwms, 2020-12-01:
I cannot replicate this 'fail with timeout'. The test suite PASSes for me on up2 at that revision 6aa16e0. (I do not have a flash script, so I used USB boot media method with grub and zephyr.strip.)
[6aa16e0_up2_tests_kernel_device.txt](https://github.com/zephyrproject-rtos/zephyr/files/5619759/6aa16e0_up2_tests_kernel_device.txt)

chen-png, 2020-12-07:
@jenmwms Hi, how long did you run over this test case and get the final result? I used west build and flash to run this test case, it spent about 5 minutes,  so for sanitycheck script, it will be failed with timeout. it seems that the same issue with #30275 .

jenmwms, 2020-12-10:
Closing this, it is now part of cluster in https://github.com/zephyrproject-rtos/zephyr/issues/30573

closed 2020-12-10 by jenmwms