zephyrproject-rtos/zephyr #30315: Build failure: zephyr/include/generated/devicetree_unfixed.h:627:29: error: 'DT_N_S_leds_S_led_0_P_gpios_IDX_0_PH_P_label' undeclared

anirudhajith, 2020-11-30:
**Description**
Running `west build -p auto -b hifive1 samples/basic/blinky` fails with a build error. This is the full dump:
```
[6/45] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/anirudh/zephyr-sdk-0.11.3/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc -DBUILD_VERSION=zephyr-v2.4.0-1845-g6aa16e01dd91 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/anirudh/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/anirudh/zephyrproject/zephyr/soc/riscv/riscv-privilege/sifive-freedom -I/home/anirudh/zephyrproject/zephyr/soc/riscv/riscv-privilege/common/. -isystem /home/anirudh/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/anirudh/zephyr-sdk-0.11.3/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/9.2.0/include -isystem /home/anirudh/zephyr-sdk-0.11.3/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/9.2.0/include-fixed -Os -imacros /home/anirudh/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mabi=ilp32 -march=rv32ima -imacros /home/anirudh/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/anirudh/zephyrproject/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/anirudh/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/anirudh/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c /home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c
In file included from /home/anirudh/zephyrproject/zephyr/soc/riscv/riscv-privilege/sifive-freedom/soc.h:15,
                 from /home/anirudh/zephyrproject/zephyr/include/arch/riscv/arch.h:28,
                 from /home/anirudh/zephyrproject/zephyr/include/arch/cpu.h:25,
                 from /home/anirudh/zephyrproject/zephyr/include/kernel_includes.h:38,
                 from /home/anirudh/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/anirudh/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c:7:
/home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c: In function 'main':
zephyr/include/generated/devicetree_unfixed.h:627:29: error: 'DT_N_S_leds_S_led_0_P_gpios_IDX_0_PH_P_label' undeclared (first use in this function); did you mean 'DT_N_S_leds_S_led_0_P_pwms_IDX_0_PH'?
  627 | #define DT_N_ALIAS_led0     DT_N_S_leds_S_led_0
      |                             ^~~~~~~~~~~~~~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:1940:24: note: in definition of macro 'DT_CAT'
 1940 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:597:2: note: in expansion of macro 'DT_PROP'
  597 |  DT_PROP(DT_PHANDLE_BY_IDX(node_id, phs, idx), prop)
      |  ^~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:390:32: note: in expansion of macro 'DT_CAT'
  390 | #define DT_PROP(node_id, prop) DT_CAT(node_id, _P_##prop)
      |                                ^~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:859:2: note: in expansion of macro 'DT_PROP'
  859 |  DT_PROP(node_id, prop##_IDX_##idx##_PH)
      |  ^~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:597:10: note: in expansion of macro 'DT_PHANDLE_BY_IDX'
  597 |  DT_PROP(DT_PHANDLE_BY_IDX(node_id, phs, idx), prop)
      |          ^~~~~~~~~~~~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree/gpio.h:61:2: note: in expansion of macro 'DT_PROP_BY_PHANDLE_IDX'
   61 |  DT_PROP_BY_PHANDLE_IDX(node_id, gpio_pha, idx, label)
      |  ^~~~~~~~~~~~~~~~~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree/gpio.h:72:2: note: in expansion of macro 'DT_GPIO_LABEL_BY_IDX'
   72 |  DT_GPIO_LABEL_BY_IDX(node_id, gpio_pha, 0)
      |  ^~~~~~~~~~~~~~~~~~~~
/home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c:19:14: note: in expansion of macro 'DT_GPIO_LABEL'
   19 | #define LED0 DT_GPIO_LABEL(LED0_NODE, gpios)
      |              ^~~~~~~~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:1940:24: note: in expansion of macro 'DT_N_ALIAS_led0'
 1940 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:202:25: note: in expansion of macro 'DT_CAT'
  202 | #define DT_ALIAS(alias) DT_CAT(DT_N_ALIAS_, alias)
      |                         ^~~~~~
/home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c:16:19: note: in expansion of macro 'DT_ALIAS'
   16 | #define LED0_NODE DT_ALIAS(led0)
      |                   ^~~~~~~~
/home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c:19:28: note: in expansion of macro 'LED0_NODE'
   19 | #define LED0 DT_GPIO_LABEL(LED0_NODE, gpios)
      |                            ^~~~~~~~~
/home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c:36:27: note: in expansion of macro 'LED0'
   36 |  dev = device_get_binding(LED0);
      |                           ^~~~
zephyr/include/generated/devicetree_unfixed.h:627:29: note: each undeclared identifier is reported only once for each function it appears in
  627 | #define DT_N_ALIAS_led0     DT_N_S_leds_S_led_0
      |                             ^~~~~~~~~~~~~~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:1940:24: note: in definition of macro 'DT_CAT'
 1940 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:597:2: note: in expansion of macro 'DT_PROP'
  597 |  DT_PROP(DT_PHANDLE_BY_IDX(node_id, phs, idx), prop)
      |  ^~~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:390:32: note: in expansion of macro 'DT_CAT'
  390 | #define DT_PROP(node_id, prop) DT_CAT(node_id, _P_##prop)
      |                                ^~~~~~
/home/anirudh/zephyrproject/zephyr/include/devicetree.h:859:2: note: in expansion of macro 'DT_PROP'
  859 |  DT_PROP(node_id, prop##_IDX_##idx##_PH)
[... 58 lines cut ...]
/home/anirudh/zephyrproject/zephyr/samples/basic/blinky/src/main.c:41:32: note: in expansion of macro 'PIN'
   41 |  ret = gpio_pin_configure(dev, PIN, GPIO_OUTPUT_ACTIVE | FLAGS);
      |                                ^~~
[15/45] Building C object zephyr/drivers/spi/CMakeFiles/drivers__spi.dir/spi_sifive.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/anirudh/zephyrproject/zephyr/build

```

It seems to be the case that there is an undeclared reference to `DT_N_S_leds_S_led_0_P_gpios_IDX_0_PH_P_label` in `zephyr/include/generated/devicetree_unfixed.h:627:29`

**To Reproduce**
Steps to reproduce the behavior:
* Follow the steps to set up SDK, etc. at [Getting Started](https://docs.zephyrproject.org/latest/getting_started/index.html) and use `hifive1` as `<your-board-name>` in step 5.

**Expected behavior**
I expect it to compile

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 1b65fd066f66596177fbce2f60f0deb2697bb2cb

brucehoult, 2020-12-10:
I have the same errors. First time attempting to use Zephyr, fresh install precisely following the getting started instructions (other than that the cmake instructions didn't work because of lack of a key so I had to use instructions from the kitware site instead)

brucehoult, 2020-12-10:
hifive1_revb instead of hifive1 builds, but of course that's not the board I want.

anirudhajith, 2020-12-10:
@brucehoult I finally went with `samples/hello_world` as a substitute to `samples/blinky`. Try running `west build -p auto -b hifive1 samples/hello_world/`.

brucehoult, 2020-12-10:
Yeah, things other than blinky seem to build. minimal, blinky_pwm. Philosophers. hello_world, as you say.

PR #31887 (2021-02-02): boards: hifive1: remove incorrect alias for GPIO use of LEDs
The devicetree for this board only provides PWM-compatible LEDs.  Remove the aliases that suggest it supports GPIO-compatible LEDs.

Fixes #30315 by producing:

```
/mnt/nordic/zp/zephyr/samples/basic/blinky/src/main.c:24:2: error: #error "Unsupported board: led0 devicetree alias is not defined"
   24 | #error "Unsupported board: led0 devicetree alias is not defined"
      |  ^~~~~
````
closed 2021-02-02 by nashif