zephyrproject-rtos/zephyr #30316: updatehub with openthread

nehalrohit, 2020-11-30:
Hello
I am doing some changes to the updatehub code with the openthread demo. and successfully connect with updatehub local server here is my code logs.
[image]

create a package  using this 
 **../../../../bootloader/mcuboot/scripts/imgtool.py sign --key ../../../../bootloader/mcuboot/root-ec-p256.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x67000 --pad-header build/zephyr/zephyr.hex build/zephyr/zephyr1.hex** 

linux config
[image]
[image]
[image]

my problem is my code not successfully update the image.

nandojve, 2020-11-30:
Hi @nehalrohit, 
There are two commands to sign image:

1) **with** --pad option (used to upload image using west flash) This can be BIN or HEX
`imgtool.py sign --key <dir>/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x67000 --pad build/zephyr/zephyr.hex build/zephyr/zephyr-1.0.0.hex`

2) **without** --pad/--pad-header options (used to create UHU package) this can be ONLY BIN
`imgtool.py sign --key <dir>/root-rsa-2048.pem --header-size 0x200 --align 8 --version 2.0.0 --slot-size 0x67000 build/zephyr/zephyr.bin build/zephyr/zephyr-2.0.0.bin`

The ***.pem** file must be the same that you used to compile MCUboot. On my case, I've been using the default **root-rsa-2048.pem**. My suggestion is, keep as simple possible at first and validate the full procedure.

CC: @otavio

nehalrohit, 2020-12-01:
Hi @nandojve 
**with --pad option (used to upload image using west flash) This can be BIN or HEX
imgtool.py sign --key <dir>/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x67000 --pad build/zephyr/zephyr.hex build/zephyr/zephyr-1.0.0.hex**

 with only --pad does not work for me. it gives me an error.
[image]

if I am using with pad-header and use **root-rsa-2048.pem** this key, successfully generate hex files but it gives me an error 
[image]

So I used  **../../../../bootloader/mcuboot/scripts/imgtool.py sign --key ../../../../bootloader/mcuboot/root-ec-p256.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x67000 --pad-header build/zephyr/zephyr.hex build/zephyr/zephyr-1.0.0.hex**
that Is work for me.

and create a package with this 
**../../../../bootloader/mcuboot/scripts/imgtool.py sign --key ../../../../bootloader/mcuboot/root-ec-p256.pem --header-size 0x200 --align 8 --version 2.0.0 --slot-size 0x67000 --pad-header build/zephyr/zephyr.bin build/zephyr/zephyr-2.0.0.bin**

still, give me an error , please give me some proper solution about it because it's very important for me my entire project stuck just because of this

nehalrohit, 2020-12-01:
Hi @nandojve 
I am doing a successful image upload but it takes too much time to update the new image about 20min.

[image]

but after reboot, the new image not update.
[image]

nandojve, 2020-12-01:
Hi @nehalrohit ,

Let's try figure out!

1) why you are using `root-ec-256.pem` instead `root-rsa-2048.pem`?
Remember, you must use same key that MCUboot have been built. The default is `root-rsa-2048.pem`. Otherwise, the image will not be validated.

2) The MCUboot that you are using have only 1 slot. You need enable dual slot because you must have a second place to download the wholly image on the device before trigger swap.

Let's use this procedure for now to build MCUboot version.
This is how I build Nordic MCUboot to proper work with UpdateHub:
```
cd <your zephyr dir>
git checkout master
git  fetch upstream or origin
git rebase upstream/master
west update
cd ../bootloader/mcuboot/boot/zephyr
rm -rf build-nrf52840dk_nrf52840
mkdir build-nrf52840dk_nrf52840 && cd build-nrf52840dk_nrf52840
cmake -GNinja -DBOARD=nrf52840dk_nrf52840 ..

# note below PEM file selected!
MCUBoot bootloader key file: <your zephyr dir>/bootloader/mcuboot/root-rsa-2048.pem

ninja
west update
```

Note that there are two slots now (primary and secondary images)
```
*** Booting Zephyr OS build zephyr-v2.4.0-1305-g2a462de260c6  ***
[00:00:00.383,575] <inf> mcuboot: Starting bootloader
[00:00:00.389,984] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.400,329] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.410,797] <inf> mcuboot: Boot source: none
[00:00:00.416,290] <wrn> mcuboot: Failed reading image headers; Image=0
[00:00:00.423,889] <err> mcuboot: Unable to find bootable image
```

3) If you upgrade a hello world sample without network support, UpdateHub will never send any information about what happened after upgrade to the server and server will wait "forever". Please, package the same image that you have running on the device that access UpdateHub-CE with a different version:

Follow this procedure:

1) build the image that you have access to UpdateHub CE
2) sign with --pad option
`imgtool.py sign --key <dir>/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.0.0 --slot-size 0x67000 --pad build/zephyr/zephyr.hex build/zephyr/zephyr-1.0.0.hex`
3) Upload the image to board
`west flash --hex-file build/zephyr/zephyr-1.0.0.hex`

4) sign without --pad/--pad-header options (now must select the bin file)
`imgtool.py sign --key <dir>/root-rsa-2048.pem --header-size 0x200 --align 8 --version 2.0.0 --slot-size 0x67000 build/zephyr/zephyr.bin build/zephyr/zephyr-2.0.0.bin`

5) Create UHU package
https://docs.zephyrproject.org/latest/samples/subsys/mgmt/updatehub/README.html#step-8-create-a-package-with-updatehub-utilities-uhu

Notes:
- UpdateHub is small and don't increase the firmware size. It selects all pieces and put it together to have a full functional OTA.
- OpenThread is a very large network layer and probably may be responsible for half image size.
- Speed, It is difficult say something without proper configuration.

My initial metrics are:
* less than 10min for IEEE 802.15.4
* less than 30min for BLE IPSP

I'm looking forward to hear from you the result.

nehalrohit, 2020-12-02:
hello @nandojve 
**sign without --pad/--pad-header options (now must select the bin file)
imgtool.py sign --key <dir>/root-rsa-2048.pem --header-size 0x200 --align 8 --version 2.0.0 --slot-size 0x67000 build/zephyr/zephyr.bin build/zephyr/zephyr-2.0.0.bin**

when I create a sign image for ota it gives error.
**Error: Header padding was not requested and image does not start with zeros**

nehalrohit, 2020-12-03:
Hi @nandojve  
Thank you very much I did it successfully.

after adding  **sysreboot** function. I did successfully update the image.
[image]
[image]

it takes around **23 min** (updatehub+openthread bin file 386kB) is it any other configuration to reduce update time.
 it is **prj.conf** file

CONFIG_NCS_SAMPLES_DEFAULTS=n

CONFIG_NEWLIB_LIBC=n

# Generic networking options
CONFIG_NETWORKING=n
CONFIG_NET_CONNECTION_MANAGER=n

# Enable multiprotocol service layer
CONFIG_MPSL=n

# Kernel options
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_INIT_STACKS=y

# Increase set for threads with meta-irq priority
CONFIG_NUM_METAIRQ_PRIORITIES=1

# Logging
CONFIG_NET_LOG=n
CONFIG_NET_STATISTICS=n

# Disable certain parts of Zephyr IPv6 stack
CONFIG_NET_IPV6_NBR_CACHE=n
CONFIG_NET_IPV6_MLD=n

# Stack sizes configuration
CONFIG_NET_TX_STACK_SIZE=1200
CONFIG_NET_RX_STACK_SIZE=1500

# Network buffers
CONFIG_NET_PKT_RX_COUNT=10
CONFIG_NET_PKT_TX_COUNT=10
CONFIG_NET_BUF_RX_COUNT=16
CONFIG_NET_BUF_TX_COUNT=16

# L2 OpenThread enabling
CONFIG_NET_L2_OPENTHREAD=y
CONFIG_OPENTHREAD_DEBUG=n
CONFIG_OPENTHREAD_L2_DEBUG=n
CONFIG_OPENTHREAD_L2_LOG_LEVEL_INF=n

# Disable vanilla mbedtls in favor of custom backends
CONFIG_OPENTHREAD_MBEDTLS=n

# Disable unneeded mbedtls modules
CONFIG_MBEDTLS_CIPHER_MODE_CBC=n
CONFIG_MBEDTLS_CIPHER_MODE_CTR=n
CONFIG_MBEDTLS_CIPHER_MODE_CFB=n
CONFIG_MBEDTLS_CIPHER_MODE_OFB=n
CONFIG_MBEDTLS_CIPHER_MODE_XTS=n
CONFIG_MBEDTLS_GCM_C=n
CONFIG_MBEDTLS_CHACHA20_C=n
CONFIG_MBEDTLS_POLY1305_C=n
CONFIG_MBEDTLS_CHACHAPOLY_C=n
CONFIG_MBEDTLS_DHM_C=n
CONFIG_MBEDTLS_RSA_C=n
CONFIG_MBEDTLS_SHA1_C=n
CONFIG_MBEDTLS_SHA512_C=n
CONFIG_MBEDTLS_HEAP_SIZE=10240
#
# Copyright (c) 2020 Nordic Semiconductor
#
# SPDX-License-Identifier: LicenseRef-BSD-5-Clause-Nordic
#

# nRF board library
CONFIG_DK_LIBRARY=y

# Enable OpenThread CoAP support API
CONFIG_OPENTHREAD_COAP=y

# Configure sample logging setting
CONFIG_COAP_SERVER_LOG_LEVEL_DBG=n
CONFIG_OT_COAP_UTILS_LOG_LEVEL_DBG=n

# Network shell
CONFIG_NET_SHELL=y
CONFIG_SHELL_ARGC_MAX=26
CONFIG_SHELL_CMD_BUFF_SIZE=416

# Network sockets
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4

# OpenThread manual start
CONFIG_OPENTHREAD_MANUAL_START=y

CONFIG_OPENTHREAD_DNS_CLIENT=y

# Enable OpenThread features set
CONFIG_OPENTHREAD_NORDIC_LIBRARY_MASTER=y

CONFIG_OPENTHREAD_MTD=y
CONFIG_OPENTHREAD_MTD_SED=y
CONFIG_OPENTHREAD_POLL_PERIOD=3000
CONFIG_RAM_POWER_DOWN_LIBRARY=y
CONFIG_DEVICE_POWER_MANAGEMENT=y

# Turn on the updatehub
CONFIG_UPDATEHUB=y

# Minimal Heap mem pool size for the updatehub working
CONFIG_HEAP_MEM_POOL_SIZE=8192
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_SHELL_STACK_SIZE=8192

# Network management events
CONFIG_NET_CONNECTION_MANAGER=y

CONFIG_NET_IPV4=n
CONFIG_NET_CONFIG_NEED_IPV4=n

CONFIG_NET_IPV6=y
CONFIG_NET_DHCPV4=n

# Optional if you would like test on the your server
CONFIG_SHELL=y
CONFIG_NET_SHELL=y
CONFIG_UPDATEHUB_SHELL=y
CONFIG_SHELL_STACK_SIZE=6144

# Debug helpers
#CONFIG_LOG=y
#CONFIG_UPDATEHUB_LOG_LEVEL_DBG=y
#CONFIG_COAP_LOG_LEVEL_DBG=y
# LOG_IMMEDIATE could infer problems on WIFI reception and is recommended keep disabled
# Use with caution!!
#CONFIG_LOG_IMMEDIATE=y

CONFIG_UPDATEHUB_PRODUCT_UID="e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f"
#CONFIG_UPDATEHUB_POLL_INTERVAL=1
#CONFIG_UPDATEHUB_MANUAL=1
CONFIG_UPDATEHUB_CE=y

CONFIG_COAP_INIT_ACK_TIMEOUT_MS=10000

# Add Google 8.8.8.8 DNS Server
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="fdaa:bb:1::2"

# Add UpdateHub Cloud
CONFIG_UPDATEHUB_SERVER="fd11:1111:1122:0:35f5:e92d:7629:b4bd"
#CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=4

#CONFIG_STDOUT_CONSOLE=y
#CONFIG_I2C=y
#CONFIG_SENSOR=y
#CONFIG_SHT3XD=y

CONFIG_REBOOT=y

nandojve, 2020-12-03:
Hi @nehalrohit ,

Below is my setup

IEEE 802.15.4:
Host: PC Linux > 4.9 + RADVD > 2.7 + ATUSB dongle or WPANUSB (can be nRF/rf2xx running Zephyr)
Remote Node:  nRF/rf2xx 

OpenThread:
Host: PC Linux > 4.9 + RADVD > 2.7 + nRF52 NCP OpenThread 1.1.1, original code (not Zephyr's)
Remote Node:  nRF52

I would like know how is your setup?
How many nodes you have at same time?

Note:
#CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=4
Above Kconfig define the CoAP block size. The default value is 6 which means 1024 bytes payload.
CoAP will request IPv6 data fragments to complete the 1024 payload before pass that data to UpdateHub.
You can decrease expoente to 4 and 2 to check best performance. I think it will be 2 for OpenThread.
CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=4
and
CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2

nehalrohit, 2020-12-04:
Hi @nandojve 
I have some questions.
1.- how did you setup **RADVD** in your host Linux?
2 - total time to update the image from your openthread setup?
3 - node and Linux system are in the same network(PC Linux --> wifi --> border router ) is it possible to ping Linux system from the node.

currently for testing, my setup same as your **PC Linux -> NCP dongle(NCP OpenThread 1.1.1)  and 1 node**  

I  change **CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2 or 4**  but it takes more time as compared to  **CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=6**
  
Nehal........

nandojve, 2020-12-11:
> 1.- how did you setup RADVD in your host Linux?

It is not necessary to run UpdateHub and have internet connectivity.

> 2 - total time to update the image from your openthread setup?

The below scenario uses OT 1.1 with NCP on a Linux PC as border router. It uses Tayga for NAT64 and OT example default configs. This is based on PR #30460. The UpdateHub-CE is a docker hosted at an US service provider at NY. I'm running at South America and ping is around 138 ms from OT node to US server.

Application un-signed
```
[513/518] Linking CXX executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      339648 B       412 KB     80.51%
            SRAM:      115631 B       256 KB     44.11%
        IDT_LIST:         120 B         2 KB      5.86%
[518/518] Linking CXX executable zephyr/zephyr.elf
```

Download: ~ 16m 30s
Swap: ~30s
Confirm: ~1 min

```
*** Booting Zephyr OS build zephyr-v2.4.0-2049-gdd5f0994f9d2  ***
[00:00:00.393,005] <inf> mcuboot: Starting bootloader
[00:00:00.399,475] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.409,942] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.420,501] <inf> mcuboot: Boot source: none
[00:00:00.426,208] <inf> mcuboot: Swap type: none
[00:00:00.878,601] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.886,535] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2239-gacd86316efd9  ***
[00:00:00.004,272] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.016,815] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.016,815] <inf> fs_nvs: alloc wra: 0, f40
[00:00:00.016,815] <inf> fs_nvs: data wra: 0, 178
[00:00:00.030,334] <inf> main: UpdateHub sample app started
[00:00:00.030,334] <inf> main: Confirming the boot image
[00:00:00.030,395] <inf> main: Network connected
[00:00:00.030,395] <inf> main: Starting UpdateHub polling mode
[00:00:00.030,395] <inf> updatehub: SHA-256 verification on download and from flash
[00:00:02.031,982] <err> updatehub: Could not receive data
uart:~$ updatehub info 
Unique device id: d7eed6b80dd18d94
Firmware Version: 1.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: XXXXXXX
[00:00:16.942,413] <inf> updatehub: No update available
[00:01:02.200,653] <inf> updatehub: No update available
[00:02:02.393,066] <inf> updatehub: Probe metadata received
[00:18:31.403,961] <inf> updatehub: Firmware download complete
�*** Booting Zephyr OS build zephyr-v2.4.0-2049-gdd5f0994f9d2  ***u can reboot now
[00:00:00.255,889] <inf> mcuboot: Starting bootloader
[00:00:00.262,390] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.272,827] <inf> mcuboot: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
[00:00:00.283,294] <inf> mcuboot: Boot source: none
[00:00:00.289,001] <inf> mcuboot: Swap type: test
[00:00:33.645,111] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:33.653,045] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2239-gacd86316efd9  ***
[00:00:00.004,272] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.016,662] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.016,662] <inf> fs_nvs: alloc wra: 0, c60
[00:00:00.016,662] <inf> fs_nvs: data wra: 0, 680
[00:00:00.045,623] <inf> main: UpdateHub sample app started
[00:00:00.045,623] <inf> main: Confirming the boot image
[00:00:00.045,745] <inf> main: Network connected
[00:00:00.045,745] <inf> main: Starting UpdateHub polling mode
[00:00:00.045,745] <inf> updatehub: SHA-256 verification on download and from flash
[00:00:02.047,363] <err> updatehub: Could not receive data
[00:01:02.214,233] <inf> updatehub: No update available
uart:~$ updatehub info 
Unique device id: d7eed6b80dd18d94
Firmware Version: 2.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: XXXXXX
uart:~$ net ping XXXXXX
PING XXXXXX
8 bytes from XXXXXX to fd11:22::6a58:2c2c:1425:7900: icmp_seq=0 ttl=48 rssi=0 time=139 ms
8 bytes from XXXXXX to fd11:22::6a58:2c2c:1425:7900: icmp_seq=1 ttl=48 rssi=0 time=136 ms
8 bytes from XXXXXX to fd11:22::6a58:2c2c:1425:7900: icmp_seq=2 ttl=48 rssi=0 time=138 ms
uart:~$ 
```

> 3 - Any reason why it takes too much time to ota update( when I did ota with using thread for Taxsas CC1352R1 MCU it takes less than 10 sec)?

This is interesting, are you sure you can download ~400k byte data from a server, with NAT, store data on flash than update in less than 10s? Maybe I can achieve using 100M bit Ethernet on Zephyr. I think Taxsas numbers that you shared seems unrealistic knowing that you can have 128 byte max air frame at 250k bit rate with compressed IPv6.

> 4 - node and Linux system are in the same network(PC Linux --> wifi --> border router ) is it possible to ping the Linux system from the node.

Yes, you can add a manual IP on node and Linux or use RADVD.
If you ping from node `net ping ff02::1` you will see Linux IPv6 unicast slaac address.

closed 2020-12-16 by nehalrohit

IHIDCHAOS, 2020-12-28:
Hi @nandojve 
I am following your guide above to do OTA too.
In the last step, I encountered this error:`<err> updatehub: Could not receive data`
```
uart:~$ updatehub info
Unique device id: 4969258d739710de
Firmware Version: 1.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: 64:ff9b::6a0d:d7b7
[00:01:02.171,905] <inf> updatehub: Probe metadata received
[00:01:30.928,161] <err> updatehub: Could not receive data
[00:01:32.928,253] <err> updatehub: Could not receive data
[00:01:34.928,344] <err> updatehub: Could not receive data
[00:01:36.928,436] <err> updatehub: Could not receive data
[00:01:38.928,527] <err> updatehub: Could not receive data
[00:01:49.753,234] <err> updatehub: Could not receive data
[00:01:51.753,356] <err> updatehub: Could not receive data
[00:01:53.753,448] <err> updatehub: Could not receive data
[00:01:55.753,540] <err> updatehub: Could not receive data
[00:01:57.753,631] <err> updatehub: Could not receive data
[00:02:57.824,279] <err> updatehub: Could not receive data
[00:02:59.824,371] <err> updatehub: Could not receive data
uart:~$ net ping 64:ff9b::8.8.8.8
PING 64:ff9b::8.8.8.8
8 bytes from 64:ff9b::808:808 to fd11:22::dfbd:cd1:d6f0:d1d4: icmp_seq=0 ttl=111 rssi=0 time=48 ms
8 bytes from 64:ff9b::808:808 to fd11:22::dfbd:cd1:d6f0:d1d4: icmp_seq=1 ttl=111 rssi=0 time=58 ms
8 bytes from 64:ff9b::808:808 to fd11:22::dfbd:cd1:d6f0:d1d4: icmp_seq=2 ttl=111 rssi=0 time=64 ms
uart:~$ 
```

Could you please help me to solve this problem? Thanks in advance!

nandojve, 2020-12-28:
Hi @IHIDCHAOS ,
UpdateHub is expecting an UpdateHub-CE running at 64:ff9b::6a0d:d7b7. Are you sure that UpdateHub-CE is running? Are there routes, no firewall blocking etc?
https://docs.zephyrproject.org/latest/samples/subsys/mgmt/updatehub/README.html#step-2-1-updatehub-ce-community-edition

IHIDCHAOS, 2020-12-28:
> Hi @IHIDCHAOS ,
> UpdateHub is expecting an UpdateHub-CE running at 64:ff9b::6a0d:d7b7. Are you sure that UpdateHub-CE is running? Are there routes, no firewall blocking etc?
> https://docs.zephyrproject.org/latest/samples/subsys/mgmt/updatehub/README.html#step-2-1-updatehub-ce-community-edition

Hi @nandojve ,
I am sure that my UpdateHub-CE is running like this (old picture):
[image]

And I think the output `[00:01:02.171,905] <inf> updatehub: Probe metadata received` can also prove this.
I have tried to capture the packet from Wireshark on My OTBR wpan0, It seems that it's always sending the same packet.
I will put some packets later. Thank you!

nandojve, 2020-12-28:
Hi @IHIDCHAOS ,

Yes, you are right! It seems something in the UpdateHub-CE side, at first. Are you using latest version or an old one?
Any data that you can share may help to find the problem. Share your diff configs from current master too.

IHIDCHAOS, 2020-12-29:
Yes I am using the latest Version: `commit 4ca7942411f70f0caee6cd872ab14200f88b221b (HEAD -> master, origin/master)`

I modified:

```
ihid@ubuntu:~/zephyrproject/zephyr$ git diff
diff --git a/samples/subsys/mgmt/updatehub/overlay-ot.conf b/samples/subsys/mgmt/updatehub/overlay-ot.conf
index d425f950a3..9f5ec77eae 100644
--- a/samples/subsys/mgmt/updatehub/overlay-ot.conf
+++ b/samples/subsys/mgmt/updatehub/overlay-ot.conf
@@ -28,10 +28,11 @@ CONFIG_OPENTHREAD_PANID=4660
 CONFIG_OPENTHREAD_CHANNEL=15
 CONFIG_OPENTHREAD_NETWORK_NAME="OpenThreadDemo"
 CONFIG_OPENTHREAD_XPANID="0x1111111122222222"
-CONFIG_OPENTHREAD_JOINER=y
-CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
-CONFIG_OPENTHREAD_JOINER_PSKD="J01NU5"
+#CONFIG_OPENTHREAD_JOINER=y
+#CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
+#CONFIG_OPENTHREAD_JOINER_PSKD="J01NME"
 CONFIG_OPENTHREAD_SLAAC=y
+CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6
 
 CONFIG_NET_IPV6_NBR_CACHE=n
 CONFIG_NET_IPV6_MLD=n
@@ -42,7 +43,6 @@ CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768
 # UpdateHub
 CONFIG_MAIN_STACK_SIZE=8192
 CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2
-
 # Only tested with 6to4
 #
 #######################
@@ -54,4 +54,4 @@ CONFIG_DNS_SERVER_IP_ADDRESSES=y
 # Add prefix        64:ff9b::
 CONFIG_DNS_SERVER1="64:ff9b::808:808"
 # Repeat the above step with your server IPv4 address translated to IPv6
-CONFIG_UPDATEHUB_SERVER="<your server ip address>"
+CONFIG_UPDATEHUB_SERVER="64:ff9b::6a0d:d7b7"
diff --git a/samples/subsys/mgmt/updatehub/src/main.c b/samples/subsys/mgmt/updatehub/src/main.c
index a2d82323ae..efde6c34db 100644
--- a/samples/subsys/mgmt/updatehub/src/main.c
+++ b/samples/subsys/mgmt/updatehub/src/main.c
@@ -11,6 +11,7 @@
 #include <net/net_conn_mgr.h>
 #include <net/wifi_mgmt.h>
 #include <dfu/mcuboot.h>
+#include <power/reboot.h>
 
 #if defined(CONFIG_UPDATEHUB_DTLS)
 #include <net/tls_credentials.h>
@@ -27,6 +28,7 @@ static struct net_mgmt_event_callback mgmt_cb;
 
 void start_updatehub(void)
 {
+       int ret;
 #if defined(CONFIG_UPDATEHUB_SAMPLE_POLLING)
        LOG_INF("Starting UpdateHub polling mode");
        updatehub_autohandler();

```

And add: `zephyrproject/zephyr/samples/subsys/mgmt/updatehub/overlay-prj.conf`

```
#
# Copyright (c) 2018 O.S.Systems
#
# SPDX-License-Identifier: Apache-2.0
#
CONFIG_SHELL=y
CONFIG_OPENTHREAD_SHELL=y
CONFIG_UPDATEHUB_SHELL=y
#Example of product_uid, you must set you product uid.
CONFIG_UPDATEHUB_PRODUCT_UID="e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f"

#If you use the autorandler, need set UPDATEHUB_POLL_INTERVAL.
#This example, the updatehub will poll on the server
#request for updates one to one minute.
#(min: 0 minute, máx: 30 days)

CONFIG_UPDATEHUB_SAMPLE_MANUAL=y
#CONFIG_UPDATEHUB_POLL_INTERVAL=1

CONFIG_UPDATEHUB_CE=y

#You need set the updatehub-ce ip Address.
#This is a example of ip address
CONFIG_UPDATEHUB_SERVER="64:ff9b::6a0d:d7b7"

```
UpdateHub-CE runs the latest version too and the port are opened.:
```
docker run -it -d -p 8080:8080 -p 5683:5683/udp --rm updatehub/updatehub-ce:latest
```