zephyrproject-rtos/zephyr #30322: tests: benchmarks: latency_measure: timing measurement values are all 0

ABOSTM, 2020-11-30:
**Describe the bug**
Running tests/benchmarks/latency_measure on nucleo_f746zg,
test is passed, but all timing measurement values are `0 ns`.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f746zg tests/benchmarks/latency_measure
2. west flash
3. See log

**Expected behavior**
Non null timing measurement value

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.4.0-1849-gc0650217cc7c  ***
START - Time Measurement
Timing results: Clock frequency: 72 MHz
Average thread context switch using yield                   :       0 cycles ,        0 ns
Average context switch time between threads (coop)          :       0 cycles ,        0 ns
Switch from ISR back to interrupted thread                  :       0 cycles ,        0 ns
Time from ISR to executing a different thread               :       0 cycles ,        0 ns
Time to create a thread (without start)                     :       0 cycles ,        0 ns
Time to start a thread                                      :       0 cycles ,        0 ns
Time to suspend a thread                                    :       0 cycles ,        0 ns
Time to resume a thread                                     :       0 cycles ,        0 ns
Time to abort a thread (not running)                        :       0 cycles ,        0 ns
Average semaphore signal time                               :       0 cycles ,        0 ns
Average semaphore test time                                 :       0 cycles ,        0 ns
Semaphore take time (context switch)                        :       0 cycles ,        0 ns
Semaphore give time (context switch)                        :       0 cycles ,        0 ns
Average time to lock a mutex                                :       0 cycles ,        0 ns
Average time to unlock a mutex                              :       0 cycles ,        0 ns
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain gnuarmemb
 - Commit SHA 1b65fd066f66596177fbce2f60f0deb2697bb2cb

**Additional context**
Add any other context about the problem here.

ABOSTM, 2020-11-30:
Issue is linked to DWT which never starts.
Writing to DWT registers is inefficient.
On Cortex M7 access to DWT register may be lock.
See #30327

nashif, 2020-11-30:
@ioannisg  FYI

ioannisg, 2020-12-01:
Will take a look, sure

PR #30327 (2020-12-02): arch: arm: aarch32:cortex_m: timing.c: cortex M7 may need DWT unlock
arch: arm: aarch32:cortex_m: timing.c: cortex M7 may need DWT unlock

On Cortex M7, we need to check the optional presence of
Lock Access Register (LAR) which is indicated in
Lock Status Register (LSR).
When present, a special access token must be written to unlock DWT
registers.

Fixes #30322
closed 2020-12-02 by ioannisg