zephyrproject-rtos/zephyr #30325: Stack overflow with http post when using civetweb

nicovince, 2020-11-30:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
I'm prototyping with a nucleo h743 an embedded web server with GET and POST capabilities. I have no issues with the GET part, but I am encountering stack overflow issues with POST requests.

The issue rises after processing the first key/value, the buffer (8k, located on the stack) containing the POST's body is memmove() onto itself to get rid of the already used part. At some points zephyr crashes with the following message:

```
[00:00:10.303,000] <err> os: ***** MPU FAULT *****
[00:00:10.303,000] <err> os:   Data Access Violation
[00:00:10.303,000] <err> os:   MMFAR Address: 0x2400e200
[00:00:10.303,000] <err> os: r0/a1:  0x2400dec8  r1/a2:  0x2400e205  r2/a3:  0x2400e200
[00:00:10.303,000] <err> os: r3/a4:  0x2400fec3 r12/ip:  0x24004828 r14/lr:  0x0800c217
[00:00:10.303,000] <err> os:  xpsr:  0x81000000
[00:00:10.303,000] <err> os: Faulting instruction address (r15/pc): 0x08010cc4
[00:00:10.303,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:10.303,000] <err> os: Current thread: 0x24009078 (unknown)
[00:00:10.324,000] <err> os: Halting system
```

The buf size is determined with the macro `#define MG_BUF_LEN (1024 * 8)` in `civetweb.c`. Reducing it to 1k fixes the stackoverflow.
I tried to increase the thread stack size from `CONFIG_MAIN_STACK_SIZE` to 16384 and leaving `MG_BUF_LEN` as is without success.
I tried to run my sample application under qemu, without success.

**To Reproduce**

1. Download and extract [civetweb_restapi.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/5616805/civetweb_restapi.tar.gz)
1. Adapt prj.conf ips addresses to your needs
1. `west build -p auto -b nucleo_h743zi -d build <folder where you extracted the attachement>`
1. `west flash -d build`
1. With a browser connect to the board, in the form, enter a **number** and submit.
1. You should have the message posted above.

**Expected behavior**
No stack overflow ;)

**Impact**
System needs to be reset

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - zephyr: fe7eda6ac83f406c46b31fb0ea6bc0c0c2e93d84
 - civetweb: zephyrproject-rtos/civetweb@e6903b80c09d17cd1a8bb32e40080005558aad29

Nukersson, 2020-12-01:
Dear @nicovince Thank you for your bug report. The CivetWeb sample were fitted by me to gave smallest footprint, to run on possible wide range of MCU's.

Generally, you can try to change following values of your `prj.conf` to reach needed memory footprint and performance:
> CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=16384
CONFIG_NET_TX_STACK_SIZE=2048
CONFIG_NET_RX_STACK_SIZE=2048
CONFIG_ISR_STACK_SIZE=2048
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_IDLE_STACK_SIZE=1024

I would say, you should try to change `CONFIG_MAIN_STACK_SIZE` to 8192 instead of 2048 or more without changing other values in `civetweb.c`. Please let me know if it helped.

Currently I can not test it on my HW, but I will let you know, if I could find a fix for your problem.

Best regards,

nicovince, 2020-12-02:
Hi @Nukersson thank for your answer.

I went up to 32768 on `CONFIG_MAIN_STACK_SIZE` without any improvements. Even when reducing `MG_BUF_LEN` to 2kB.
The only way to avoid a crash is setting `MG_BUF_LEN` to 1kB.

btw, I updated Zephyr to 302db10537468170a40c35b852b6f0e3ffecba9c (latest at time of test).

Nukersson, 2020-12-02:
> Hi @Nukersson thank for your answer.
>
> I went up to 32768 on `CONFIG_MAIN_STACK_SIZE` without any improvements. Even when reducing `MG_BUF_LEN` to 2kB.
> The only way to avoid a crash is setting [...]

@nicovince Thank you. Looks like we have to do with a bug here. I need time to investigate it.

nvincent-vossloh, 2021-02-04:
I have not re-tested yet with latest zephyr tree yet, but if it is present I'm interested in keeping this issue open.

Thanks.

closed 2021-04-22 by github-actions[bot]