zephyrproject-rtos/zephyr #30329: drivers: gpio: Combined drive strength flags

statropy, 2020-11-30:
**Is your enhancement proposal related to a problem? Please describe.**
Most microcontrollers support some sort of drive strength option for GPIO. The Zephyr implementation provides flags for using an alternate drive strength, but these flags are targeted at a specific vendor implementation. The `GPIO_DS_ALT_HIGH` and `GPIO_DS_ALT_LOW` flags are used to set the drive strength of a pin when the output of that pin is High or Low, respectively. As most microcontrollers do not support this type of behavior (per output state drive setting), the initial assumption is that the High and Low correspond to drive strength, not output state, causing a lot of confusion. It also requires non-nordic hardware to use a combination of `(GPIO_DS_ALT_HIGH | GPIO_DS_ALT_LOW)` to set the drive strength.

**Describe the solution you'd like**
New drive strength flags `GPIO_DS_ALT` and `GPIO_DS_DFLT` should be added. These flags should be called out in the documentation as the primary drive strength flags for use in most situations. 

By defining these flags as
`#define GPIO_DS_DFLT		(GPIO_DS_DFLT_LOW | GPIO_DS_DFLT_HIGH)`
`#define GPIO_DS_ALT			(GPIO_DS_ALT_LOW | GPIO_DS_ALT_HIGH)`
the per-output-state functionality of the nordic GPIO driver will be maintained, but the more general-purpose usage of drive strength flags will be available.

**Describe alternatives you've considered**
None.

**Additional context**
When implementing the drive strength option for the cc13xx_cc26xx gpio driver, it was not apparent which combination of flags should be used to set the alternate drive strength. In both my development of the driver feature and in code review of the PR, the High/Low confusion came into play, and is the motivation for this Enhancement request.

pabigot, 2020-11-30:
Added for API review @carlescufi

pabigot, 2020-11-30:
@mnkp FYI, this follows up on discussion #30220.

pabigot, 2021-01-05:
In the API telecon 2020-01-05 we discussed this, and came up with several concerns:

* Some vendors have more than two drive strength configurations.
* It seems unlikely there will be a clear set of common functionality that can be supported cross-vendors with consistent behavior in vendor-agnostic code.  (That being exactly the problem with the existing gpio DS flags).
* Most vendors handle drive strength as a function of pinmux configuration, not GPIO, as it also applies when the signals are used with peripherals like I2C, SPI, and PWM.

This enhancement doesn't identify the platform where the existing functionality is unsatisfactory.

My position at this point is any generic solution to managing drive strength belongs in a pinmux API, not a GPIO API.  It may also be easier to support vendor-specific capabilities than it is with GPIO, e.g. by doing it through driver-specific devicetree properties associated with the controller.

I'd recommend that we instead take steps to deprecating the GPIO configuration of drive strength, and replace it with something in the pinmux API.  Until that happens it would make sense to add vendor-specific API extensions to drivers (pinmux or GPIO) to provide access to capability needed in the short-term.

carlescufi, 2021-01-19:
@statropy could you please respond to the concerns above?

statropy, 2021-01-20:
Evidently alternate drive-strength is not an often used feature since until I found this issue it had only been implemented for one device. So whether or not this is adopted probably isn't that important. But my proposed solution:

- Alleviates confusion from poor naming convention
- Doesn't rely on understanding implementation details of NRF I/O pins
- Doesn't eliminate features/functionality of current NRF implementation
- Perfectly suitable as an interim solution until a pinmux-based design is implemented

If this isn't the right approach, I'm happy to withdraw it.

pabigot, 2021-01-20:
@mnkp @anangl please provide your input.

I'm willing to prototype a common driver-specific API in the Nordic, SX1509, and cc13xx* drivers if we agree on deprecating use of these flags in the GPIO API and aren't going to get pinmux in place anytime soon.  But I'd really prefer to not start adding more complexity to the use of common flags for this purpose.

PR #40150 (2021-11-11): drivers: gpio: add combined drive strength flags and mask
Introduce combined GPIO drive strength flags for GPIO controllers only supporting either default or alternative drive strength regardless if the pin is driven to a high or a low level.

Fixes: #30329
commit 2e8cc9f9 (2021-11-11): drivers: gpio: add combined drive strength flags and mask

Introduce combined GPIO drive strength flags for GPIO controllers only
supporting either default or alternative drive strength regardless if
the pin is driven to a high or a low level.

Fixes: #30329
closed 2021-11-11 by cfriedt