zephyrproject-rtos/zephyr #30344: Bluetooth: host: Add support for multiple advertising sets for legacy advertising

lukasbrchl, 2020-12-01:
Hello,

In April this year, support for multiple advertising sets for Bluetooth 5 AE was added (#24491). However, there is no info on planned support for multiple advertising sets for legacy advertising (Bluetooth 4). Is this somewhere in the Zephyr roadmap? I was unable to find any issue or comments on this feature.

Thanks.

cvinayak, 2020-12-04:
As the PR #24491 states, the feature is fully support in Zephyr host stack. What is the issue?

lukasbrchl, 2020-12-04:
Hello @cvinayak, thank you for your reply. Apologies if I didn't understand the implementation correctly. However, the https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L446 seems it is implemented as a singleton. which means I cannot have more instances of the legacy advertisements. If I try to do so, then it returns NULL. And it also seems that there is no possibility to have control over the legacy advertising sets (e.g. I'm not able to see which one is running and I cannot stop a particular one).

Is my assumption above correct or do we completely missed something?

joerchan, 2020-12-07:
@lukasbrchl  In order have multiple advertising sets you will have to use the new API i.e bt_le_ext_adv_* functions in order to create and manage multiple advertising sets.

The type of advertising is determined by the advertising option `BT_LE_ADV_OPT_EXT_ADV`, if you don't set this option then the advertising set will be a "legacy" advertiser.

The singleton you refer to is for the legacy API i.e bt_le_adv_* functions, which only supports one advertiser and is kept for backwards compatibility.

closed 2020-12-07 by joerchan