zephyrproject-rtos/zephyr #30348: XIP can't be enabled with ARC MWDT toolchain

evgeniy-paltsev, 2020-12-01:
**Describe the bug**
If CONFIG_XIP is enabled we use `ALIGN_WITH_INPUT` option in linker script. However `ALIGN_WITH_INPUT`  isn't supported by ARC MWDT toolchain so we get error on linkage.

**To Reproduce**
Steps to reproduce the behavior:

Enable XIP for nsim_hs:
```
diff --git a/boards/arc/nsim/nsim_hs_defconfig b/boards/arc/nsim/nsim_hs_defconfig
index 3842f44225..7d4540fc4a 100644
--- a/boards/arc/nsim/nsim_hs_defconfig
+++ b/boards/arc/nsim/nsim_hs_defconfig
@@ -5,7 +5,7 @@ CONFIG_SOC_NSIM=y
 CONFIG_SOC_NSIM_HS=y
 CONFIG_BOARD_NSIM=y
 CONFIG_SYS_CLOCK_TICKS_PER_SEC=100
-CONFIG_XIP=n
+CONFIG_XIP=y
 CONFIG_BUILD_OUTPUT_BIN=n
 CONFIG_PRINTK=y
```

Build with ARC MWDT toolchain:

```
export ZEPHYR_TOOLCHAIN_VARIANT=arcmwdt
export ARCMWDT_TOOLCHAIN_PATH=$METAWARE_ROOT/..
. ./zephyr-env.sh
west build -b nsim_hs samples/hello_world
```

**Expected behavior**
No build issues.

**Logs and console output**
```
west build -b nsim_hs samples/hello_world
 [**snip**]
[108/113] Linking C executable zephyr/zephyr_prebuilt.elf
lldac: error: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/linker.cmd:221: symbol not found: ALIGN_WITH_INPUT
lldac: error: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/linker.cmd:236: symbol not found: ALIGN_WITH_INPUT
lldac: error: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/linker.cmd:243: symbol not found: ALIGN_WITH_INPUT
lldac: error: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/linker.cmd:252: symbol not found: ALIGN_WITH_INPUT
lldac: error: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/linker.cmd:259: symbol not found: ALIGN_WITH_INPUT
lldac: error: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/linker.cmd:271: symbol not found: ALIGN_WITH_INPUT
lldac: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: MWDT 2020.09
 - Zephyr commit SHA: 1e674ad815

**Additional context**
Looks like it isn't MWDT quirk but limitation came from LLVM.

hongshui3000, 2021-04-02:
https://github.com/zephyrproject-rtos/zephyr/issues/32563
In fact, this is the same problem as the one I mentioned. I suggest that the LD script should be reviewed and overhauled. So as not to make the script become messy after being larger.

galak, 2021-05-27:
@evgeniy-paltsev is this still an issue?

evgeniy-paltsev, 2021-05-28:
@galak yes, it is.

ruuddw, 2022-04-19:
I raised the priority, as this is an issue for a specific ARC user to move to MetaWare.

jacoba-at, 2022-05-15:
What is the planning for this issue?

ruuddw, 2022-05-23:
I'm not sure if the 'ALIGN_WITH_INPUT' is actually required for the llvm linker, we're checking/discussing this with the Zephyr Toolchain working group and others. I don't have a firm planning, but if the attribute indeed is superfluous for lld a solution could be simple.

evgeniy-paltsev, 2022-09-05:
Hi,

now we have draft implementation of XIP with ARC MWDT toolchain, which can be found in this branch:
https://github.com/foss-for-synopsys-dwc-arc-processors/zephyr/tree/xip_with_mwdt_v2

The implementation was tested with qemu_arc_hs platform which has XIP enabled.

Note that currently it was tested with kernel mode only (without userspace), so userspace testing is disabled in the branch.
Also branch includes some tests disabling (mostly new logging tests) which currently has issues with MWDT - these are issues not related to XIP.

evgeniy-paltsev, 2022-12-04:
Fixed by #52148

closed 2022-12-04 by evgeniy-paltsev