zephyrproject-rtos/zephyr #30349: Memory protection unit fault when running socket CAN program

mejtejt, 2020-12-01:
**Describe the bug**
When running socket CAN code for my board, I get MPU FAULT. It did not happen a month ago, when I last modified by code. Since then, I only updated (by executing git pull) Zephyr directory.  I initialize a single thread from this function:

```
bool FazuaHandler::Initialize() {
  LOG(INFO) << "Initializing fazua handler";

  if (!can_driver::Open(&_socket_id)) {
    LOG(ERROR) << "Failed to open CAN interface";
    return false;
  }

  const std::vector<uint32_t> hmi_messages = {0x600, 0x601};
  if (!can_driver::SubscribeToMessages(_socket_id, hmi_messages)) {
    LOG(ERROR) << "Failed to subscribe to HMI messages";
    return false;
  }

  k_tid_t ret = k_thread_create(&_read_thread, _read_thread_stack,
                                K_THREAD_STACK_SIZEOF(_read_thread_stack),
                                Callback<FazuaHandler>::callback, NULL, NULL,
                                NULL, K_PRIO_PREEMPT(5), 0, K_NO_WAIT);
  if (ret == NULL) {
    LOG(ERROR) << "Failed to create CAN reading thread";
    _running = false;
	  return false;
  }
  LOG(INFO) << "Fazua handler successfully initialized";
  return true;
}
```

The problem seems to occur only when `SubscribeToMessage` function is called, since when this function is not there, no MPU FAULT happens (because this function enables reception of the CAN frames). Function is implemented as following (although I believe it's implementation is not relevant for the problem, since the problem lies deeper in the code). 

```
bool SubscribeToMessages(const uint32_t socket_id,
                         const std::vector<uint32_t> &can_ids) {
  auto message_ids = can_ids;
  std::sort(message_ids.begin(),message_ids.end());
  const auto mask = CreateControllerAreaNetworkMask(message_ids);

  LOG(INFO) << "Setting filter ID: " << std::hex << std::showbase
            << message_ids[0] << ", mask: " << mask;

#pragma GCC diagnostic push      // save the current warning state
#pragma GCC diagnostic ignored "-Wnarrowing" // disable the warning
  const struct zcan_filter zfilter = {
      .id_type = CAN_STANDARD_IDENTIFIER,
      .rtr = CAN_DATAFRAME,
      .std_id = message_ids[0],
      .rtr_mask = 1,
      .std_id_mask = mask};
#pragma GCC diagnostic pop  // restore previous default behaviour

  struct can_filter filter;
  can_copy_zfilter_to_filter(&zfilter, &filter);

  const auto ret = setsockopt(socket_id, SOL_CAN_RAW, CAN_RAW_FILTER, &filter,
                       sizeof(filter));
  if (-1 == ret) {
    LOG(ERROR) << "Failed to subscribe to CAN messages, reason: "
              << strerror(errno);
    return false;
  }

  return true;
}
```

**Impact**
I have no CAN communication with my system and project is at the halt at the moment. 

**Logs and console output**
```
 000151 [INFO] power_sequence.cpp:147:[WaitModuleOk] | All power pins initialized
000159 [INFO] fazua_handler.cpp:76:[Initialize] | Initializing fazua handler
[00:00:00.166,000] <dbg> net_ctx.net_context_bind: (main): Context 0x20001158 binding to 1 iface[1] 0x200005f0
[00:00:00.179,000] <dbg> net_conn.conn_register_debug: (main): [0x200015b8/1/4/0x05] remote -/0 
[00:00:00.190,000] <dbg> net_conn.conn_register_debug: (main):   local ?/0 cb 0x8010071 ud (nil)
000202 [INFO] can.cpp:119:[SubscribeToMessages] | Setting filter ID: 0x600, mask: 0x7fe
[00:00:00.210,000] <dbg> net_sock_can.can_register_filters: (main): Registering 1 filters
[00:00:00.221,000] <dbg> net_sock_can.can_register_receiver: (main): Max 1 receivers
[00:00:00.231,000] <dbg> can_driver.can_stm32_set_filter: Setting filter ID: 0x600, mask: 0x7fe
[00:00:00.242,000] <dbg> can_driver.can_stm32_set_filter: Filter type: standard ID with mask (0)
[00:00:00.254,000] <dbg> can_driver.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
000266 [INFO] fazua_handler.cpp:98:[Initialize] | Fazua handler successfully initialized
000274 [INFO] fazua_handler.cpp:49:[read_thread] | Executing read thread
[00:00:00.335,000] <dbg> can_driver.can_stm32_rx_isr_handler: Message on filter index 0
[00:00:00.346,000] <dbg> can_driver.can_stm32_rx_isr_handler: Message on filter index 0
[00:00:00.356,000] <dbg> net_core.net_recv_data: (): prio 0 iface 0x200005f0 pkt 0x20006180 len 14
[00:00:00.368,000] <err> os: ***** MPU FAULT *****
[00:00:00.376,000] <err> os:   Instruction Access Violation
[00:00:00.384,000] <err> os: r0/a1:  0x20008790  r1/a2:  0x20008790  r2/a3:  0x20004628
[00:00:00.395,000] <err> os: r3/a4:  0x00000004 r12/ip:  0x4f464e49 r14/lr:  0x00000000
[00:00:00.406,000] <err> os:  xpsr:  0x61000000
[00:00:00.413,000] <err> os: Faulting instruction address (r15/pc): 0x4f464e48
[00:00:00.423,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.433,000] <err> os: Current thread: 0x20000890 (rx_q[0])
[00:00:00.442,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 16.04
 - Toolchain: Zephyr SDK version 0.11.4
 - Commit: on the latest Zephyr master commit at the time of writing (aa07f8a drivers: wifi: winc1500: remove unused send_cb and send_user_data)

**Additional context**
My code for the CAN communication has not been modified from my side for over a month, and it used to work then. That is why I believe it could have something to do with the Zephyr repository. I tried increasing stack size of my thread to 2048 (normally I used 1024), but it did not help. Has this problem been reported maybe by someone else already lately, or the problem lies somewhere else (indirectly, in my code)? How do I trace MPU fault?

jukkar, 2020-12-01:
Is the fault address `0x4f464e48` a valid one, at a quick glance it does not look like a valid address?
If it is not a valid address, then there is probably a stack problem somewhere.

mejtejt, 2020-12-01:
I understand there is a stack problem, but I do not understand how I can fix it. I added a few more logs, where it can be seen that read_thread is the one experiencing problems. Also, now it returns BUS FAULT, instead of MPU FAULT.

```
000274 [INFO] main.cpp:30:[main] | main thread id = 0x20000ae8
000280 [INFO] fazua_handler.cpp:49:[read_thread] | Executing read thread
000287 [INFO] fazua_handler.cpp:53:[read_thread] | read_thread id = 0x20004880
[00:00:00.294,000] <err> os: ***** BUS FAULT *****
[00:00:00.302,000] <err> os:   Imprecise data bus error
[00:00:00.310,000] <err> os: r0/a1:  0x200001c0  r1/a2:  0x20002cec  r2/a3:  0x20002d4c
[00:00:00.321,000] <err> os: r3/a4:  0x08040098 r12/ip:  0x080414bc r14/lr:  0x0802392b
[00:00:00.332,000] <err> os:  xpsr:  0x210f0000
[00:00:00.339,000] <err> os: Faulting instruction address (r15/pc): 0x08023976
[00:00:00.349,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.359,000] <err> os: Current thread: 0x20004880 (unknown)
[00:00:00.368,000] <err> os: Halting system
```

Furthermore, now PC is at another location. This is different from the last time, and it seems to me since stack overflow happens, everything messes up. 

Read thread has the following implementation: 
```
void FazuaHandler::read_thread(void *p1, void *p2, void *p3) {
  LOG(INFO) << "Executing read thread";
  int32_t can_id_read;
  uint32_t data_length;
  std::array<uint8_t, can_driver::kMaxDataLength> can_data;
  LOG(INFO) << "read_thread id = " << k_current_get();

  while (_running) {
    if (!can_driver::Read(_socket_id, &can_id_read, &data_length, can_data)) {
      k_msleep(0);
      continue;
    }
    switch (can_id_read) {
      case 0x600:
        can_driver::Write(_socket_id, 0x300, can_data, data_length);
        break;
      case 0x601:
        can_driver::Write(_socket_id, 0x301, can_data, data_length);
        break;
      default:
        LOG(INFO) << "Message " << std::hex << std::showbase << can_id_read
                  << " not defined";
        break;
    }
  }
  LOG(INFO) << "Exiting read thread" << std::endl;
}
```
And it's stack size is set to 1024. 

```
constexpr int kStackSize = 1024;
K_THREAD_STACK_DEFINE(_read_thread_stack, kStackSize);
```

Is there some general stack configuration I am missing? As in prj.conf file?

jukkar, 2020-12-01:
> And it's stack size is set to 1024.

And you tried to increase this size?

You can see all the stack sizes like this
```
$ ack STACK <you app build dir>/zephyr/.config
CONFIG_X86_EXCEPTION_STACK_TRACE=y
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_IDLE_STACK_SIZE=256
CONFIG_ISR_STACK_SIZE=2048
CONFIG_TEST_EXTRA_STACKSIZE=0
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=1024
# CONFIG_HW_STACK_PROTECTION is not set
CONFIG_PRIVILEGED_STACK_SIZE=1024
# CONFIG_STACK_GROWS_UP is not set
CONFIG_ARCH_HAS_STACK_PROTECTION=y
CONFIG_THREAD_STACK_INFO=y
CONFIG_INIT_STACKS=y
# CONFIG_STACK_CANARIES is not set
CONFIG_STACK_POINTER_RANDOM=0
CONFIG_UART_MUX_RX_STACK_SIZE=512
# CONFIG_STACK_USAGE is not set
# CONFIG_STACK_SENTINEL is not set
CONFIG_EXCEPTION_STACK_TRACE=y
CONFIG_LOG_PROCESS_THREAD_STACK_SIZE=768
CONFIG_NET_TX_STACK_SIZE=1200
CONFIG_NET_RX_STACK_SIZE=1500
CONFIG_NET_MGMT_EVENT_STACK_SIZE=768
# CONFIG_NET_DEBUG_MGMT_EVENT_STACK is not set
CONFIG_NET_CONNECTION_MANAGER_STACK_SIZE=512
CONFIG_SHELL_STACK_SIZE=2048
```

mejtejt, 2020-12-01:
Some more info. I will refer to the log shown below:

```
006416 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
006723 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
007030 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
007338 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
007645 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
007952 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
008259 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
008566 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
008873 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
009180 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
009487 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
009794 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
010101 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
010408 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
010716 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
011023 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
011330 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
011637 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
011944 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
012251 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
012558 [ERROR] can.cpp:196:[Read] | Timeout while receiving CAN message
[00:00:12.858,000] <dbg> can_driver.can_stm32_rx_isr_handler: Message on filter index 0
[00:00:12.869,000] <dbg> can_driver.can_stm32_rx_isr_handler: Message on filter index 0
[00:00:12.880,000] <dbg> net_core.net_recv_data: (): prio 0 iface 0x200005f0 pkt 0x20007d80 len 14
[00:00:12.891,000] <err> os: ***** MPU FAULT *****
[00:00:12.899,000] <err> os:   Instruction Access Violation
[00:00:12.907,000] <err> os: r0/a1:  0x2000a390  r1/a2:  0x2000a390  r2/a3:  0x20006228
[00:00:12.918,000] <err> os: r3/a4:  0x00000004 r12/ip:  0x4f464e49 r14/lr:  0x00000000
[00:00:12.929,000] <err> os:  xpsr:  0x61000000
[00:00:12.936,000] <err> os: Faulting instruction address (r15/pc): 0x4f464e48
[00:00:12.946,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:12.956,000] <err> os: Current thread: 0x20000890 (rx_q[0])
[00:00:12.965,000] <err> os: Halting system
```

I have two parts of a system, for the sake of discussion, two CAN nodes. Timeout is written when those two nodes are disconnected from each other. As soon as I connect them, there is a MPU FAULT. This has to be some low level memory corruption or kernel issue. Hopefully, this can be resolved at some point.

mejtejt, 2020-12-01:
> 
> And you tried to increase this size?
> 

Yes, to 2048, 4096 and 8192. I did not cause any change.

jukkar, 2020-12-01:
@mejtejt do you have an application that you could share to replicate the issue?

You said your setup used to work earlier. Are you able to bisect the issue?

cc: @alexanderwachter

alexanderwachter, 2020-12-01:
@mejtejt can you check the fault address with addr2line?
Is the thread running in user mode?

mejtejt, 2020-12-03:
Ok guys, I did some investigation, and I have some interesting results. 

[image]

The above image shows part of the code which is reading messages 0x600 and 0x601,  and forwards them as 0x300 and 0x301. 

I have PEAK analyzer connected to the board. When I comment out the following defines in prj.conf file, everything works fine.
[image]

[image]

[image]

Now, when those "debug" lines in prj.conf file are enabled, there is an error. 

[image]

[image]

This is on the Zephyr side. On my side, there is a problem also, which I am able to avoid, but I do not understand it at the moment. I will write it down, maybe you can tell me is it related to some flash/memory stuff. 

Debug messages can be sent out to UART either through std::cout, or through the wrapper which I will show you later. 

The example usage is like this:

[image]

If line 61 is commented out, and problematic defines in prj.conf file are also commented out, I get normal CAN behavior like the one shown above. However, as soon as uncomment line 61, I get no more communication on CAN. 

[image]

LOG is a macro, which calls a class StreamLogger. 

`#define LOG(severity) logger::StreamLogger{LOG_LEVEL(#severity), FILE_LINE, FUNCTION_NAME}`

StreamLogger.h file:
[image]

StreamLogger.cpp file:
[image]

Macros are defined as follows:
[image]

And their implementation:
[image]

As you can see, it writes out to the port as expected, but CAN thread is not working anymore.
[image]

As I said, I can (temporarily) get around this by using std::cout directly, but I would really like to understand why is this implementation problematic. It seems as it takes some resources/memory, which kernel/other parts of the underlying OS are using. 

Looking forward to your inputs.

alexanderwachter, 2020-12-05:
You cannot use log immediate in an ISR. There is LOG code in the can driver which crashes your system. Either decrease the log level for CAN or use the threaded logging.

mejtejt, 2020-12-07:
@alexanderwachter , I don't understand quite your last response. I will break it up in a sections. First of all, what do you mean by "cannot use log in ISR"? I have no ISR which is using log lines. `read_thread` is (one of) foreground thread(s), created with `k_thread_create`. 

[image]

Secondly, how do I "decrease the log of CAN"? I tried the following configuration:
CONFIG_LOG=y
CONFIG_CAN_LOG_LEVEL_OFF=y

Didn't help. 

I also tried

alexanderwachter, 2020-12-07:
There is a log from the can rx ISR in your screenshot. I don't recommend to use immediate logging. Some other drivers might also do logging from ISRs

mejtejt, 2020-12-14:
I have spent a couple of days trying to really understand the problem, so I am better prepared to understand your answers. The question really boils down to: I cannot have socket CAN working when using std::ostringstream. 

The following code snippets illustrate the problem: 
[image]

As you can see, I have replaced std::cout line 76 with std::ostringstream. 

And inside of the `SubscribeToMessages`, I have done similar thing:
[image]

And as soon as I replace these 2 std::cout's with std::ostringstream, CAN crashes. 

[image]

In prj.conf file, only two defines related to CAN debugging are enabled. 
[image]

Why does CAN/Zephyr not like when I use std::ostringstream instead of std::cout? When I use std::cout, CAN is working without any problems.

mejtejt, 2020-12-28:
I would like to provide an update on this issue. I have been able to resolve it. What made a difference, was in addition to increasing stack size of `_read_thread`, I also had to set the following in prj.conf file: `CONFIG_MAIN_STACK_SIZE=2048`. Now, CAN thread is not crashing any more. But there was no error in Zephyr CAN implementation, it was a coincidence that the stack which was corrupted was the one belonging to the CAN. I will therefore mark this issue as resolved.

closed 2020-12-28 by mejtejt