zephyrproject-rtos/zephyr #30355: Multiple vlan interfaces on same interface not working

rtzu6, 2020-12-01:
**Describe the bug**
This issue was already hinted at in issue #26235, but the author was asked to create a separate issue. I could not find such a followup issue, if I missed it please link it here and close this one.

I want to run two vlan interfaces on the same ethernet interface in addition to the normal interface. To achieve this, I extended the vlan sample with an additional vlan interface, similar to the one that is already present (I extended the code, the KConfig file and the prj.conf accordingly). On my machine, I used the zeth-vlan.conf from the net-tools repo. 
I can ping all interfaces indivually. I can also ping one vlan interface and the normal interface at the same time. But pinging both vlan interfaces at the same time stops working quickly. 
In the previous issue, it was already discovered that the problem shows itself in `ethernet_send` in ethernet.c. It can easily be observed by enabling the debug output and checking the output of `set_vlan_tag`, which discovers that the interface indicated in the packet and the interface given to the function are different. `set_vlan_tag` handles this by overwriting the interface that was passed as parameter with the interface returned by `net_if_ipv6_addr_lookup(&NET_IPV6_HDR(pkt)->src, &target))`. As far as I can tell, the interface returned by `net_if_ipv6_addr_lookup(&NET_IPV6_HDR(pkt)->src, &target))` is the correct one. 
This change is never propagated to `ethernet_send`, which then proceeds with the wrong vlan interface, resulting in an error. 

The interface parameter `iface` is created in `process_tx` in net_if.c:
```
pkt = CONTAINER_OF(work, struct net_pkt, work);
net_pkt_set_tx_stats_tick(pkt, k_cycle_get_32());
iface = net_pkt_iface(pkt);
```
Unfortunately, I am not familiar enough with the inner workings to know where the packet originates from. 

**To Reproduce**
Steps to reproduce the behavior:
1. Extend the vlan sample with an additional vlan interface.
2. Compile and flash it to the board with debug output enabled. 
3. Ping multiple vlan interfaces at the same time. 
4. After a short amount of time, the pings stop working. You can also check the output in a serial terminal: `Iface <iface1> should be <iface2>`

**Expected behavior**
Pings with multiple vlans should work normally.

**Logs and console output**
prj.conf:
```
CONFIG_NETWORKING=y
CONFIG_NET_LOG=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=n
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=n
CONFIG_NET_STATISTICS=y
CONFIG_TEST_RANDOM_GENERATOR=n
CONFIG_NET_PKT_RX_COUNT=32
CONFIG_NET_PKT_TX_COUNT=32
CONFIG_NET_BUF_RX_COUNT=32
CONFIG_NET_BUF_TX_COUNT=32
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=10
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=10
#CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1
CONFIG_NET_MAX_CONTEXTS=20
CONFIG_INIT_STACKS=y
CONFIG_PRINTK=y
CONFIG_NET_SHELL=y
# Ethernet is needed for VLAN
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=n
CONFIG_NET_CONFIG_SETTINGS=y
# First ethernet interface will use these settings
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
# VLAN tag for the first interface
CONFIG_SAMPLE_VLAN_TAG_1=100
CONFIG_SAMPLE_IPV6_ADDR_1="2001:db8:100::1"
CONFIG_LOG_STRDUP_BUF_COUNT=40
# Settings for the second network interface
CONFIG_SAMPLE_IPV6_ADDR_2="2001:db8:200::1"
CONFIG_SAMPLE_VLAN_TAG_2=200
CONFIG_LOG=y
CONFIG_NET_VLAN=y
CONFIG_NET_VLAN_COUNT=3
CONFIG_NET_ROUTING=y
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: (arm-none-eabi-gcc 10.2.0)
 - Version: 2.4.99 (latest master)

**Additional context**
pr ##29865 is from the same author that mentioned the problem in the issue but does not seem to address this problem.

jukkar, 2020-12-01:
I just tested this and it definitely works for me (with echo-server sample app). I tested it like this:

Terminal 1 (net-tools):
```
cd net-tools
./net-setup.sh -c zeth-vlan.conf
```

Terminal 2 (zephyr):
```
cd samples/net/sockets/echo_server
cmake -Bbuild -DBOARD=qemu_x86 . -DOVERLAY_CONFIG="overlay-e1000.conf overlay-vlan.conf"
make -C build run
```

Then in net-shell in terminal 2:
```
uart:~$ net iface

Interface 0x14d8b4 (Ethernet) [1]
===================================
Link addr : 52:54:00:12:34:56
MTU       : 1500
Ethernet capabilities supported:
	Virtual LAN
	10 Mbits
	100 Mbits
	1 Gbits
IPv6 unicast addresses (max 3):
	fe80::5054:ff:fe12:3456 autoconf preferred infinite
	2001:db8::1 manual preferred infinite
IPv6 multicast addresses (max 4):
	ff02::1
	ff02::1:ff12:3456
	ff02::1:ff00:1
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 15012
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
	192.0.2.1 manual preferred infinite
IPv4 multicast addresses (max 1):
	<none>
IPv4 gateway : 0.0.0.0
IPv4 netmask : 255.255.255.0

Interface 0x14d978 (Ethernet) [2]
===================================
Link addr : 52:54:00:12:34:56
MTU       : 1500
VLAN tag  : 100 (0x64)
Ethernet capabilities supported:
	Virtual LAN
	10 Mbits
	100 Mbits
	1 Gbits
IPv6 unicast addresses (max 3):
	fe80::a9b6:7cff:fe3c:8617 autoconf preferred infinite
	2001:db8:100::1 manual preferred infinite
IPv6 multicast addresses (max 4):
	ff02::1
	ff02::1:ff3c:8617
	ff02::1:ff00:1
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 27477
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
	198.51.100.1 manual preferred infinite
IPv4 multicast addresses (max 1):
	<none>
IPv4 gateway : 0.0.0.0
IPv4 netmask : 0.0.0.0

Interface 0x14da3c (Ethernet) [3]
===================================
Link addr : 52:54:00:12:34:56
MTU       : 1500
VLAN tag  : 200 (0xc8)
Ethernet capabilities supported:
	Virtual LAN
	10 Mbits
	100 Mbits
	1 Gbits
IPv6 unicast addresses (max 3):
	fe80::a40c:b4ff:fee8:d64e autoconf preferred infinite
	2001:db8:200::1 manual preferred infinite
IPv6 multicast addresses (max 4):
	ff02::1
	ff02::1:ffe8:d64e
	ff02::1:ff00:1
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 38215
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
	203.0.113.1 manual preferred infinite
IPv4 multicast addresses (max 1):
	<none>
IPv4 gateway : 0.0.0.0
IPv4 netmask : 0.0.0.0
```
So there is 3 interfaces, the master one is the first 1, the other two are the VLAN interfaces.

In Linux side, I am seeing also three interfaces:
```
$ ifconfig
zeth: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.0.2.2  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::200:5eff:fe00:53ff  prefixlen 64  scopeid 0x20<link>
        inet6 2001:db8::2  prefixlen 128  scopeid 0x0<global>
        ether 00:00:5e:00:53:ff  txqueuelen 1000  (Ethernet)
        RX packets 39  bytes 3546 (3.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 57  bytes 5338 (5.2 KiB)
        TX errors 0  dropped 1 overruns 0  carrier 0  collisions 0

vlan.100: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 198.51.100.2  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 2001:db8:100::2  prefixlen 128  scopeid 0x0<global>
        inet6 fe80::200:5eff:fe00:53ff  prefixlen 64  scopeid 0x20<link>
        ether 00:00:5e:00:53:ff  txqueuelen 1000  (Ethernet)
        RX packets 2  bytes 156 (156.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 8  bytes 832 (832.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

vlan.200: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 203.0.113.2  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::200:5eff:fe00:53ff  prefixlen 64  scopeid 0x20<link>
        inet6 2001:db8:200::2  prefixlen 128  scopeid 0x0<global>
        ether 00:00:5e:00:53:ff  txqueuelen 1000  (Ethernet)
        RX packets 2  bytes 156 (156.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 8  bytes 832 (832.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

I can then use echo-client in net-tools
```
$ ./echo-client 203.0.113.1
.......
Average round trip 4 ms
Sent 7 packets
```
here the data will be routed via vlan.200 interface in Linux and interface 3 in zephyr.

jukkar, 2020-12-01:
Can you elaborate more detail what you are doing so I can try to replicate the issue?

rtzu6, 2020-12-03:
Sorry for the late reply.

Here are my exact steps starting from a fresh zephyr setup with the echo_server example.

1. Modify the prj.conf to enable vlan. I do this by copying the config from overlay-vlan.conf to prj.conf. Nothing else in this project was touched.

This is the output of west diff for the prj.conf file:

```
# Generic networking options
 CONFIG_NETWORKING=y
 CONFIG_NET_UDP=y
-CONFIG_NET_TCP=y
+CONFIG_NET_TCP=n
 CONFIG_NET_IPV6=y
-CONFIG_NET_IPV4=y
+CONFIG_NET_IPV4=n
 CONFIG_NET_SOCKETS=y
 CONFIG_NET_SOCKETS_POSIX_NAMES=y
 CONFIG_POSIX_MAX_FDS=6
@@ -20,6 +20,7 @@ CONFIG_NET_LOG=y
 CONFIG_LOG=y
 CONFIG_NET_STATISTICS=y
 CONFIG_PRINTK=y
+CONFIG_NET_SHELL=y
 
 # Network buffers
 CONFIG_NET_PKT_RX_COUNT=16
@@ -34,7 +35,7 @@ CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
 CONFIG_NET_MAX_CONTEXTS=10
 
 # Network shell
CONFIG_SHELL=y
 
 # Network application options and configuration
@@ -42,7 +43,7 @@ CONFIG_NET_CONFIG_SETTINGS=y
 CONFIG_NET_CONFIG_NEED_IPV6=y
 CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
 CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
-CONFIG_NET_CONFIG_NEED_IPV4=y
+CONFIG_NET_CONFIG_NEED_IPV4=n
 CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
 CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
 
@@ -52,3 +53,27 @@ CONFIG_POSIX_MAX_FDS=12
 
 # How many client can connect to echo-server simultaneously
-CONFIG_NET_SAMPLE_NUM_HANDLERS=1
+CONFIG_NET_SAMPLE_NUM_HANDLERS=5
+
+
+CONFIG_NET_VLAN=y
+
+# We have one non-vlan interface and two VLAN interfaces
+CONFIG_NET_VLAN_COUNT=3
+
+# There will be three network interfaces.
+
+# First ethernet interface will use these settings
+CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
+CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
+
+# Second ethernet interface will have these settings
+CONFIG_NET_SAMPLE_IFACE2_MY_IPV6_ADDR="2001:db8:100::1"
+# TEST-NET-2 from RFC 5737
+# VLAN tag for the second interface
+CONFIG_NET_SAMPLE_IFACE2_VLAN_TAG=100
+
+# Settings for the third network interface
+CONFIG_NET_SAMPLE_IFACE3_MY_IPV6_ADDR="2001:db8:200::1"
[... 214 lines cut ...]
IPv6 ND recv   129      sent    16      drop    124
IPv6 MLD recv  0        sent    3       drop    0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
UDP recv       0        sent    0       drop    0
UDP chkerr     0
Bytes received 16468
Bytes sent     5986
Processing err 2

Interface 0x20024f08 (Ethernet) [3]
===================================
IPv6 recv      128      sent    115     drop    41      forwarded       0
IPv6 ND recv   42       sent    28      drop    40
IPv6 MLD recv  0        sent    3       drop    0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
UDP recv       0        sent    0       drop    0
UDP chkerr     0
Bytes received 14264
Bytes sent     10296
Processing err 0
```

8. Enable debug output and repeat 7. to reproduce the `Iface <iface1> should be <iface2>` debug output line on the serial console.

jukkar, 2020-12-17:
Retried this with the same nucleo board.

```
cd samples/net/sockets/echo_server
cmake -Bbuild . -DBOARD=nucleo_f767zi -DOVERLAY_CONFIG=overlay-vlan.conf
make -C build flash
```

Setting up the interfaces:
```
cd net-tools
sudo bash zeth-vlan.conf enp0s20u2u4u2
```

pings work ok
```
$ ping 192.0.2.1
PING 192.0.2.1 (192.0.2.1) 56(84) bytes of data.
64 bytes from 192.0.2.1: icmp_seq=1 ttl=64 time=0.585 ms
64 bytes from 192.0.2.1: icmp_seq=2 ttl=64 time=0.339 ms
^C
--- 192.0.2.1 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1031ms
rtt min/avg/max/mdev = 0.339/0.462/0.585/0.123 ms
$ ping 2001:db8::1
PING 2001:db8::1(2001:db8::1) 56 data bytes
64 bytes from 2001:db8::1: icmp_seq=1 ttl=64 time=0.791 ms
64 bytes from 2001:db8::1: icmp_seq=2 ttl=64 time=0.423 ms
^C
--- 2001:db8::1 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1059ms
rtt min/avg/max/mdev = 0.423/0.607/0.791/0.184 ms
$ ping 2001:db8:100::1
PING 2001:db8:100::1(2001:db8:100::1) 56 data bytes
64 bytes from 2001:db8:100::1: icmp_seq=1 ttl=64 time=0.965 ms
64 bytes from 2001:db8:100::1: icmp_seq=2 ttl=64 time=0.371 ms
^C
--- 2001:db8:100::1 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 0.371/0.668/0.965/0.297 ms
$ ping 2001:db8:200::1
PING 2001:db8:200::1(2001:db8:200::1) 56 data bytes
64 bytes from 2001:db8:200::1: icmp_seq=1 ttl=64 time=1060 ms
64 bytes from 2001:db8:200::1: icmp_seq=2 ttl=64 time=1.19 ms
64 bytes from 2001:db8:200::1: icmp_seq=3 ttl=64 time=0.363 ms
^C
--- 2001:db8:200::1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2060ms
rtt min/avg/max/mdev = 0.363/353.826/1059.926/499.288 ms, pipe 2
$ ./echo-client 2001:db8::1
.......
Average round trip 648 us
Sent 7 packets
```

Then some UDP traffic
```
$ ./echo-client 2001:db8:100::1
.......
Average round trip 657 us
Sent 7 packets
$ ./echo-client 2001:db8:200::1
.......
Average round trip 697 us
Sent 7 packets
$ ./echo-client 203.0.113.1
.......
Average round trip 650 us
Sent 7 packets
$ ./echo-client 198.51.100.1
.......
Average round trip 592 us
Sent 7 packets
```

I had some issues with my firewall settings, so I disabled firewall for the vlan.100, vlan.200 and the main interface.

rtzu6, 2020-12-21:
Interesting. Just to confirm: you ran the pings/UDP traffic in parallel I assume?
The issue does not necessarily happen immediately. Maybe you can run the described test setup for a longer period of time, e.g. 10 minutes. 
Also: what connections are shown in the statistics and why is there a large amount of dropped packets?

jukkar, 2021-02-10:
Hi @rtzu6 sorry for very late reply. This issue was just lost to a sea of other issues. I tried this again, and noticed one issue in the echo-server sample application for IPv4 connections. The sample app did not set the netmask of the VLAN network interfaces, which then meant that the network packet could be sent to wrong virtual interface. For example, I was pinging 203.0.113.2 from zephyr, the packet should go to third interface but the second interface (src address 198.51.100.1) was selected instead. This could lead to packets to be dropped unexpectedly or some other connection failures. I have a fix for the echo-server sample at #32163.

rtzu6, 2021-02-11:
Thanks for looking into this and for the fix. Does this also affect IPv6? I am testing with IPv6 and it had the same issues, but as far as I can tell this is not affected by your modification.

jukkar, 2021-02-14:
> Does this also affect IPv6?

No, my fix was only for IPv4.

I tried again with `nucleo_f767zi` with echo-server sample (without any modifications) and with this config
```
cmake . -Bbuild -DBOARD=nucleo_f767zi -DCONFIG_LOG_BUFFER_SIZE=65536 -DCONFIG_LOG_STRDUP_BUF_COUNT=400 -DCONFIG_NET_DEBUG_NET_PKT_ALLOC=y -DCONFIG_NET_BUF_POOL_USAGE=y DCONFIG_NET_PKT_RX_COUNT=100 -DCONFIG_NET_PKT_TX_COUNT=100 -DCONFIG_NET_BUF_RX_COUNT=640 -DCONFIG_NET_BUF_TX_COUNT=640 -DOVERLAY_CONFIG=overlay-vlan.conf
```
In Linux side, I had 6 terminals running with these commands
```
ping 2001:db8::1
ping 2001:db8:100::1
ping 2001:db8:200::1
```
and
```
./echo-client -e -r 2001:db8::1
./echo-client -e -r 2001:db8:100::1
./echo-client -e -r 2001:db8:200::1
```
Zephyr net-shell becomes unresponsive with this load but I did not see excessive dropped packets. This load is too much for this board.

Also it helps if one disables the output from echo-server using this command
```
log disable net_echo_server_sample
```

Please note that if you have multiple terminals sending data to echo-server, you need to increase the amount of net_pkt and net_buf used. So these options `CONFIG_NET_PKT_RX_COUNT`, `CONFIG_NET_PKT_TX_COUNT`, `CONFIG_NET_BUF_RX_COUNT` and `CONFIG_NET_BUF_TX_COUNT` need to be increased from the default values.

rtzu6, 2021-02-17:
Interesting. I modified the config to your recommendations
```
CONFIG_LOG_BUFFER_SIZE=65536
CONFIG_LOG_STRDUP_BUF_COUNT=400
CONFIG_NET_DEBUG_NET_PKT_ALLOC=y
CONFIG_NET_BUF_POOL_USAGE=y 
CONFIG_NET_PKT_RX_COUNT=100 
CONFIG_NET_PKT_TX_COUNT=100 
CONFIG_NET_BUF_RX_COUNT=640 
CONFIG_NET_BUF_TX_COUNT=640 
```
but I still have the same problem. Also, the shell it not unresponsive for me and I still have the high number of dropped packets.
```
Interface 0x20022684 (Ethernet) [1]
===================================
IPv6 recv      64       sent    71      drop    1       forwarded       0
IPv6 ND recv   7        sent    16      drop    0
IPv6 MLD recv  0        sent    3       drop    0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
UDP recv       0        sent    0       drop    0
UDP chkerr     0
TCP bytes recv 0        sent    0       resent  0
TCP seg recv   0        sent    0       drop    0
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0
TCP conn drop  0        connrst 0
TCP pkt drop   0
Bytes received 7320
Bytes sent     7886
Processing err 0

Interface 0x20022734 (Ethernet) [2]
===================================
IPv6 recv      63       sent    16      drop    54      forwarded       0
IPv6 ND recv   56       sent    9       drop    53
IPv6 MLD recv  0        sent    3       drop    0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
UDP recv       0        sent    0       drop    0
UDP chkerr     0
TCP bytes recv 0        sent    0       resent  0
TCP seg recv   0        sent    0       drop    0
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0
TCP conn drop  0        connrst 0
TCP pkt drop   0
Bytes received 5886
Bytes sent     1664
Processing err 0

Interface 0x200227e4 (Ethernet) [3]
===================================
IPv6 recv      55       sent    33      drop    27      forwarded       0
IPv6 ND recv   24       sent    2       drop    24
IPv6 MLD recv  0        sent    3       drop    0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
UDP recv       0        sent    0       drop    0
UDP chkerr     0
TCP bytes recv 0        sent    0       resent  0
TCP seg recv   0        sent    0       drop    0
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0
TCP conn drop  0        connrst 0
TCP pkt drop   0
Bytes received 5838
Bytes sent     3898
Processing err 0
```

Would you mind sharing you autoconf.h just to confirm that we have the same config?

jukkar, 2021-02-17:
> Would you mind sharing you autoconf.h just to confirm that we have the same config?

Sure, although if using same Kconfig options, board and same revision, we should end up with identical compilation.
I am at commit 8f8cdc18c4a7af5b833bc3e5b630bbe527a942b7 in master

[autoconf.h.gz](https://github.com/zephyrproject-rtos/zephyr/files/5997480/autoconf.h.gz)

rtzu6, 2021-02-23:
Strange. 
I also tried a complete rebuild, zephyr is at commit 1ff7a843.

Scenario 1:
I built the echo_server example via west without touching anything code-wise. Autoconf: [autoconf_west.h.gz](https://github.com/zephyrproject-rtos/zephyr/files/6029135/autoconf_west.h.gz)
A bit different. This results in the same error as always: pinging two different vlan interfaces in parallel as described above does not work.

Scenario 2:
I built the echo_server example with the exact cmake command from above: 
`cmake . -Bbuild -DBOARD=nucleo_f767zi -DCONFIG_LOG_BUFFER_SIZE=65536 -DCONFIG_LOG_STRDUP_BUF_COUNT=400 -DCONFIG_NET_DEBUG_NET_PKT_ALLOC=y -DCONFIG_NET_BUF_POOL_USAGE=y -DCONFIG_NET_PKT_RX_COUNT=100 -DCONFIG_NET_PKT_TX_COUNT=100 -DCONFIG_NET_BUF_RX_COUNT=640 -DCONFIG_NET_BUF_TX_COUNT=640 -DOVERLAY_CONFIG=overlay-vlan.conf`
Autoconf: [autoconf_cmake.h.gz](https://github.com/zephyrproject-rtos/zephyr/files/6029151/autoconf_cmake.h.gz)
Diff: 
```
35d34
< #define CONFIG_ZEPHYR_TRUSTED_FIRMWARE_M_MODULE 1
463d461
< #define CONFIG_SRAM_OFFSET 0x0
```
Same issue: the pings dont work reliably after a certain amount of time.

Scenario 3:
I used the same setup as in Scenario 2 and waited until at least one got "stuck". Then I sent data via the echo client:
```
./echo-client -e -r 2001:db8::1
./echo-client -e -r 2001:db8:100::1
./echo-client -e -r 2001:db8:200::1
```
(Every command in a different terminal). The interface that got "stuck" with the ping also did not get a response from the server.

Scenario 4: 
Same setup as in Scenario 3, but I sent the data via the echo client at a point in time where all three pings worked (they work only for a few seconds in parallel, but I started the clients in this timeframe). This resulted in the scenario that you described: all three clients get a response albeit a bit slow (the board is under heavy load, as you said). The pings continue to work, at least for some amount of time (after around 5 minutes, one echo-client did not get responses anymore and the ping stopped working. I don't know if this always happens).

How do these scenarios behave for you?

jukkar, 2021-03-03:
Perhaps this is not related to anything you are seeing, but I noticed that if we have multiple network interfaces, then the network packet might be sent to wrong network interface (always to first interface even if it was meant to be sent to second or third one) in some rare cases.
This issue should be fixed in #32783

rtzu6, 2021-03-10:
I tried the same setup (3 parallel pings) on 7567f232 and the result is still the same. Either immediately or after a short while, one of the vlan pings stops working.

closed 2021-05-24 by github-actions[bot]