zephyrproject-rtos/zephyr #30356: DAC header file not included in stm32 soc.h

UberDoefus, 2020-12-01:
**Describe the bug**
When building for board 'nucleo_h743zi' and with enabled DAC in the prj.conf:

CONFIG_DAC=y
CONFIG_DAC_STM32=y

the compiler fails to build 'zephyrproject/zephyr/drivers/dac/dac_stm32.c' due to missing defines and functions.
The reason for this failure is the missing header file inclusion in the zephyrproject/zephyr/soc/arm/st_stm32/stm32h7/soc.h

#ifdef CONFIG_DAC_STM32
#include <stm32h7xx_ll_dac.h>
#endif /* CONFIG_DAC_STM32 */

**Expected behavior**
No compiler errors

**Impact**
This blocks using the DAC

**Logs and console output**
zephyrproject/zephyr/drivers/dac/dac_stm32.c:33:18: error: 'LL_DAC_CHANNEL_1' undeclared here (not in a function); did you mean 'LL_ADC_CHANNEL_1'?
   33 | #define CHAN(n)  LL_DAC_CHANNEL_##n
      |                  ^~~~~~~~~~~~~~~
zephyrproject/zephyr/drivers/dac/dac_stm32.c:33:18: note: in definition of macro 'CHAN'
   33 | #define CHAN(n)  LL_DAC_CHANNEL_##n
      |                  ^~~~~~~~~~~~~~~
zephyrproject/zephyr/drivers/dac/dac_stm32.c: In function 'dac_stm32_write_value':
zephyrproject/zephyr/drivers/dac/dac_stm32.c:66:3: warning: implicit declaration of function 'LL_DAC_ConvertData8RightAligned' [-Wimplicit-function-declaration]
   66 |   LL_DAC_ConvertData8RightAligned(cfg->base,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyrproject/zephyr/drivers/dac/dac_stm32.c:69:3: warning: implicit declaration of function 'LL_DAC_ConvertData12RightAligned' [-Wimplicit-function-declaration]
   69 |   LL_DAC_ConvertData12RightAligned(cfg->base,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyrproject/zephyr/drivers/dac/dac_stm32.c: In function 'dac_stm32_channel_setup':
zephyrproject/zephyr/drivers/dac/dac_stm32.c:98:2: warning: implicit declaration of function 'LL_DAC_SetOutputBuffer' [-Wimplicit-function-declaration]
   98 |  LL_DAC_SetOutputBuffer(cfg->base,
      |  ^~~~~~~~~~~~~~~~~~~~~~
zephyrproject/zephyr/drivers/dac/dac_stm32.c:100:3: error: 'LL_DAC_OUTPUT_BUFFER_ENABLE' undeclared (first use in this function); did you mean 'DAC_OUTPUTBUFFER_ENABLE'?
  100 |   LL_DAC_OUTPUT_BUFFER_ENABLE);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   DAC_OUTPUTBUFFER_ENABLE
zephyrproject/zephyr/drivers/dac/dac_stm32.c:100:3: note: each undeclared identifier is reported only once for each function it appears in
zephyrproject/zephyr/drivers/dac/dac_stm32.c:102:2: warning: implicit declaration of function 'LL_DAC_Enable'; did you mean 'LL_ADC_Enable'? [-Wimplicit-function-declaration]
  102 |  LL_DAC_Enable(cfg->base,
      |  ^~~~~~~~~~~~~
      |  LL_ADC_Enable
[106/116] Building CXX object CMakeFiles/app.dir/applications/poc/ioc/src/main.cc.obj

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04
 - Commit: 1ea7bf0db026617a52ea7cf5c037c8c932584cb1

erwango, 2020-12-01:
@UberDoefus can you run `west update` ?

UberDoefus, 2020-12-01:
@erwango Just did a git fetch && git pull. I get a lot of changes, compiler error changed to

```
/home/jeroen/zephyrproject/zephyr/soc/arm/st_stm32/stm32h7/soc_m7.c:16:10: fatal error: stm32_ll_bus.h: No such file or directory
   16 | #include <stm32_ll_bus.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
```

UberDoefus, 2020-12-01:
Ok, never mind. Mistake by me. Indeed solved after commit 1ea7bf0

closed 2020-12-01 by UberDoefus