zephyrproject-rtos/zephyr #30365: TCP2 does not implement Nagle algorithm

jukkar, 2020-12-01:
**Is your enhancement proposal related to a problem? Please describe.**
Currently TCP Nagle algorithm is not implemented when sending data. https://en.wikipedia.org/wiki/Nagle%27s_algorithm

Implementing Nagle would improve network utilization as it collects smaller TCP segments into larger ones and then send one larger segment instead of multiple smaller ones.

lowlander, 2020-12-11:
Would be good if this can be disabled/enabled at runtime on a per socket basis.

jimparis, 2021-04-02:
Lack of Nagle and the flood of small packets is causing problems in my application at the moment, both in terms of performance and running out of net buffers.  Here is my naive attempt at implementing it:

```diff
diff --git a/subsys/net/ip/tcp2.c b/subsys/net/ip/tcp2.c
index 7aa5c3988cda..c6cf60c9a8fe 100644
--- a/subsys/net/ip/tcp2.c
+++ b/subsys/net/ip/tcp2.c
@@ -962,13 +962,22 @@ static int tcp_send_queued_data(struct tcp *conn)
                goto out;
        }
 
-       while (tcp_unsent_len(conn) > 0) {
+       int unsent;
+       int mss = conn_mss(conn);
+       while ((unsent = tcp_unsent_len(conn)) > 0) {
 
                if (tcp_window_full(conn)) {
                        subscribe = true;
                        break;
                }
 
+               /* Nagle's algorithm: don't send new TCP data yet if we still
+                  have unacked data, and the total unsent length is less than
+                  the MSS */
+               if (conn->unacked_len > 0 && unsent < mss) {
+                       break;
+               }
+
                ret = tcp_send_data(conn);
                if (ret < 0) {
                        break;
```

Agreed on having a way to disable it.  The normal way to disable Nagle on a socket would be `setsockopt(sock, SOL_TCP, TCP_NODELAY, ...)`, so I'd vote for that.

jukkar, 2021-04-06:
Looks reasonable, but one can probably only see if it really works as expected after running the code. Can you send a PR for this + supporting the setsockopt too?

Agreed on TCP_NODELAY option, we should have that supported from day one.

There is also some MSS work going on in #32076

ssharks, 2022-06-06:
On top of the fixes of PR https://github.com/zephyrproject-rtos/zephyr/pull/45890, I build an implementation of Nagle's algorithm.

It includes the TCP_NODELAY option to switch Nagle's algorithm off.
The full branch:
https://github.com/ssharks/zephyr/pull/new/net/tcp_nagle

The specific commit:
https://github.com/ssharks/zephyr/commit/a6863fd1f761d752b80d43a0869664a26bfc9866

Please have a good look at it before I submit an official PR. There is quite some new code to make options on TCP specific possible.

ssharks, 2022-06-07:
After review comment from @rlubos, a new commit at https://github.com/ssharks/zephyr/commit/8ab3c5a8fefb5580af1f505cd390528cedb40ef7

ssharks, 2022-06-07:
Now an official PR at https://github.com/zephyrproject-rtos/zephyr/pull/46348

closed 2022-06-09 by carlescufi