zephyrproject-rtos/zephyr #30366: TCP2 does not implement delayed ACKs

jukkar, 2020-12-01:
**Is your enhancement proposal related to a problem? Please describe.**
We do not implement delayed ACK mechanishm. https://en.wikipedia.org/wiki/TCP_delayed_acknowledgment

Implementing delayed ACKs would improve network utilization as we would be able to send fewer ACK packets to peer by collecting more data before sending a single ACK instead of multiple ACKs.

kartben, 2023-09-19:
related: #46584

rlubos, 2025-09-29:
I think the issue can be closed as resolved, we have a delayed ACK mechanism implemented already:
https://github.com/zephyrproject-rtos/zephyr/blob/06daf0e29ee1d7cfd8242f45f2ad04f2fd2b03b9/subsys/net/ip/tcp.c#L2828-L2837
Specifically added in https://github.com/zephyrproject-rtos/zephyr/pull/75014

closed 2025-09-29 by rlubos