zephyrproject-rtos/zephyr #30378: bleutooth: controller: tx buffer overflow error

hardik-proxy, 2020-12-02:
**Describe the bug**
I have multiple central devices connecting to a single peripheral asynchronously. The central device writes to one characteristic (around 40 bytes) and reads from another characteristic (around 170 bytes). After few hours of run, the peripheral device runs into the error. The error occurs when the central device tries to read the characteristic.

```
bt_ctlr_hci: Tx Buffer Overflow
bt_hci_core: Data buffer overflow (link type 0x01)
bt_conn: Unable to send to driver (err -55)
```

**To Reproduce**
I've tried developing standalone peripheral and central app to reproduce the issue. But I am unable to do so. I still run into errors on my production code.

**Expected behavior**
the central device should be able to read the characteristic without tx buffer overflow error.

**Impact**
impact on upcoming delivery of the project

**Logs and console output**
I've tried to add values of free buffer count into an array and print the array on the error. 
something like this ```tx_mem_free_arr_acq[i_acq].cnt = mem_free_count_get(mem_conn_tx.free);``` in functions ```void *ll_tx_mem_acquire(void)```  and ```void ll_tx_mem_release(void *tx)```

the result is as below:
arr_acp is count in mem_acquire and arr_free is count in mem_release
[image]

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
 - Version: Zephyr 2.3.0, b8c78e254ff875680e99c9f131fbe285c4575927

**Additional context**

the above screenshot is with these config options related to tx buffer. 

```
CONFIG_BT_CTLR_TX_BUFFERS=6
CONFIG_BT_CTLR_TX_BUFFER_SIZE=251
CONFIG_BT_L2CAP_TX_BUF_COUNT=6
```

cvinayak, 2020-12-02:
@hardik-proxy Please attached the diff file as to the changes you have done with reference to the printing

hardik-proxy, 2020-12-02:
Here it is:

[git_diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/5627234/git_diff.txt)

cvinayak, 2020-12-02:
From the `arr_acp` values, the value 0 returned by `mem_free_count_get` should cause the `ll_tx_mem_acquire` to return NULL, and subsequently lead to `bt_ctlr_hci: Tx Buffer Overflow`, correct me that is the case? 
Why do I see another `0` in `arr_acp` in your logging, if there was already a tx buffer overflow?
Also, `CONFIG_BT_CTLR_TX_BUFFERS=6` hence your `arr_acp` should show numbers from 6 to 1.

hardik-proxy, 2020-12-02:
during testing, what I've observed is,
--> tx buffer overflow occurred
--> again buffer were available (not able to understand why it is the case)
--> again application worked as expected
--> again tx buffer overflow occurred
--> the cycle continues few times and then system hangs. Soft reset is the only option to get out of it.

secondly, I even tried to set CONFIG_BT_CTLR_TX_BUFFERS=12 and the result was almost similar with values in the range of 3 to 0

for the last question, the central disconnects as soon as it reads the characteristic value. And as long as central devices and peripherals are in the range, flow is something like this:
the central device --> connect to peripheral --> write to a characteristic --> read a characteristic --> disconnect.

cvinayak, 2020-12-03:
> I even tried to set CONFIG_BT_CTLR_TX_BUFFERS=12 and the result was almost similar with values in the range of 3 to 0

This indicates loss of tx buffer count very early in your application. Please check if your application has sufficient thread call stack allocated, use the THREAD_ANALYZER  feature to get the runtime usage details. Try to keep a free bytes margin of >128 bytes when subjected to worst case scenarios of possible nested interrupts of priority levels.

> I've tried developing standalone peripheral and central app to reproduce the issue. But I am unable to do so. I still run into errors on my production code.

You will need to make a small reproducible sample, so that issue, if any, can be analysed.

Experiments that can be done:
- Use nRF sniffer to compare on air traffic between your application and the simple test application to compare if the usecase behavior is similar, in terms of procedures, values/length, and timings
- Experiment with single device, connection and reconnection
- Experiment with multiple devices one after the other, connection and reconnection
- Experiment with multiple devices simultaneously, connection and reconnection

hardik-proxy, 2020-12-04:
**As an update:**

> looks like there's enough stack available when error occurs.

[image]

> This indicates loss of tx buffer count very early in your application.

the error occurs after around 4 hours of correct behaviour. curious how it runs for few hours before failing.

> Experiment with single device, connection and reconnection

no issue with one/two/three peripheral and one central device
if more than three central are running simultaneously issue occurs.

> You will need to make a small reproducible sample, so that issue, if any, can be analysed.

yeah, sure.

hardik-proxy, 2020-12-08:
**Update:**
the issue doesn't occur if I remove GATT read operation of around 170 bytes, which is not I want but just to see where is the problem.

carlescufi, 2020-12-10:
>Version: Zephyr 2.3.0, b8c78e2

Is there a chance you could try to reproduce this on the current master @hardik-proxy?
This might well be an issue in the Host, but without being able to reproduce it consistently it is hard to pinpoint exactly where.

carlescufi, 2020-12-10:
>I have multiple central devices connecting to a single peripheral asynchronously.

Are the centrals all connected at the same time to the single peripheral, or does this happens sequentially?

hardik-proxy, 2020-12-11:
@carlescufi 
> Is there a chance you could try to reproduce this on the current master @hardik-proxy?
Yes, I was able to reproduce it on latest master.

> without being able to reproduce it consistently it is hard to pinpoint exactly where
I understand. I am trying to reproduce it but with minimal GATT read and write operations I am not able to do so. Our application is more than just those operations. So trying to figure out the best minimal way to reproduce it.

> Are the centrals all connected at the same time to the single peripheral, or does this happens sequentially?
it's like if Centrals are within given RSSI range of the peripheral, all of them will try to connect and do the required GATT operations. So if at a time five Centrals are in the range, all five will initiate connection and GATT operations afterwards and disconnection once GATT operations are completed.

hardik-proxy, 2020-12-17:
So I tried to perform GATT write on a characteristic of a Central device, instead of the central reading the characteristic.
The tx buffer overflow error along with few more errors started appearing very early in the application. Logs attached here.
[more_logs.txt](https://github.com/zephyrproject-rtos/zephyr/files/5706258/more_logs.txt)

cvinayak, 2020-12-22:
Your log starts after ~28 minutes of operations, without a sample to reproduce the issue, it is difficult to provide an analysis.

It also appears that asserts do not halt the CPU, have you changed the default assert implementation?
Also, use CONFIG_ASSERT=y, try to halt at exceptions and backtrace the function call tree.

vrahane, 2020-12-28:
Hi @cvinayak, so, yes the asserts are disabled. Probably due to code size. We did play around a bit with the settings and did some investigation, I tried it with 3 peripherals and 1 central, I have come to a point where I understand that there are 4 connections getting initiated trying to securely talk to our central whereas only 3 devices are advertising and they get disconnected if they do not "securely pair" with the central. I say securely pair because this is our proprietary protocol and I do not want to go into the details.
After bumping up pool sizes, disabling certain functionality among others things I have come to a point where I see the following:
```
(gdb) p tx_mem_free_arr_free
$8 = {{cnt = 17 '\021', now = 161252720}, {cnt = 17 '\021', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 17 '\021', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 18 '\022', now = 161287981}, {cnt = 18 '\022', now = 161287981}, {
    cnt = 18 '\022', now = 161287981}, {cnt = 17 '\021', now = 161287981}, {cnt = 18 '\022', now = 161287981}, {cnt = 18 '\022', now = 161287981}, {cnt = 18 '\022', now = 161287981}, {cnt = 17 '\021', now = 157421298}, {cnt = 18 '\022', now = 157421298}, {
    cnt = 18 '\022', now = 157617909}, {cnt = 18 '\022', now = 157683446}, {cnt = 18 '\022', now = 157683446}, {cnt = 17 '\021', now = 157683446}, {cnt = 18 '\022', now = 157683446}, {cnt = 18 '\022', now = 157683446}, {cnt = 18 '\022', now = 157683446}, {
    cnt = 17 '\021', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 17 '\021', now = 161252720}, {cnt = 17 '\021', now = 161252720}, {cnt = 17 '\021', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 15 '\017', now = 161252720}, {
    cnt = 16 '\020', now = 161252720}, {cnt = 17 '\021', now = 161252720}}
(gdb) p tx_mem_free_arr_acq
$9 = {{cnt = 16 '\020', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 19 '\023', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 19 '\023', now = 161287981}, {cnt = 19 '\023', now = 161287981}, {
    cnt = 19 '\023', now = 161287981}, {cnt = 19 '\023', now = 161287981}, {cnt = 18 '\022', now = 161287981}, {cnt = 19 '\023', now = 161287981}, {cnt = 19 '\023', now = 161287981}, {cnt = 18 '\022', now = 157421298}, {cnt = 18 '\022', now = 157421298}, {
    cnt = 19 '\023', now = 157617909}, {cnt = 19 '\023', now = 157683446}, {cnt = 19 '\023', now = 157683446}, {cnt = 19 '\023', now = 157683446}, {cnt = 18 '\022', now = 157683446}, {cnt = 19 '\023', now = 157683446}, {cnt = 19 '\023', now = 157683446}, {
    cnt = 19 '\023', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 19 '\023', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 18 '\022', now = 161252720}, {cnt = 19 '\023', now = 161252720}, {
    cnt = 18 '\022', now = 161252720}, {cnt = 17 '\021', now = 161252720}}
(gdb) p tx_buf_count
$10 = {1, 1, 1, 1, 0, 0}
(gdb) bt
#0  0x000321c8 in arch_irq_lock () at /Users/vipul/dev/fw-signal/repos/zephyr/include/arch/arm/aarch32/asm_inline_gcc.h:56
#1  arch_system_halt (reason=reason@entry=3) at /Users/vipul/dev/fw-signal/repos/zephyr/kernel/fatal.c:28
#2  0x000285f2 in k_sys_fatal_error_handler (reason=reason@entry=3, esf=esf@entry=0x2000be98 <z_interrupt_stacks+960>) at /Users/vipul/dev/fw-signal/repos/zephyr/kernel/fatal.c:43
#3  0x000286ba in z_fatal_error (reason=3, esf=0x2000be98 <z_interrupt_stacks+960>) at /Users/vipul/dev/fw-signal/repos/zephyr/kernel/fatal.c:123
#4  0x000110c2 in _oops () at /Users/vipul/dev/fw-signal/repos/zephyr/arch/arm/core/aarch32/swap_helper.S:465
#5  <signal handler called>
#6  0x00025024 in isr_done (param=0x200042cc <conn_pool+948>) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:613
#7  0x00023f82 in radio_nrf5_isr_body () at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:76
#8  radio_nrf5_isr () at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:72
#9  <signal handler called>
#10 0x16323770 in ?? ()
#11 0x8a5296e2 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) p (struct lll_conn *)param
No symbol "param" in current context.
(gdb) fr 6
#6  0x00025024 in isr_done (param=0x200042cc <conn_pool+948>) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:613
613		LL_ASSERT(e);
(gdb) p (struct lll_conn *)param
$11 = (struct lll_conn *) 0x200042cc <conn_pool+948>
(gdb) p *(struct lll_conn *)param
$12 = {hdr = {parent = 0x200042b0 <conn_pool+920>, is_stop = 0 '\000'}, access_addr = "\345\244\326N", crc_init = <incomplete sequence \364\107\130>, handle = 2, interval = 24, latency = 1, latency_prepare = 0, latency_event = 0, event_counter = 3, 
  data_chan_map = "\377\377\377\377\037", data_chan_count = 37 '%', data_chan_sel = 0 '\000', role = 1 '\001', {{data_chan_hop = 15 '\017', data_chan_use = 8 '\b'}, data_chan_id = 2063}, slave = {latency_enabled = 0 '\000', window_widening_periodic_us = 3, 
    window_widening_max_us = 14850, window_widening_prepare_us = 0, window_widening_event_us = 3, window_size_prepare_us = 0, window_size_event_us = 0}, max_tx_octets = 27, max_rx_octets = 27, max_tx_time = 328, max_rx_time = 328, phy_tx = 1 '\001', 
[... cut ...]
 u16_t mem_free_count_get(void *mem_head);
 void *mem_get(void *mem_pool, u16_t mem_size, u16_t index);
 u16_t mem_index_get(void *mem, void *mem_pool, u16_t mem_size);
 ```

So, I have couple of questions:
1. Is the above workaround/solution a problem in anyway ? 
2. Why am I seeing the assert I am seeing, any insights on that ? The FIFO gets enqueued with events which the ISR lookups and processes. In this case the FIFO seems to be empty. Any suggestions on debugging this scenario.

Thank you for your help.

cvinayak, 2021-01-02:
> yes the asserts are disabled

If there has been any assert check failures that does not halt the CPU, then there is possibility of memory corruption or incorrect state in the subsystem. This will make it difficult to analyse any bugs.

> I have divided the conn_tx.free buffer pool into 3 max allowable buffers per ACL connection. So even if one connection freezes up others operate just fine as opposed to earlier where all of them where freezing up.

As you are making changes to the Bluetooth controller design, deviating from the standard HCI rules of using the Number of Completed Packets and HCI Read Buffer Size counts, will impact the Zephyr Host subsystem as well, causing it to be out of sync with the buffers counts maintained between host and controller.

> To Reproduce
I've tried developing standalone peripheral and central app to reproduce the issue. But I am unable to do so. I still run into errors on my production code.

If you are making changes to host and/or controller subsystem, please send a pull request so that the design changes can be formally reviewed. You will need to include a sample or a test covering the changes to subsystems.

I suggest you go back to unmodified host and controller subsystem with asserts enabled, and provide details of issue symptoms faced. You can use the nRF sniffer (https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE/Download) to capture air trace, in case if it can be of any help analyze the issue.

carlescufi, 2021-01-07:
@vrahane and @hardik-proxy could you please reply to @cvinayak?

vrahane, 2021-01-07:
Thanks @cvinayak and @carlescufi for the response. I have enabled asserts in our system as per your suggestion. I am still trying to overcome some internal issues. I wanted to send you the traces but without my patch the issue happens once every 8 hours after increasing log size buffers and stack sizes to accommodate asserts. Also, one of the issue I am facing is that the nRF Sniffer does not work over Mac OSX Catalina . Currently trying to get that setup so that I can send traces. With our current code base 

I am seeing an assert but the only information I can give you right now is the register dump because file names and line numbers cannot be included since they are compiled out to free up RAM. The current running thread seems to be the HCI receive thread. I am trying to enable the file name and line number for the Bluetooth controller using the ‘bt_ctlr_assert_handler()’ but seeing issues compiling that in. Including the app in the prototype is giving some errors which I need to take a look at. I got distracted with some other work but today I will get back to it.

```
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
 CONFIG_MAIN_STACK_SIZE=3072
 CONFIG_BT_RX_STACK_SIZE=3072
 CONFIG_IDLE_STACK_SIZE=160
 CONFIG_ISR_STACK_SIZE=2048
 CONFIG_PRIVILEGED_STACK_SIZE=1024

[20210106144935][00:00:41.808,959] <err> os: ***** BUS FAULT *****
[20210106144935][00:00:41.808,959] <err> os:   Precise data bus error
[20210106144935][00:00:41.808,990] <err> os:   BFAR Address: 0xe39b7d52
[20210106144935][00:00:41.808,990] <err> os: r0/a1:  0xffffffff  r1/a2:  0x200098e4  r2/a3:  0xffffffff
[20210106144935][00:00:41.808,990] <err> os: r3/a4:  0xe39b7d52 r12/ip:  0x00000000 r14/lr:  0x000270c5
[20210106144935][00:00:41.808,990] <err> os:  xpsr:  0x61000000
[20210106144935][00:00:41.809,020] <err> os: Faulting instruction address (r15/pc): 0x0002714c
[20210106144935][00:00:41.809,020] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[20210106144935][00:00:41.809,020] <err> os: Current thread: 0x20002c48 (unknown)

```

```
[20210106144935][00:00:42.175,323] <err> os: Halting system
Connection closed by foreign host.

```

cvinayak, 2021-01-07:
@vrahane 
you can run the following to get the filename and line number:
`arm-none-eabi-addr2line -e "<path to zephyr.elf file>" 0x0002714c`

vrahane, 2021-01-07:
Thanks, that helped a bit:
`zepyr/subsys/net/buf.c`
```
 199 static uint8_t *data_alloc(struct net_buf *buf, size_t *size, k_timeout_t timeout)
 200 {
 201         struct net_buf_pool *pool = net_buf_pool_get(buf->pool_id);
 202 
 203         return pool->alloc->cb->alloc(buf, size, timeout);
 204 }
 ```

cvinayak, 2021-01-08:
@vrahane This seems like stack overflow issue.

Do you still see the "tx buffer overflow error"? or you hit the "***** BUS FAULT *****" first after 8 hours?

If this is a BUS Fault issue, you can turn on CONFIG_THREAD_ANALYZER_AUTO=y to check if all your threads have sufficient headroom.

vrahane, 2021-01-08:
@cvinayak We had turned on the `CONFIG_THREAD_ANALYZER_AUTO` and there was no issue with the stacks. Right now what we are seeing is a fault after the device connects. If this crash was not present then we would see an overflow after 8 hours. Not sure if the issues are related but definitely could be.

vrahane, 2021-01-08:
Asserts are enabled this time and also logging is enabled. We did play around with it a bit and now are getting a crash consistently:
```
[00:00:24.463,958] <err> os: ***** USAGE FAULT *****
[00:00:24.463,989] <err> os:   Illegal use of the EPSR
[00:00:24.463,989] <err> os: r0/a1:  0x2000da90  r1/a2:  0x20009424  r2/a3:  0xffffffff
[00:00:24.464,019] <err> os: r3/a4:  0xffffffff r12/ip:  0x200058e8 r14/lr:  0x00026967
[00:00:24.464,019] <err> os:  xpsr:  0x20000000
[00:00:24.464,019] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:24.464,019] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:24.464,050] <err> os: Current thread: 0x20002d38 (unknown)
[00:00:24.747,619] <err> os: Halting system
```

cvinayak, 2021-01-08:
please check the filename and line number for return address LR: 0x00027643

vrahane, 2021-01-08:
```
 199 static uint8_t *data_alloc(struct net_buf *buf, size_t *size, k_timeout_t timeout)
 200 {
 201         struct net_buf_pool *pool = net_buf_pool_get(buf->pool_id);
 202 
 203         return pool->alloc->cb->alloc(buf, size, timeout);
 204 }
 ```
 `/Users/vipul/dev/fw-signal/repos/zephyr/subsys/net/buf.c`

vrahane, 2021-01-08:
We are using Zephyr: 2.4.0 now
```
commit 7a3b253ced7333f5c0269387a7f3ed1dee69739d (HEAD, tag: zephyr-v2.4.0, tag: v2.4.0)
Author: Maureen Helm <maureen.helm@nxp.com>
Date:   Sat Sep 26 11:40:18 2020 -0500

    release: Zephyr 2.4.0
    
    Set version to 2.4.0.
    
    Signed-off-by: Maureen Helm <maureen.helm@nxp.com>
    ```

vrahane, 2021-01-09:
One investigating more and with @cvinayak’s help, we were able to come to conclusion for the crash:

There is memory corruption happening because of the bt_smp_pool.

`7491  .bss.db_hash_work
7492                 0x0000000020001c50       0x30 zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a(gatt.c.obj)
7493                 0x0000000020001c50                db_hash_work
7494  .bss.gatt_sc   0x0000000020001c80       0x50 zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a(gatt.c.obj)
7495  .bss.bt_smp_pool
7496                 0x0000000020001cd0      0xf40 zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a(smp.c.obj)
7497  .bss.prio_recv_thread_data
7498                 0x0000000020002c10       0x98 zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a(hci_driver.c.obj)
7499                 0x0000000020002c10                prio_recv_thread_data
7500  .bss.recv_thread_data
7501                 0x0000000020002ca8       0x98 zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a(hci_driver.c.obj)
7502                 0x0000000020002ca8                recv_thread_data `

vrahane, 2021-01-11:
Update: I am seeing a similar crash after running the test for > 21 hours:
```
8:46
arch_system_halt (reason=reason@entry=0) at /Users/vipul/dev/fw-signal/repos/zephyr/kernel/fatal.c:32
32		}
(gdb) bt
#0  arch_system_halt (reason=reason@entry=0) at /Users/vipul/dev/fw-signal/repos/zephyr/kernel/fatal.c:32
#1  0x00023542 in k_sys_fatal_error_handler (reason=reason@entry=0, esf=esf@entry=0x20009700 <z_interrupt_stacks+992>) at /Users/vipul/dev/fw-signal/repos/zephyr/kernel/fatal.c:44
#2  0x000235da in z_fatal_error (reason=reason@entry=0, esf=esf@entry=0x20009700 <z_interrupt_stacks+992>) at /Users/vipul/dev/fw-signal/repos/zephyr/kernel/fatal.c:126
#3  0x00011a88 in z_arm_fatal_error (reason=reason@entry=0, esf=esf@entry=0x20009700 <z_interrupt_stacks+992>) at /Users/vipul/dev/fw-signal/repos/zephyr/arch/arm/core/aarch32/fatal.c:59
#4  0x00011f2a in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /Users/vipul/dev/fw-signal/repos/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:996
#5  0x00011b6c in z_arm_usage_fault () at /Users/vipul/dev/fw-signal/repos/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:102
#6  <signal handler called>
#7  0xd4ffbf98 in ?? ()
#8  0x45912e40 in ?? ()
```
This seems to happening in the `recv_thread` which hands of data from the controller's `prio_recv_thread`. Both are in the controller but not sure why the other one is crashing, I have increased the stack size for the `prio_recv_thread` to `4096` just for the test and after this crash happened I upped the stack size for the `recv_thread` to `4096` from `3072`.

```
(gdb) i r
r0             0x0                 0
r1             0x200054cc          536892620
r2             0x20                32
r3             0x20                32
r4             0x0                 0
r5             0x0                 0
r6             0x20001ff0          536879088
r7             0x20009700          536909568
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x0                 0
sp             0x200096d0          0x200096d0 <z_interrupt_stacks+944>
lr             0x23543             144707
pc             0x29f50             0x29f50 <arch_system_halt+16>
xpsr           0x61000004          1627389956
msp            0x200096d0          536909520
psp            0x200084a8          536904872
primask        0x0                 0
basepri        0x20                32
faultmask      0x0                 0
control        0x0                 0
fpscr          0x0                 0
(gdb) info symbol 0x200084a8
recv_thread_stack + 2856 in section noinit
(gdb) info symbol 0x200096d0
z_interrupt_stacks + 944 in section noinit
(gdb) info symbol 0x20001ff0
recv_thread_data in section bss
(gdb) info symbol 0x20009700
z_interrupt_stacks + 992 in section noinit
(gdb) info symbol 0x200054cc
lock in section bss
```

joerchan, 2021-01-11:
@vrahane I don't understand what you mean by the memory corruption being caused by the bt_smp_pool, could you explain your conclusion?

Have you tried enabling CONFIG_ASSERT? Maybe that could provide some clues.

vrahane, 2021-01-11:
The pool whose allocation pointer is 0 which causes the crash is actually part of the bt_smp_pool. Yes, As mentioned above CONFIG_ASSERT is now enabled. We also debugged quite a lot together.

joerchan, 2021-01-11:
> The pool whose allocation pointer is 0 which causes the crash is actually part of the bt_smp_pool. Yes, As mentioned above CONFIG_ASSERT is now enabled. We also debugged quite a lot together.

bt_smp_pool is not a net_buf pool. It is just a regular array. Is the pool pointer pointing to bt_smp_pool?

vrahane, 2021-01-11:
Yes, it is. I just added a “bkpt” just before the line that crashes in ‘data_alloc()’
```

 199 static uint8_t *data_alloc(struct net_buf *buf, size_t *size, k_timeout_t timeout)
 200 {
 201         struct net_buf_pool *pool = net_buf_pool_get(buf->pool_id);
 202 
 203         if (!pool || !pool->alloc || !pool->alloc->cb || !pool->alloc->cb->alloc) {
 204             printk("buf: %lu, size: %lu, timeout: %lu\n", buf, size, timeout);
 205             __asm("bkpt");
 206         }
 207 
 208         return pool->alloc->cb->alloc(buf, size, timeout);
 209 }
```

```
[00:00:12.415,069] <err> net_buf: NBD: alloc pool: 0x20001d3c, pool->alloc: 0x00000000,pool->alloc->cb: 0x20005480 pool->alloc->cb->alloc: 0x00000000

(gdb) p ((struct net_buf_pool *)0x20001d3c)
$17 = (struct net_buf_pool *) 0x20001d3c <bt_smp_pool+108>
```

vrahane, 2021-01-11:
Also, please check the following backtrace for it:

```
data_alloc (buf=0x2000e7d0 <net_buf_num_complete_pool>, size=0x20009184 <prio_recv_thread_stack+964>, timeout=...) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/net/buf.c:205
205	            __asm("bkpt");
(gdb) bt
#0  data_alloc (buf=0x2000e7d0 <net_buf_num_complete_pool>, size=0x20009184 <prio_recv_thread_stack+964>, timeout=...) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/net/buf.c:205
#1  0x00026ea2 in net_buf_alloc_len (pool=0x20000d3c <num_complete_pool>, size=<optimized out>, timeout=...) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/net/buf.c:330
#2  0x00030544 in net_buf_alloc_fixed (pool=<optimized out>, timeout=...) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/net/buf.c:372
#3  0x00014fac in bt_buf_get_evt (evt=<optimized out>, discardable=<optimized out>, timeout=...) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:8819
#4  0x0001d08e in prio_recv_thread (p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/controller/hci/hci_driver.c:103
#5  0x0002c8a2 in z_thread_entry (entry=0x1d07d <prio_recv_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /Users/vipul/dev/fw-signal/repos/zephyr/lib/os/thread_entry.c:29
#6  0xaaaaaaaa in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

joerchan, 2021-01-11:
@vrahane What is buf->pool_id? at the moment of the crash? It seems that the pointer to the pool is wrong.
num complete pool: `net_buf_alloc_len (pool=0x20000d3c <num_complete_pool>`
buf pointer belonging in net_buf_num_complete_pool: `data_alloc (buf=0x2000e7d0 <net_buf_num_complete_pool>`
Now we don't get from the buf pointer the correct pool: `net_buf: NBD: alloc pool: 0x20001d3c` returned from `net_buf_pool_get(buf->pool_id);`, appears buf->pool_id is `0x1000`, which gives you an invalid pointer.

vrahane, 2021-01-11:
I see, I am currently running the code with SMP disabled and L2CAP channel unregistered. Once I know for sure that this issue gets fixed by doing the above, I will go back and check the pool_id. It takes hours to reproduce the overflow issue, hence, do not want to stop the tests. I do have the following logs some of which I sort of added in the net but code and some I converted to be error logs so that I do not have too much logging information in output:

```
[00:00:00.042,449] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

--- 280 messages dropped ---
[00:00:00.042,633] <err> net_buf:  buf 0x2000e758 fifo 0x2000010c
[00:00:00.042,663] <err> net_buf: buf 0x2000e758 (old) ref 2 pool_id 4
[00:00:00.042,694] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:00.042,755] <err> net_buf: buf 0x2000e758 ref 3 pool_id 4 frags 0x00000000
[00:00:00.042,785] <err> net_buf: buf 0x2000e758 ref 2 pool_id 4 frags 0x00000000
[00:00:00.042,877] <err> net_buf: buf 0x2000e758 ref 1 pool_id 4 frags 0x00000000
[00:00:00.042,907] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

[00:00:00.042,938] <err> net_buf: NBD: pool 0x20000c64 size 258 timeout: 18446744073709551615
[00:00:00.042,968] <err> net_buf: allocated buf 0x2000e758
[00:00:00.042,968] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:00.042,999] <err> m
[00:00:00.042,053] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

--- 280 messages dropped ---
[00:00:00.042,236] <err> net_buf:  buf 0x2000e758 fifo 0x2000010c
[00:00:00.042,266] <err> net_buf: buf 0x2000e758 (old) ref 2 pool_id 4
[00:00:00.042,297] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:00.042,358] <err> net_buf: buf 0x2000e758 ref 3 pool_id 4 frags 0x00000000
[00:00:00.042,388] <err> net_buf: buf 0x2000e758 ref 2 pool_id 4 frags 0x00000000
[00:00:00.042,480] <err> net_buf: buf 0x2000e758 ref 1 pool_id 4 frags 0x00000000
[00:00:00.042,510] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

[00:00:00.042,541] <err> net_buf: NBD: pool 0x20000c64 size 258 timeout: 18446744073709551615
[00:00:00.042,572] <err> net_buf: allocated buf 0x2000e758
[00:00:00.042,602] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:00.042,633] <err> net_buf: buf 0x2000e758 (old) ref 1 pool_id 4
[00:00:00.042,694] <err> net_buf:  fifo 0x2000010c
[00:00:00.042,694] <err> net_buf:  buf 0x2000e758 fifo 0x2000010c
[00:00:00.042,724] <err> net_buf: buf 0x2000e758 (old) ref 2 pool_id 4
[00:00:00.042,846] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:00.043,029] <err> net_buf: buf 0x2000e758 ref 3 pool_id 4 frags 0x00000000
[00:00:00.043,060] <err> net_buf: buf 0x2000e758 ref 2 pool_id 4 frags 0x00000000
[00:00:00.043,151] <err> net_buf: buf 0x2000e758 ref 1 pool_id 4 frags 0x00000000
[00:00:00.043,182] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

[00:00:00.332,122] <err> net_buf: NBD: pool 0x20000c9c size 258 timeout: 18446744073709551615
[00:00:00.332,153] <err> net_buf: allocated buf 0x2000e788
[00:00:00.332,183] <err> net_buf: buf 0x2000e790 reserve 0
[00:00:00.332,275] <err> net_buf: buf 0x2000e788 ref 1 pool_id 5 frags 0x00000000
[00:00:00.332,275] <err> net_buf: NBD: unref buf: 0x2000e788, data: 0x2000f07c

[00:00:01.043,762] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

--- 35 messages dropped ---
[00:00:01.043,792] <err> net_buf: NBD: pool 0x20000c64 size 258 timeout: 18446744073709551615
[00:00:01.043,823] <err> net_buf: allocated buf 0x2000e758
[00:00:01.043,853] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:01.043,884] <err> net_buf: buf 0x2000e758 (old) ref 1 pool_id 4
[00:00:01.043,945] <err> net_buf:  fifo 0x2000010c
[00:00:01.043,975] <err> net_buf:  buf 0x2000e758 fifo 0x2000010c
[00:00:01.044,006] <err> net_buf: buf 0x2000e758 (old) ref 2 pool_id 4
[00:00:01.044,036] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:01.044,128] <err> net_buf: buf 0x2000e758 ref 3 pool_id 4 frags 0x00000000
[00:00:01.044,158] <err> net_buf: buf 0x2000e758 ref 2 pool_id 4 frags 0x00000000
[00:00:01.044,219] <err> net_buf: buf 0x2000e758 ref 1 pool_id 4 frags 0x00000000
[00:00:01.044,250] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

[00:00:01.044,281] <err> net_buf: NBD: pool 0x20000c64 size 258 timeout: 18446744073709551615
[00:00:01.044,311] <err> net_buf: allocated buf 0x2000e758
[00:00:01.044,342] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:01.044,372] <err> net_buf: buf 0x2000e758 (old) ref 1 pool_id 4
[00:00:01.044,433] <err> net_buf:  fifo 0x2000010c
[00:00:01.044,464] <err> net_buf:  buf 0x2000e758 fifo 0x2000010c
[... 51 lines cut ...]
[00:00:05.938,385] <err> net_buf: buf 0x2000e758 (old) ref 1 pool_id 4
[00:00:05.938,415] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:05.938,476] <err> net_buf: buf 0x2000e758 ref 2 pool_id 4 frags 0x00000000
[00:00:05.938,507] <err> net_buf: buf 0x2000e758 ref 1 pool_id 4 frags 0x00000000
[00:00:05.938,598] <err> net_buf: NBD: unref buf: 0x2000e758, data: 0x2000ee78

[00:00:06.236,236] <err> net_buf: allocated buf 0x2000e788
--- 1 messages dropped ---
[00:00:06.236,267] <err> net_buf: buf 0x2000e790 reserve 0
[00:00:06.236,297] <err> net_buf: NBD: pool 0x20000c64 size 258 timeout: 18446744073709551615
[00:00:06.236,297] <err> net_buf: allocated buf 0x2000e758
[00:00:06.236,328] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:06.236,328] <err> net_buf: buf 0x2000e758 (old) ref 1 pool_id 4
[00:00:06.236,389] <err> net_buf:  fifo 0x2000010c
[... cut ...]
[00:00:06.236,419] <err> net_buf: buf 0x2000e758 (old) ref 2 pool_id 4
--- 13 messages dropped ---
[00:00:06.236,785] <err> net_buf: buf 0x2000e760 reserve 0
[00:00:06.236,816] <err> net_buf: buf 0x2000e758 ref 2 pool_id 4 frags 0x00000000
[00:00:06.236,846] <err> net_buf: buf 0x2000e758 ref 1 pool_id 4 frags 0x00000000
--- 2 messages dropped ---
[00:00:06.298,553] <err> net_buf: allocated buf 0x2000e788
[00:00:06.298,553] <err> net_buf: buf 0x2000e790 reserve 0
[00:00:06.298,583] <dbg> bt_att.bt_att_recv: Received ATT chan 0x2000d298 code 0x02 len 2
```

vrahane, 2021-01-11:
Update: @cvinayak @joerchan The over flow is still there even after disabling L2CAP and BT_SMP. The crash is no more but overflow still exists.
```
[20210110204916][18:04:05.263,549] <err> bt_ctlr_hci: Tx Buffer Overflow
[20210110204916][18:04:05.263,580] <wrn> bt_hci_core: Data buffer overflow (link type 0x01)
[20210110204916][18:04:05.263,610] <err> bt_conn: Unable to send to driver (err -55)
[20210110204916][18:04:05.400,390] <dbg> proxy_id_sdk.peer_connected: proxy_id_peer: peer connected, conn=0x20000f30 addr=d5:5f:62:65:a4:50
[20210110204916][18:04:05.469,543] <err> bt_ctlr_hci: Tx Buffer Overflow
[20210110204916][18:04:05.469,573] <wrn> bt_hci_core: Data buffer overflow (link type 0x01)
[20210110204916][18:04:05.469,573] <err> bt_conn: Unable to send to driver (err -55)
[20210110204916][18:04:05.493,041] <err> bt_ctlr_hci: Tx Buffer Overflow
[20210110204916][18:04:05.493,041] <wrn> bt_hci_core: Data buffer overflow (link type 0x01)
[20210110204916][18:04:05.493,072] <err> bt_conn: Unable to send to driver (err -55)
[20210110204916][18:04:05.497,680] <err> bt_ctlr_hci: Tx Buffer Overflow
[20210110204916][18:04:05.497,741] <wrn> bt_hci_core: Data buffer overflow (link type 0x01)
[20210110204916][18:04:05.497,741] <err> bt_conn: Unable to send to driver (err -55)
```

hardik-proxy, 2021-01-12:
[image]

Here's the snapshot from the sniffer. We will be sharing a sniffer file on Slack.

vrahane, 2021-01-12:
@joerchan What do you suggest we look at next ?

hardik-proxy, 2021-01-12:
[image]

[image]

We have tried to match the time stamp on console logs and Sniffer software to get the frame near the error occurred. Above two images represents the same.

cvinayak, 2021-01-13:
@joerchan Is there any reason the ATT protocol would fragment the packet into 27 bytes after ATT Exchange MTU transaction and Data Length Update have completed?

See screenshot:
[image]

@vrahane and @hardik-proxy please include this length check to rule out memory corruption if there was any ACL data length violation: https://github.com/zephyrproject-rtos/zephyr/pull/31277

vrahane, 2021-01-13:
@cvinayak  Sure, we will include that. We are running tests at the moment without any issue for 27 hours straight. We will keep you guys updated. Thank you.

hardik-proxy, 2021-01-13:
@cvinayak we include #31277 but it didn't help to overcome tx buffer overflow issue. Overflow appeared after 15 hours of runtime.

joerchan, 2021-01-14:
> joerchan Is there any reason the ATT protocol would fragment the packet into 27 bytes after ATT Exchange MTU transaction and Data Length Update have completed?

Because the ATT requests can be queued, and they buffers are created before they are put into the queue. They must have been put into the queue before the MTU exchange was finished.

With the changes here : https://github.com/zephyrproject-rtos/zephyr/pull/31184 improving this wouldn't be too hard I think.

cvinayak, 2021-01-14:
@joerchan it is the 3rd ATT transaction response from the ATT server on the zephyr peripheral after the completion of ATT exchange MTU; and, over 200 ms and 400 ms after Length update and ATT exchange transactions have completed. See the screenshot here: https://github.com/zephyrproject-rtos/zephyr/issues/30378#issuecomment-759163656

@hardik-proxy ok, the added length check ensures that there is no linker layer buffer overwrites.

joerchan, 2021-01-14:
@vrahane Did you check the buffer pool_id? If it has been changed then that would be the root of the problem I think.
We should try to find out how this was changed, somewhere an invalid pointer probably leads to this issue.

It might be possible to use the below code to find out where the overwrite comes from.
Added to `subsys/bluetooth/host/buf.c` and calling `bt_buf_debug` from `bt_enable` in `subsys/bluetooth/host/hci_core.c`

```
#include <nrf.h>
static void debug_write_addr(uint8_t *address)
{
	uint8_t region = 0;

	NRF_MWU->REGION[region].START = (uint32_t)address;
	NRF_MWU->REGION[region].END = ((uint32_t)address) + 1;
	NRF_MWU->REGIONENSET = 1 << (region*2);

	NRF_MWU->INTENSET = 1 << (region * 2);
}

static void C_mwu_isr(uint32_t *sp)
{
	uint32_t pc;
	uint8_t region = 0;

  	/* Extract the stacked PC using the given stack pointer */
  	pc = sp[6];

	if (NRF_MWU->EVENTS_REGION[region].WA) {
		uint8_t *address = (uint8_t *)NRF_MWU->REGION[region].START;

		BT_ERR("Address %u", *address);

		BT_ERR("MWU write %d stacked PC: 0x%x", *address, pc);
		NRF_MWU->EVENTS_REGION[region].WA = 0;	
	}
}

void mwu_isr(void)
{
  	if ((uint32_t)__builtin_return_address(0) == 0xFFFFFFFDUL) {
  		C_mwu_isr((uint32_t*)__get_PSP());
  	} else {
  		C_mwu_isr((uint32_t*)__get_MSP());
  	}
}

void bt_buf_debug(void)
{
	struct net_buf *num_complete_buf = &net_buf_num_complete_pool[0];

	debug_write_addr(&num_complete_buf->pool_id);

	IRQ_DIRECT_CONNECT(MWU_IRQn, 0, mwu_isr, IRQ_ZERO_LATENCY);
	irq_enable(MWU_IRQn);
}
```

vrahane, 2021-01-15:
@joerchan apologies for the delay, we are actually testing our stuff with migrating from L2CAP to GATT to avoid the issue from happening in the first place. We do still see large L2CAP frames being sent. could fine the actual reason for it. I will keep you updated with the progress. We are sort of on a strict timeline hence, cannot go back at debugging the crash more. Will get back to it once we stop seeing the overflow.

vrahane, 2021-01-19:
@joerchan The patch above does not work. Can you please provide a git diff instead.

```
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:166:38: error: 'net_buf_num_complete_pool' undeclared (first use in this function)
  166 |  struct net_buf *num_complete_buf = &net_buf_num_complete_pool[0];
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:166:38: note: each undeclared identifier is reported only once for each function it appears in
In file included from /Users/vipul/dev/fw-signal/repos/zephyr/include/arch/arm/aarch32/arch.h:27,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/arch/cpu.h:19,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/kernel_includes.h:38,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/kernel.h:17,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/zephyr.h:18,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:10:
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:170:43: error: 'IRQ_ZERO_LATENCY' undeclared (first use in this function)
  170 |  IRQ_DIRECT_CONNECT(MWU_IRQn, 0, mwu_isr, IRQ_ZERO_LATENCY);
      |                                           ^~~~~~~~~~~~~~~~
/Users/vipul/dev/fw-signal/repos/zephyr/include/arch/arm/aarch32/irq.h:117:44: note: in definition of macro 'ARCH_IRQ_DIRECT_CONNECT'
  117 |  z_arm_irq_priority_set(irq_p, priority_p, flags_p); \
      |                                            ^~~~~~~
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:170:2: note: in expansion of macro 'IRQ_DIRECT_CONNECT'
  170 |  IRQ_DIRECT_CONNECT(MWU_IRQn, 0, mwu_isr, IRQ_ZERO_LATENCY);
      |  ^~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/cmake/cmake --build /Users/vipul/dev/fw-signal/build
make: *** [sdk] Error 1
Vipuls-MacBook-Pro-16-inch-2019:fw-signal vipul$ vim /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c

repos/zephyr/subsys/bluetooth/host/hci_core.c
166:	struct net_buf *num_complete_buf = &net_buf_num_complete_pool[0];

Press ENTER or type command to continue
```

vrahane, 2021-01-19:
@joerchan the patch above does not build. Can you please give me the `git diff` instead. I just put it in `hci_core.c`. I also put `nrf.h` in the path.

```
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:167:38: error: 'net_buf_num_complete_pool' undeclared (first use in this function)
  167 |  struct net_buf *num_complete_buf = &net_buf_num_complete_pool[0];
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:167:38: note: each undeclared identifier is reported only once for each function it appears in
In file included from /Users/vipul/dev/fw-signal/repos/zephyr/include/arch/arm/aarch32/arch.h:27,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/arch/cpu.h:19,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/kernel_includes.h:38,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/kernel.h:17,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/include/zephyr.h:18,
                 from /Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:10:
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:171:43: error: 'IRQ_ZERO_LATENCY' undeclared (first use in this function)
  171 |  IRQ_DIRECT_CONNECT(MWU_IRQn, 0, mwu_isr, IRQ_ZERO_LATENCY);
      |                                           ^~~~~~~~~~~~~~~~
/Users/vipul/dev/fw-signal/repos/zephyr/include/arch/arm/aarch32/irq.h:117:44: note: in definition of macro 'ARCH_IRQ_DIRECT_CONNECT'
  117 |  z_arm_irq_priority_set(irq_p, priority_p, flags_p); \
      |                                            ^~~~~~~
/Users/vipul/dev/fw-signal/repos/zephyr/subsys/bluetooth/host/hci_core.c:171:2: note: in expansion of macro 'IRQ_DIRECT_CONNECT'
  171 |  IRQ_DIRECT_CONNECT(MWU_IRQn, 0, mwu_isr, IRQ_ZERO_LATENCY);
      |  ^~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/cmake/cmake --build /Users/vipul/dev/fw-signal/build
make: *** [sdk] Error 1
```

joerchan, 2021-01-19:
@vrahane Wrong file, put it in `subsys/bluetooth/host/buf.c`
And either enable `CONFIG_ZERO_LATENCY_IRQS`, or replace the flag `IRQ_ZERO_LATENCY` with `0`.

Updated my previous comment to the full diff.

vrahane, 2021-01-21:
@carlescufi @joerchan @cvinayak We are not seeing the "Tx buffer overflow" issue happening with the Nordic Soft Device, we have been testing for over 17 hours now. We will keep you updated.

cvinayak, 2021-01-22:
@vrahane "Tx buffer overflow"  message is not present in the Nordic SoftDevice based controller, but the other two from the host:
```
bt_hci_core: Data buffer overflow (link type 0x01)
bt_conn: Unable to send to driver (err -55)
```

I am assuming you are not getting those. You will need to look for other functional failures, if any.

At this point, we can narrow down to the Zephyr controller if you do not see any functional failures when using the Nordic SoftDevice based controller.

To rule out ISR Latencies, please try the following option to retest with Zephyr controller:
```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_LLL_PRIO=0
CONFIG_BT_CTLR_ULL_HIGH_PRIO=1
CONFIG_BT_CTLR_ULL_LOW_PRIO=1
```
If that does not solve, then try this:
```
CONFIG_ZERO_LATENCY_IRQS=y
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_ZLI=y
```

(reference: https://github.com/zephyrproject-rtos/zephyr/issues/29994#issuecomment-739623736)

cvinayak, 2021-01-27:
@vrahane @hardik-proxy did you get a chance to verify the issue with modified controller ISR configuration stated here: https://github.com/zephyrproject-rtos/zephyr/issues/30378#issuecomment-765087304

cvinayak, 2021-01-30:
@carlescufi Compiler optimizations  without use of Data Synchronization Barrier is causing Tx Buffer Leak. Was able to reproduce incorrect memory usage by controller's tx buffer loop through Thread->ULL->LLL->ULL->Thread data path.

[image]

joerchan, 2021-02-01:
@vrahane @hardik-proxy If you have time could you verify that the fix solves the issue you were seeing? As I recall you could reproduce it quite quickly in the last setup.

PR #31813 (2021-02-02): Bluetooth: controller: Fix Tx Buffer Overflow
Fix Tx Buffer Overflow caused by uninitialized node_tx
memory being used by ULL ISR context due to Compiler
Instructions Reordering in the use of MFIFO_ENQUEUE.

The MFIFO last index was committed before the data element
was stored in the MFIFO due to Compiler Instructions
Reordering.

This is fixed now by adding a Data Memory Barrier
instruction alongwith a compiler memory clobber.

Fixes #30378.
commit 1af2b91c (2021-02-02): Bluetooth: controller: Fix Tx Buffer Overflow

Fix Tx Buffer Overflow caused by uninitialized node_tx
memory being used by ULL ISR context due to Compiler
Instructions Reordering in the use of MFIFO_ENQUEUE.

The MFIFO last index was committed before the data element
was stored in the MFIFO due to Compiler Instructions
Reordering.

This is fixed now by adding a Data Memory Barrier
instruction alongwith a compiler memory clobber.

Fixes #30378.
closed 2021-02-02 by nashif