zephyrproject-rtos/zephyr #30384: Scheduler doesn't activate sleeping threads on native_posix

Tehieh, 2020-12-02:
**Describe the bug**
I have the following code built to run as a native_posix application:
```
#include <kernel.h>
#include <iostream>

constexpr int kStackSize = 1024;
K_THREAD_STACK_DEFINE(thread_stack, kStackSize);

bool _dummy_running = true;
void DummyThread(void*, void*, void*) {
  while(_dummy_running) {
    k_yield();
  }
}

void main(void)
{
  std::cout << "Create the thread." << std::endl;  
  struct k_thread dummy_thread;
  k_tid_t ret = k_thread_create(&dummy_thread, thread_stack, K_THREAD_STACK_SIZEOF(thread_stack), DummyThread, NULL, NULL, NULL, 1, 0, K_NO_WAIT);
  
  std::cout << "About to sleep." << std::endl;
  k_msleep(10);
  std::cout << "Finished sleeping." << std::endl;
  
  _dummy_running = false;
  k_thread_join(&dummy_thread, K_FOREVER);
  std::cout << "Exiting." << std::endl;
}
```
The code stops on the ``` k_msleep(10); ``` and has the following output:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1789-g47ebde30b9ea  ***
Create the thread.
About to sleep.
```
**Expected behavior**
The code reaches the line to  ```std::cout << "Exiting." << std::endl;```

**Environment (please complete the following information):**
 - OS: Linux, application running inside a docker container

aescolar, 2020-12-02:
@Tehieh That while loop is an infinite loop in that case. You need to be aware that in native_posix, code executes in 0 simulated time. So the code calls k_yield(), there is nothing to yield to, and continues looping, all in zero time.

You need to do something that causes it to spend time on that while loop. For example a `k_busy_wait(some_time);`

aescolar, 2020-12-05:
Closing assuming the issue has been clarified with the previous answer. If that is not the case, please reopen.

closed 2020-12-05 by aescolar