zephyrproject-rtos/zephyr #30386: Building confirmed images does not work

de-nordic, 2020-12-02:
**Describe the bug**
According to the documentation to following Kconfig options:
https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_MCUBOOT_SIGNATURE_KEY_FILE.html#cmdoption-arg-CONFIG_MCUBOOT_SIGNATURE_KEY_FILE
https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE.html?highlight=config_mcuboot#cmdoption-arg-CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE
It should be possible to build signed and confirmed image in one step.
It does not work, build fails.

**To Reproduce**
Best to reproduce on smp_svr sample as it has mcuboot ready configuration
Steps to reproduce the behavior, assuming running ../ from zephyr repo clone
1. Prepare smp_svr build with menuconfig
`west build -b nrf52840dk_nrf52840 --build-dir smp_svr zephyr/samples/subsys/mgmt/mcumgr/smp_svr/ -t menuconfig`
2. Change following Kconfig options:
`CONFIG_MCUBOOT_SIGNATURE_KEY_FILE=bootloader/mcuboot/root-rsa-2048.pem`
`CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE=y`
and close Kconfig
3. Build:
`west build --build-dir smp_svr`
4. See error
```
[177/177] Linking C executable zephyr/zephyr.elf
FAILED: zephyr/zephyr.elf zephyr/zephyr.hex zephyr/zephyr.bin zephyr/zephyr.lst zephyr/zephyr.stat zephyr/zephyr.signed.bin zephyr/zephyr.signed.confirmed.bin zephyr/zephyr.signed.hex zephyr/zephyr.signed.confirmed.hex
: && ccache <path_to_your_sdk>/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj  -o zephyr/zephyr.elf  -Wl,-T zephyr/linker_pass_final.cmd -Wl,-Map=<path_to_your_repos>/smp_svr/zephyr/zephyr_final.map -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/common/libarch__common.a zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a zephyr/arch/arch/arm/core/aarch32/cortex_m/mpu/libarch__arm__core__aarch32__cortex_m__mpu.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/lib/posix/liblib__posix.a zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a zephyr/subsys/mgmt/mcumgr/libsubsys__mgmt__mcumgr.a zephyr/subsys/net/libsubsys__net.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/flash/libdrivers__flash.a zephyr/drivers/serial/libdrivers__serial.a modules/nordic/lib..__modules__hal__nordic.a modules/mcumgr/lib..__modules__lib__mcumgr.a modules/tinycbor/lib..__modules__lib__tinycbor__zephyr.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"<path_to_your_sdk>/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/thumb/v7e-m/nofp" -L<path_to_your_repos>/smp_svr/zephyr -lgcc zephyr/arch/common/libisr_tables.a -mcpu=cortex-m4 -mthumb -mabi=aapcs -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -Wl,-u,_OffsetAbsSyms -Wl,-u,_ConfigAbsSyms -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--orphan-handling=warn && cd <path_to_your_repos>/smp_svr/zephyr && <path_to_your_home>/.local/lib/python3.7/site-packages/cmake/data/bin/cmake -E rename zephyr_final.map zephyr.map && <path_to_your_sdk>/arm-zephyr-eabi/bin/arm-zephyr-eabi-objcopy --gap-fill 0xff --output-target=ihex --remove-section=.comment --remove-section=COMMON --remove-section=.eh_frame zephyr.elf zephyr.hex && <path_to_your_sdk>/arm-zephyr-eabi/bin/arm-zephyr-eabi-objcopy --gap-fill 0xff --output-target=binary --remove-section=.comment --remove-section=COMMON --remove-section=.eh_frame zephyr.elf zephyr.bin && <path_to_your_sdk>/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -d -S zephyr.elf > zephyr.lst && <path_to_your_sdk>/arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf -e zephyr.elf > zephyr.stat && /usr/bin/python3.7 -m west.app.main sign --quiet --tool imgtool --tool-path <path_to_your_home>/.local/bin/imgtool --build-dir <path_to_your_repos>/smp_svr --bin --sbin <path_to_your_repos>/smp_svr/zephyr/zephyr.signed.bin --hex --shex <path_to_your_repos>/smp_svr/zephyr/zephyr.signed.hex -- --key <path_to_your_repos>/bootloader/mcuboot/root-rsa-2048.pem && /usr/bin/python3.7 -m west.app.main sign --quiet --tool imgtool --tool-path <path_to_your_home>/.local/bin/imgtool --build-dir <path_to_your_repos>/smp_svr --bin --sbin <path_to_your_repos>/smp_svr/zephyr/zephyr.signed.confirmed.bin --hex --shex <path_to_your_repos>/smp_svr/zephyr/zephyr.signed.confirmed.hex -- --key <path_to_your_repos>/bootloader/mcuboot/root-rsa-2048.pem --pad --confirm
Usage: imgtool sign [OPTIONS] INFILE OUTFILE
Try "imgtool sign -h" for help.

Error: no such option: --confirm
FATAL ERROR: command exited with status 2: <path_to_your_home>/.local/bin/imgtool sign --version 0.0.0+0 --align 4 --header-size 512 --slot-size 421888 --key <path_to_your_repos>/bootloader/mcuboot/root-rsa-2048.pem --pad --confirm <path_to_your_repos>/smp_svr/zephyr/zephyr.bin <path_to_your_repos>/smp_svr/zephyr/zephyr.signed.confirmed.bin
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: <path_to_your_home>/.local/bin/cmake --build <path_to_your_repos>/smp_svr
```

**Expected behavior**
Build signed and confirmed image.

**Impact**
Annoyance that provided Kconfig options does not behave as described.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.11.3
 - Commit SHA zephyrproject-rtos/zephyr:4cf1d4380ed9d83b5f0aed08cec441d414834040

utzig, 2020-12-02:
The `--confirm` flag exists since the alphas of 1.6.0, which happened in April. Any reasonable recent version should work, so `pip install --upgrade imgtool` might be needed.

de-nordic, 2020-12-02:
My bad. That was the case.

closed 2020-12-02 by de-nordic