zephyrproject-rtos/zephyr #30391: Unit Testing in Zephyr

jakub-uC, 2020-12-02:
## Introduction

While testing the modules written for Zephyr, I noticed that there is no environment to create Unit Tests. Currently, it is not possible to test a module isolated from others. This problem affects both Zephyr developers and those who develop applications based on it.

### Problem description

The "z_test" does not allow to create of module tests in isolation from included modules.
The "z_test" does not provide any mock functionality.

### Proposed change

Integrate Unity and Cmock with Zephyr. 
In the next step maybe we could add unit tests to CI? Unity allows for easy generation of test results into XML format.

## Detailed RFC

Proposal:
https://github.com/jakub-uC/zephyr/pull/1

### Dependencies

In my opinion, it is only added value and will not impact existing projects.

### Concerns and Unresolved Questions

Maintaining z_test and unit test framework at the same time

## Alternatives

Keep using z_test only. Some other vendors like Nordic will keep Unity and Cmock on their Zephyr fork.

carlescufi, 2020-12-02:
@PerMac and @nashif  is this something you want to discuss in the context of the testing working group, or should we leave Unit Testing out of that?

nashif, 2020-12-02:
adding unity and cmock support would be a great addition IMO. I would like to see concrete examples of what can be tested with the combo and if we should long term move to one single framework if we can cover everything ztest does with unity/cmock.

ztest does have some mocking features, those however were not used widely. mocking in general with zephyr is going to be hard, depending on what you are testing.

There was also some effort to integrate cmocka, need to check what happened with that..

PerMac, 2020-12-02:
I have no experience with Unity nor CMock at all. We can raise the topic and see if there is someone with some experience and eager to help within the testing working group.

carlescufi, 2020-12-02:
@jakub-uC side note: CMock uses Unity as a submodule, but west doesn't support submodules yet. I believe that Kamil K. from Nordic is working on adding submodule support to west to avoid the "hack" that is in the manifest in your branch now.

thomasstenersen, 2020-12-04:
@carlescufi you don't need submodule support AFAIK, Unity is a submodule of CMock for conveniece/self-testing. I seem to recall having them cloned separately when I worked with it a while back in the nRF5 SDK for Mesh.

There we have used CMock + Unity for a while. You can take a look at https://github.com/NordicSemiconductor/nRF5-SDK-for-Mesh/blob/master/CMake/UnitTest.cmake and https://github.com/NordicSemiconductor/nRF5-SDK-for-Mesh/blob/master/mesh/test for inspiration.

nordic-krch, 2020-12-04:
See user guide for how it is done in nRF Connect SDK: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_unity_testing.html
Short summary:
Since it is hard to remove modules from zephyr build (due to cross dependencies and kconfig options) it builds full zephyr with mocked modules as well. During linking calls to mocked modules are replaced with corresponding mocks which are prefix with `__wrap_`. Python script is using to parse the header before generating mocks: static inline function replaced with declarations, removing syscalls.
Mocks are indicated in CMakeLists.txt.
The main limitation is that kernel API cannot be mocked. That is because kernel is needed to start the test (jump to main). Maybe some CMake tricks could be done to apply linking trick only to the module under test.

carlescufi, 2020-12-06:
> @carlescufi you don't need submodule support AFAIK, Unity is a submodule of CMock for conveniece/self-testing. I seem to recall having them cloned separately when I worked with it a while back in the nRF5 SDK for Mesh.

Right, but it is convenient to have the version that CMock packs instead of having yet another dependency.
 
> There we have used CMock + Unity for a while. You can take a look at https://github.com/NordicSemiconductor/nRF5-SDK-for-Mesh/blob/master/CMake/UnitTest.cmake and https://github.com/NordicSemiconductor/nRF5-SDK-for-Mesh/blob/master/mesh/test for inspiration.

Yes, and you are able to do this thanks to the "hack" I mentioned (add another entry to the manifest for the submodule)

tsvehagen, 2021-04-09:
Has there been any further discussion regarding this? Personally I think it would be really nice to see something like this to ease with application testing as well.

henrikbrixandersen, 2022-11-02:
Seeing that we now have FFF support in the tree, can this be closed?

jakub-uC, 2022-11-02:
sure

closed 2022-11-02 by jakub-uC